evcc-io/evcc #6213: PV modus startet Laden des E-Autos, wenn die Batterie nicht mehr geladen wird

binderth, 2023-02-12:
### Describe the bug

Wenn ein E-Auto angesteckt ist und der Modus auf "pv" im evcc steht, wird das Auto in dem Moment geladen, wenn die Haus-Batterie nicht mehr geladen wird - was meist abends der Fall ist, wenn die PV-Leistung eh nicht mehr ausreichend ist:
[image]
hier im Bild zu sehen ab 16:36 wurde geladen, bei Haus-SoC 75% wurde gestoppt.

### Steps to reproduce

1. E-Auto anstecken
2. Warten, bis die Haus-Batterie nicht mehr geladen wird
3. zusehen, wie der Lade-Befehl an den Ladepunkt geht, an dem das Auto angesteckt ist
4. trotz nicht ausreichender PV-Leistung

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http

  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. Host must be set to the docker host's name
  #host: evcc.local
  host: 0.0.0.0

  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

  # per default, evcc should be available at http://evcc.local:7070

interval: 10s # control cycle interval
log: info

meters:
# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters
- type: custom
  name: PVErzeugung
  power:
    source: http
    uri: http://192.168.78.10:8080/rest/items/KOS_PVPower
    method: GET
    headers:
    - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]
- type: custom
  name: Netzbezug
  power:
    source: http
    uri: http://192.168.78.10:8080/rest/items/KOS_HomeConsumptionGrid
    method: GET
    headers:
    - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]
    scale: 1
- type: custom
  name: BYD10
  power:
    source: http
    uri: http://192.168.78.10:8080/rest/items/KOS_BatteryChargeDischarge
    method: GET
    headers:
    - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]
    scale: 1
  soc:
    source: http
    uri: http://192.168.78.10:8080/rest/items/KOS_BatterySoC
    method: GET
    headers:
    - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]

chargers:
# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/evcc-io/config#chargers
- type: template
  template: tinkerforge-warp
  fw2: true
  host: 192.168.78.18
  port: 1883
  topic: devices/Warp2
  timeout: 30s
  name: WarpWallbox

vehicles:
# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/evcc-io/config#vehicles
- name: 500e
  type: template
  template: offline
  title: Fiat 500e
  capacity: 22
- name: EV6
  type: custom
  title: Kia EV6 # display name for UI
  capacity: 77 # byttery capacity (kWh)
  soc: # battery soc (%)
    source: http
    uri: http://192.168.78.10:8080/rest/items/EV6_batteryStatus
    method: GET
    headers:
    - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]
  range: # optional electric range (km)
    source: http
    uri: http://192.168.78.10:8080/rest/items/EV6_availableRange 
    method: GET
    headers:
    - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]
  status: # FahrzeugStatus (%)
    source: http
    uri: http://192.168.78.10:8080/rest/items/EV6_vehicleStatus
    method: GET
    headers:
    - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]
  onIdentify: # set defaults when vehicle is identified
    mode: pv
    minSoC: 25
    targetSoC: 80

tariffs:
# tariffs are the fixed or variable tariffs
# cheap (tibber/awattar) can be used to define a tariff rate considered cheap enough for charging
  currency: EUR # (default EUR)
  grid:
    #type: fixed
    #price: 0.319 # [currency]/kWh
    type: tibber
    cheap: 0.25
    token: "xxx"
    homeid: "yyy"
  feedin:
    type: fixed
    price: 0.065 # [currency]/kWh

loadpoints:
# loadpoint describes the charger, charge meter and connected vehicle
- title: Warp
  charger: WarpWallbox
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  vehicle: 500e

site:
# site describes the EVU connection, PV and home battery
  title: Zuhause
  meters:
    grid: Netzbezug
    pvs:
    - PVErzeugung
    batteries:
    - BYD10
  prioritySoC: 66
  bufferSoC: 75
  residualPower: 600

mqtt:
  broker: 192.168.78.18:1883
  topic: servers/evcc

# sponsor token enables optional features (request at https://cloud.evcc.io)
sponsortoken: xxxx
plant: yyyy
```

### Log details

```text
zum Zeitpunkt leider kein DEBUG-Log vorhanden, Verhalten kann aber immer nachvollzogen werden, wenn:

* der aktuelle Haus-SoC > bufferSoC ist
* die Haus-Batterie nicht mehr geladen wird
* auch wenn PV-Leistung nicht mehr ausreichend ist
```

### What type of operating system are you running?

Docker container

### Version

0.112.5

VolkerK62, 2023-02-12:
Durch mehrere Diskussionen und einen Selbstversuch habe ich folgendes herausgefunden:
Ausgangssituation: Auto ist angesteckt, Modus PV, es wird nicht geladen, BatterySoc > bufferSoc

Solange die Hausbatterie geladen wird, startet die Fahrzeugladung nicht.
Wird die Hausbatterie nicht mehr geladen bzw. entladen (Kipppunkt konnte ich nicht testen), dann startet die Fahrzeugladung sofort, ohne Timer.

Offensichtlich hat bufferSoc keine Wirkung, solange die Batterie geladen wird.

andig, 2023-02-12:
Siehe https://github.com/evcc-io/evcc/discussions/1361

andig, 2023-02-12:
> Solange die Hausbatterie geladen wird, startet die Fahrzeugladung nicht.

Das ergibt wenig Sinn, da Batterieladung *immer* als verfügbare Leistung gewertet wird. Sollte das so sein wärs ein Bug, ist aber wirklich schwer zu glauben.

Mangels Logfile mache ich hier zu. Gerne weiter wenn es etwas Nachvollziehbares gibt.

andig, 2023-02-12:
Im Eingangspost (anderes Thema!) startet die Ladung nicht weil einfach nicht genug Leistung dafür vorhanden ist. Die Batterie zählt _nicht_ weil nicht bekannt ist, welche Leistung sie beisteuern könnte. Würde einmal geladen (Min+PV) so würde die Ladung bis `bufferSoc` fortgesetzt sofern die Batterie ausreichend Leistung liefert.

binderth, 2023-02-12:
> Im Eingangspost (anderes Thema!) startet die Ladung nicht weil einfach nicht genug Leistung dafür vorhanden ist. Die Batterie zählt nicht weil nicht bekannt ist, welche Leistung sie beisteuern könnte.

Problem ist ja eben, dass exakt dann der Ladevorgang beginnt, wenn die Haus-Batterie nicht mehr geladen wird und (und das sehe ich als Fehler) die PV-Leistung weiterhin nicht ausreicht, selbst mit `residualPower` nicht. 

Kanns morgen abend gern nochmal nachstellen, morgen sollte das Auto bewegt werden und <100% haben.

andig, 2023-02-13:
Ja, das wäre falsch.

VolkerK62, 2023-02-13:
Gestern diesen Test gemacht.
PV Modus, nur wenig PV Power, Hausbatterie wurde leicht geladen. bufferSoc auf 10% gestellt.
Erstmal ist nix passiert, Dann habe ich die Herdplatte eingeschaltet,
Batterie ging in die Entladung und das Auto fing sofort an zu laden.

```
Feb 12 12:20:24 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:24 ----
Feb 12 12:20:24 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:24 charge power: 0W
Feb 12 12:20:24 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:24 pv power: 839W
Feb 12 12:20:24 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:24 battery soc: 20%
Feb 12 12:20:24 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:24 battery power: -232W
Feb 12 12:20:25 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:25 grid power: -10W
Feb 12 12:20:25 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:25 grid powers: [-12 21 -19]W
Feb 12 12:20:25 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:25 grid currents: [-1.14 0.86 -1.03]A
Feb 12 12:20:25 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:25 site power: -143W
Feb 12 12:20:25 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:25 charge currents: [0 0 0]A
Feb 12 12:20:25 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:25 charger status: B
Feb 12 12:20:25 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:25 next soc poll remaining time: 47m14s
Feb 12 12:20:25 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:25 pv charge current: 0.62A = 0A + 0.62A (-143W @ 1p)
Feb 12 12:20:39 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:39 ----
Feb 12 12:20:39 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:39 charge power: 0W
Feb 12 12:20:39 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:39 pv power: 836W
Feb 12 12:20:40 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:40 battery soc: 21%
Feb 12 12:20:40 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:40 battery power: 14W
Feb 12 12:20:40 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:40 grid power: 1720W
Feb 12 12:20:40 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:40 grid powers: [1769 4 -52]W
Feb 12 12:20:40 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:40 grid currents: [7.75 0.86 -1.02]A
Feb 12 12:20:40 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:40 site power: 1834W
Feb 12 12:20:40 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:40 charge currents: [0 0 0]A
Feb 12 12:20:40 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:40 charger status: B
Feb 12 12:20:40 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:40 next soc poll remaining time: 46m59s
Feb 12 12:20:40 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:40 pv charge current: 0A = 0A + -7.97A (1834W @ 1p)
Feb 12 12:20:40 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:40 max charge current: 6A
Feb 12 12:20:40 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:40 charger enable
Feb 12 12:20:40 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:40 wake-up timer: start
Feb 12 12:20:54 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:54 ----
Feb 12 12:20:54 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:54 charge power: 400W
Feb 12 12:20:54 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:54 pv power: 837W
Feb 12 12:20:54 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:54 battery soc: 21%
Feb 12 12:20:54 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:54 battery power: 1059W
Feb 12 12:20:55 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:55 grid power: 1955W
Feb 12 12:20:55 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:55 grid powers: [722 4 1230]W
Feb 12 12:20:55 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:55 grid currents: [3.19 0.88 5.66]A
Feb 12 12:20:55 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:55 site power: 3114W
Feb 12 12:20:55 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:55 charge currents: [2.6 0 0]A
Feb 12 12:20:55 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:55 charger status: C
Feb 12 12:20:55 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:55 charge start energy: 99.3kWh
Feb 12 12:20:55 raspberrypi evcc[10762]: [lp-1  ] INFO 2023/02/12 12:20:55 start charging ->
Feb 12 12:21:11 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:11 soc estimated: 59.00% (vehicle: 59.00%)
Feb 12 12:21:11 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:11 vehicle soc: 59%
Feb 12 12:21:11 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:11 vehicle soc limit: 100%
Feb 12 12:21:11 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:11 vehicle range: 218km
Feb 12 12:21:11 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:11 pv charge current: 0A = 4.6A + -13.5A (3114W @ 1p)
Feb 12 12:21:11 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:11 ----
Feb 12 12:21:11 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:11 telemetry: charge: Δ2/3Wh @ 400W
Feb 12 12:21:11 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:11 charge power: 1200W
```

andig, 2023-02-13:
> Problem ist ja eben, dass exakt dann der Ladevorgang beginnt, wenn die Haus-Batterie nicht mehr geladen wird und (und das sehe ich als Fehler) die PV-Leistung weiterhin nicht ausreicht, selbst mit residualPower nicht.

@VolkerK62 also gleiche Fehlerbeschreibung?

VolkerK62, 2023-02-13:
Ich habe dann kurz nach dem Einschalten bufferSoc wieder hochgesetzt, dann fing der disable Timer an

```
Feb 12 12:20:55 raspberrypi evcc[10762]: [lp-1  ] INFO 2023/02/12 12:20:55 start charging ->
Feb 12 12:21:11 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:11 soc estimated: 59.00% (vehicle: 59.00%)
Feb 12 12:21:11 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:11 vehicle soc: 59%
Feb 12 12:21:11 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:11 vehicle soc limit: 100%
Feb 12 12:21:11 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:11 vehicle range: 218km
Feb 12 12:21:11 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:11 pv charge current: 0A = 4.6A + -13.5A (3114W @ 1p)
Feb 12 12:21:11 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:11 ----
Feb 12 12:21:11 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:11 telemetry: charge: Δ2/3Wh @ 400W
Feb 12 12:21:11 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:11 charge power: 1200W
Feb 12 12:21:11 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:11 pv power: 835W
Feb 12 12:21:11 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:11 battery soc: 21%
Feb 12 12:21:11 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:11 battery power: 1015W
Feb 12 12:21:11 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:11 grid power: -69W
Feb 12 12:21:12 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:12 grid powers: [-1271 9 1176]W
Feb 12 12:21:12 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:12 grid currents: [-5.54 0.9 5.48]A
Feb 12 12:21:12 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:12 site power: 1047W
Feb 12 12:21:12 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:12 charge currents: [5.5 0 0]A
Feb 12 12:21:12 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:12 detected active phases: 1p
Feb 12 12:21:12 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:12 charger status: C
Feb 12 12:21:12 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:12 next soc poll remaining time: 3h59m43s
Feb 12 12:21:12 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:12 soc estimated: 59.00% (vehicle: 59.00%)
Feb 12 12:21:12 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:12 vehicle soc: 59%
Feb 12 12:21:12 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:12 vehicle soc limit: 100%
Feb 12 12:21:12 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:12 vehicle range: 218km
Feb 12 12:21:12 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:12 pv charge current: 1.45A = 6A + -4.55A (1047W @ 1p)
Feb 12 12:21:24 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:24 ----
Feb 12 12:21:24 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:24 charge power: 1200W
Feb 12 12:21:24 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:24 pv power: 843W
Feb 12 12:21:24 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:24 battery soc: 21%
Feb 12 12:21:24 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:24 battery power: 1073W
Feb 12 12:21:25 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:25 grid power: -54W
Feb 12 12:21:25 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:25 grid powers: [-1287 19 1214]W
Feb 12 12:21:25 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:25 grid currents: [-5.59 0.89 5.69]A
Feb 12 12:21:25 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:25 site power: 1119W
Feb 12 12:21:25 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:25 charge currents: [5.5 0 0]A
Feb 12 12:21:25 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:25 detected active phases: 1p
Feb 12 12:21:25 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:25 charger status: C
Feb 12 12:21:25 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:25 next soc poll remaining time: 3h59m46s
Feb 12 12:21:25 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:25 soc estimated: 59.00% (vehicle: 59.00%)
Feb 12 12:21:25 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:25 vehicle soc: 59%
Feb 12 12:21:25 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:25 vehicle soc limit: 100%
Feb 12 12:21:25 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:25 vehicle range: 218km
Feb 12 12:21:25 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:25 pv charge current: 1.13A = 6A + -4.87A (1119W @ 1p)
Feb 12 12:21:39 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:39 ----
Feb 12 12:21:39 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:39 charge power: 1200W
Feb 12 12:21:39 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:39 pv power: 841W
Feb 12 12:21:40 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:40 battery soc: 21%
Feb 12 12:21:40 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:40 battery power: 1117W
Feb 12 12:21:40 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:40 grid power: -67W
Feb 12 12:21:40 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:40 grid powers: [-1278 0 1212]W
Feb 12 12:21:40 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:40 grid currents: [-5.55 0.91 5.69]A
Feb 12 12:21:40 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:21:40 site power: 1151W
Feb 12 12:21:40 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:21:40 charge currents: [5.5 0 0]A
[... cut ...]
Feb 12 12:22:25 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:22:25 site power: -152W
Feb 12 12:22:25 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:22:25 charge currents: [0 0 0]A
Feb 12 12:22:25 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:22:25 charger status: B
Feb 12 12:22:25 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:22:25 next soc poll remaining time: 3h59m44s
Feb 12 12:22:38 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:22:38 set charge mode: pv
Feb 12 12:22:38 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:22:38 ----
```

andig, 2023-02-13:
> @VolkerK62 also gleiche Fehlerbeschreibung? 

Ja?

VolkerK62, 2023-02-13:
Sorry, das hab ich überlesen. Ja.
Residualpower steht auf +100.

Ich glaube, das "Problem" ist https://github.com/evcc-io/evcc/discussions/1361#discussioncomment-1274270

BufferSoc wird als "Schalter" genommen um so eine Art "Min+PV" einzuschalten.
Das ist mMn okay, wenn es um das AUSschaltverhalten geht.
Aber ich denke, es ist falsch, wenn es um das EINschaltverhalten geht.

Ich werde den obigen Test heute Abend mal wiederholen, wenn keine PV-Erzeugung mehr da ist.
Ich vermute stark, dass auch dann eingeschaltet wird. (gegen dich wetten würde ich aber nicht ;-)

andig, 2023-02-13:
Also die Werte stimmen:

```
Feb 12 12:20:39 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:39 charge power: 0W
Feb 12 12:20:39 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:39 pv power: 836W
Feb 12 12:20:40 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:40 battery power: 14W
Feb 12 12:20:40 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:40 grid power: 1720W
Feb 12 12:20:40 raspberrypi evcc[10762]: [site  ] DEBUG 2023/02/12 12:20:40 site power: 1834W
```

Was nicht passt ist:

```
Feb 12 12:20:40 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:40 pv charge current: 0A = 0A + -7.97A (1834W @ 1p)
Feb 12 12:20:40 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:40 max charge current: 6A
Feb 12 12:20:40 raspberrypi evcc[10762]: [lp-1  ] DEBUG 2023/02/12 12:20:40 charger enable
```

Warum passiert das?

```go
case mode == api.ModeMinPV || mode == api.ModePV:
	targetCurrent := lp.pvMaxCurrent(mode, sitePower, batteryBuffered)

	var required bool // false
	if targetCurrent == 0 && lp.climateActive() {
		targetCurrent = lp.GetMinCurrent()
		required = true
	}

	// Sunny Home Manager
	if lp.remoteControlled(loadpoint.RemoteSoftDisable) {
		remoteDisabled = loadpoint.RemoteSoftDisable
		targetCurrent = 0
		required = true
	}

	err = lp.setLimit(targetCurrent, required)
}

```

Kann es sein, dass beim Auto Klimatisierung detektiert wurde? Sonst fällt mir kein guter Grund ein warum die 0 wieder auf 6 korrigiert werden sollte???

VolkerK62, 2023-02-13:
>Kann es sein, dass beim Auto Klimatisierung detektiert wurde?

Nein. Wird bei Bluelink nicht unterstützt.

andig, 2023-02-13:
Letzte Idee: bitte nochmal trace log davon, oder wie müssen eine Debugsession machen. Da ist ja sonst nix :/

binderth, 2023-02-13:
ok, ich starte meinen Docker grad mal mit TRACE neu und warte ab. könnte bissle größer werden das logfile dann! ;)

andig, 2023-02-13:
Ich hab in `lp.pvMaxCurrent()` noch was gefunden, aber noch nicht versucht, die Logik zu verstehen:

	// in MinPV mode or under special conditions return at least minCurrent
	if (mode == api.ModeMinPV || batteryBuffered || lp.climateActive()) && targetCurrent < minCurrent {
		return minCurrent
	}

Im Prinzip: falls abgeschaltet würde und MinPV, dann minimaler Strom. Analog machen wir das aber auch falls Batteriepuffer vorhanden ist. Das sieht erstmal unklar aus.

@binderth denke Log hat sich damit erledigt, Ursache gefunden.

binderth, 2023-02-13:
ok. alles klar. hab eh gerade rausgefunden, dass meine Batterie über Tags nicht über 10% kam. Habs grad noch simulieren können mit 5% bufferSoC, hat aber bei Start von evcc gleich bei 6% zu laden begonnen.
Falls noch Notwendigkeit besteht - kurz anstupsen, das trace-log ist eben fertig.

VolkerK62, 2023-02-13:
>das ging also noch nie richtig.

ist dir Frage, was "richtig" ist. Wenn man sich das Issue #1361 anschaut, ist es works as designed.

Webwanze, 2023-02-13:
Hallo,

eigentlich sollte bei Überschreiten von buffersoc in den Modus Min+PV geschaltet werden. Beim Unterschreiten zurück nach PV.  

Ob der Hausakku geladen wird ist nicht ist völlig relevant. 

Usecase für das Verhalten:
- Umladen des Hausakkus, damit dieser nivht bei 100% steht und am Folgetag wieder Platz hat. 
- Starten des Ladevorgangs wenn Hausakku voll und Fahrzeug angesteckt wird. Wichtig für ZeroExport und Inseln. 
- Verhindern von Ladeunterbrechungen bei Wolken usw.

thebrainkafka, 2023-02-13:
Ich hatte es so verstanden dass der Hausbatterie Speicher die Ladung unterstützt um kurzzeitige Unterbrechungen (Wolke z.b. damit kleiner Mindestleistung) zu verhindern.
Ein entladen der Hausbatterie über dies Hinaus halte ich für wenig sinnvoll (auch auf Basis der gängigen Batteriespeichergrößen)

Webwanze, 2023-02-13:
Kannst du ja einstellen, wie dein buffersoc sein soll.

binderth, 2023-02-13:
Der normale Use Case sind sicherlich 10kWh Speicher und 40-70kWh E-Auto. Und nein, wenn evcc automatisch den Lademodus wechselt empfinde ich das nicht sinnvoll.
Die Haus-Batterie zur Überbrückung von Wolken zu verwenden ist ein sinnvoller UseCase, aber es macht in der überwiegenden Mehrheit keinen Sinn, "teureren" Haus-Speicher-Strom über zweimal Wandeln in E-Auto Speicher zu laden.
Wer im Sommer verhindern will, mit voller Batterie aus der Nacht zu kommen, kann ja Ziel-Laden oder per Schalter/API den Lademodus am Morgen entsprechend auf MinPV ändern.

Aus User-Sicht ist ein Lademodus-Wechsel nur sinnvoll, wenn man das vorab so plant und drauf hingewiesen wird. Auch das ein Edge-Case: Start des Ladevorgangs mit "pv" und Ziel-Laden bis 8h morgen. Mit nicht ausreichend PV-Leistung stoppt "PV" nach eingestellten Intervall wie konfiguriert (3min sind glaub Standard). Wenn noch nicht fertig geladen, wird rechtzeitig nachgeladen. 
Das ist netzdienlich und gibt die Chance, mit genügend grosser Batterie dann doch den UseCase "ich möchte im Sommer möglichst viele Zyklen erreichen" dadurch zu lösen. Aber aus meiner Sicht sollte evcc dabei helfen, möglichst viel PV-Strom direkt zu verwenden.

Wer mag und eine entsprechend riesig dimensionierte Anlage hat, kann sich ja per API selbst einen "Battery"-Modus basteln. Ist aber nicht die Regelgröße, leider.

Webwanze, 2023-02-13:
Dann stellst du halt buffersoc auf 100 und hast was du möchtest. Ist ja auch der Default.  

Anlagen mit ZeroExport oder Inseln sind aber auf das Feature angewiesen.

VolkerK62, 2023-02-14:
Alle Usecase sind sicherlich sinnvoll, aber auch sehr individuell.
Es sollte die Möglichkeit geben, das man zusätzlich zu "nicht beenden" auch noch "neu starten" auswählen kann. 
Z.B. ein `bufferStart: true/false`.
Ja, es wäre ein neuer Parameter, aber aufgrund der immer wieder aufkommenden Diskussionen rund um `bufferSoc` und der großen Unterschiede zwischen den Usecase halte ich den für durchaus nützlich.

binderth, 2023-02-14:
> Dann stellst du halt buffersoc auf 100 und hast was du möchtest. Ist ja auch der Default.

Dann sind allerdings 90% der Anlagen außen vor, weil keine Inseln. 100 bufferSoC ist halt meist sinnfrei. 

> Nachtrag: Die Doku spricht von "Ladung nicht unterbrechen". 

Darauf beziehe ich mich und das sollte auch so bleiben. Ansonsten sehe

premultiply, 2023-02-14:
Ich ziehe aus der ganzen Diskussion nur folgendes Fazit:

Gewünscht ist, dass die Fahrzeugladung oberhalb bufferSoc zuverlässig und eindeutig (wie bei Min+PV) startet.
Das ist bisher nur unter auf den ersten Blick nicht ganz nachvollziehbaren Bedingungen der Fall.

Mir war das vorher nicht bewusst dass dies derzeit nicht immer so funktioniert wie dokumentiert.

binderth, 2023-02-14:
> Gewünscht ist, dass die Fahrzeugladung oberhalb bufferSoc zuverlässig und eindeutig (wie bei Min+PV) startet.

Im Modus 'pv' ist ein Start rein aus Batterie allerdings entgegen des Sinnes des Modus. Weil dann isses eben nicht mehr 'pv', sondern 'minpv'. Daher finde ich den von Volker ins Spiel gebrachten Schalter eine gute Idee.
Weil, wenn ich 'pv' gewählt habe, möchte ich mit PV-Überschuss laden, nicht mit Batterie-Überschuss. In der Regel können handelsübliche Akkus 5kW, das reicht zwar erstmal für 3p Laden, aber gerade wenn abends z.B. gekocht wird, eben nicht mehr.

Oder es gibt einen "Batterie" Modus, der explizit Nutzung der Batterie erlaubt. Dann aber mit Angabe der maximalen Batterieentladung. Und das wiederum führt zu nix.

Ergo: 'pv' sollte PV-Laden bleiben, mit 'bufferSoc' für Wolken o.ä. wie in der Doku beschrieben. Das sollte der Standardfall bei den meisten Anlagen sein.
Nächtliches Entladen der Batterie bei mega-Akkus kann durch Zielladen ja jederzeit erfolgen.

thebrainkafka, 2023-02-14:
> Ich ziehe aus der ganzen Diskussion nur folgendes Fazit:
>
> Gewünscht ist, dass die Fahrzeugladung oberhalb bufferSoc zuverlässig und eindeutig (wie bei Min+PV) startet. Das ist bisher nur unter auf [...]

Genau. Danke.
Und bis BufferSoc die Hausbatterie als Puffer nutzt.

thebrainkafka, 2023-02-14:
> 
> Ergo: 'pv' sollte PV-Laden bleiben, mit 'bufferSoc' für Wolken o.ä. wie in der Doku beschrieben. Das sollte der Standardfall bei den meisten Anlagen sein. Nächtliches Entladen der Batterie bei mega-Akkus kann durch Zielladen ja jederzeit erfolgen.

Exakt.

andig, 2023-02-14:
Das bedeutet im Klartext aber auch, dass PV sich >`bufferSoc` wie MinPV verhält sobald die Ladung gestartet ist.

> Jeder hat eben ein individuelles Setup.

Da haben wir tatsächlich eine absolute Sonderlösung für Dich gebaut.

> Gewünscht ist, dass die Fahrzeugladung oberhalb bufferSoc zuverlässig und eindeutig (wie bei Min+PV) startet. 

Das mag sein, ist aber einfach nicht möglich da (heute) keine Aussage über die Leistung der Batterie getroffen werden kann, es also nicht klar ist ob `minCurrent` erreichbar ist. Weitere "mal probieren ob" Logik ist zu vermeiden.

Was wir aber machen könnten wäre, dass die Ladung `bufferSoc` *aufrecht* erhalten wird wenn sie einmal läuft. Es gäbe also keinen automatischen Start, wohl aber Ladeerhaltung bis `bufferSoc`. Aus Anwendersicht brauchts also einen Schubs über `minPV`, dann würde `PV` die Ladung halten und abschalten.

Alternativ (und ich möchte das nur sehr ungern) bräuchte es noch einen `batteryDischargePower` Parameter mit dem sich berechnen liesse, ob ein weiteres Fahrzeug innerhalb dieser geladen werden könnte. Dann:
- würde das Limit aber auch voll ausgeschöpft
- weitere Verbraucher würden genau analog PV zur Unterbrechung führen
- es bräuchte einen weiteren Parameter

Das würde allerdings die Modi nicht weiter verwirren.

binderth, 2023-02-14:
> Was wir aber machen könnten wäre, dass die Ladung bufferSoc aufrecht erhalten wird wenn sie einmal läuft. Es gäbe also keinen automatischen Start, wohl aber Ladeerhaltung bis bufferSoc. Aus Anwendersicht brauchts also einen Schubs über minPV, dann würde PV die Ladung halten und abschalten.

und das ist soweit ich die Doku lese auch der entsprechende Use-Case:

1. zu geringe PV-Erzeugung bei Überschreitung des `bufferSoc`: `pv` bleibt auch aus (weil ja eben die dafür _eigtl._ nötigen Umgebungsvariablen fehlen)
2. Ladevorgang ist vor Überschreitung des `bufferSoc` zu gering: `pv` lädt weiterhin - bis `threshold` erreicht ist.

Auch:
Wenn jemand überhalb `bufferSoc` auch ohne PV-Zutun laden möchte, kann er `minPV` nutzen, `residualPower` einstellen oder direkt Ziel-Laden nutzen. 

> Alternativ (und ich möchte das nur sehr ungern) bräuchte es noch einen batteryDischargePower Parameter mit dem sich berechnen liesse, ob ein weiteres Fahrzeug innerhalb dieser geladen werden könnte. 

Dann wüsste evcc, ob und wieviel die Batterie beitragen könnte - und in diesem Fall, das käme, plädierte ich dann für einen extra `PVplusBattery` Modus (oder Schalter), der dann den PV-Überschuss-kW mit der maximalen Entladungs-kW addiert und dann in einem "selbstgemachter verfügbarer Strom" zusammenfasst. Abgesehen von den von Andi erwähnten Schwierigkeiten könnte evcc dann den "PVplusBattery"-Strom so behandeln wie bisher den PV-Überschuss - und entsprechend des sonstigen Hausverbrauches dann im `pv`-Lademodus abriegeln - _mit der Ausnahme, dass `bufferSoc` dann natürlich nicht mehr geht, weil die Batterie ja zu 100% in den verfügbaren eigenen Ladestrom einfliesst._

andig, 2023-02-14:
> plädierte ich dann für einen extra PVplusBattery Modus

Weitere Modi wirds noch weniger geben als weitere Parameter ;)

thebrainkafka, 2023-02-14:
> und das ist soweit ich die Doku lese auch der entsprechende Use-Case:
>
> 1. zu geringe PV-Erzeugung bei Überschreitung des `bufferSoc`: `pv` bleibt auch aus (weil ja eben die dafür _eigtl._ nötigen [...]

Exakt so lese ich es auch aus der Doku

VolkerK62, 2023-02-14:
>Alternativ (und ich möchte das nur sehr ungern) bräuchte es noch einen batteryDischargePower Parameter mit dem sich berechnen liesse, ob ein weiteres Fahrzeug innerhalb dieser geladen werden könnte

Ich denke, auch in Bezug auf "Redesign PV Modus" wäre dieser Parameter nützlich. Da gab es auch immer wieder Wünsche nach Batterieunterstützung.

andig, 2023-02-14:
Dann bliebe die Frage, was der tun soll:

- nur den Start ermöglichen?
- oder voll als "verfügbare" Leistung am Netzpunkt ausgeschöpft werden?

Ersteres würde am Loadpoint passieren, letzteres wäre analog der `aux` Zähler.

Webwanze, 2023-02-14:
Zweiteres. 

Batterie kann mit spezifizierter Leistung entladen werden, wenn SoC > buffersoc.

binderth, 2023-02-14:
> Batterie kann mit spezifizierter Leistung entladen werden, wenn SoC > buffersoc.

mir fehlt weiterhin eine schlüssige Erläuterung, warum im Modus `pv` eine reine Batterieentladung ohne PV-Erzeugung gestartet werden sollte. PV = PV-Überschuss ist ausreichend vorhanden!?

thebrainkafka, 2023-02-14:
> Dann bliebe die Frage, was der tun soll:
> 
> * nur den Start ermöglichen?
> * oder voll als "verfügbare" Leistung am Netzpunkt ausgeschöpft werden?
> 
> Ersteres würde am Loadpoint passieren, letzteres wäre analog der `aux` Zähler, aber eben nur >`bufferSoc`.

Den Start ermöglichen und puffern.

VolkerK62, 2023-02-14:
> PV = PV-Überschuss 

Aufgrund der Entwicklung (Zielladen, Tarif, CO2, Planner) sollte man sich nicht mehr so sehr an `PV`´ festhalten.
Der PV-Modus ist mittlerweile zu einem "Automatik" Modus gewachsen.

binderth, 2023-02-14:
> Der PV-Modus ist mittlerweile zu einem "Automatik" Modus gewachsen.

Dann brauchts aber einen "Automatik"-Modus. Ich sehe evcc wirklich in erster Linie als PV-Überschuss-Tool. Automatik-Modus geht Richtung EMS und sollte daher im parallel diskutierten "Profi-Modus" münden. 

und ganz ehrlich: Wenn `pv` nicht so konfiguriert werden kann, dass es PV-Überschuss verwendet und Wolken puffert -> dann brauch ich persönlich kein evcc, weil es am Thema vorbei geht.

VolkerK62, 2023-02-14:
>Batterie kann mit spezifizierter Leistung entladen werden, wenn SoC > buffersoc.

Nur mal als Gedankenspiel: 
`batteryDischargePower: 1000` und Soc > buffersoc ; Mindestladeleistung 1,4 kW

- Fall 1: Dann würde bei 300 W Überschuss nix passieren und bei 500 W Überschuss würde das Laden gestartet.
- Fall 2: Ladung läuft, Erzeugung geht zurück, Batterie wird mit mehr als 1000 W entladen, dann geht es in die Abschaltung, auch wenn Soc immer noch > bufferSoc.
-

binderth, 2023-02-14:
> buffersoc auf z.B. 95%: Batterie dient als Puffer, bis diese auf buffersoc entladen wurde. Dabei wird allerdings nicht geprüft, ob die Batterie überhaupt die Min-Leistung hergibt. Ggf. wird der Rest aus dem Netz geholt.

**Widespruch!** Ich möchte nicht erst einen Puffer haben, wenn die Batterie voll ist.
Nochmal: `pv` heißt erstmal PV und nicht Battery. und nur weil eine PV-Ladung mal Wolken überdauern soll, kann man nicht auf 95% warten. weil dann kann man den Puffer außer für 4 Monate kicken.... und täglich per yaml oder API den bufferSoc anzupassen kann nicht der Sinn von evcc sein.

Webwanze, 2023-02-14:
So ist das aber heute mit dem buffersoc.
Wie möchtest du es denn genau haben? Wie weit darf der Speicher geleert werden, wenn während des Ladevorgangs die Sonne untergeht?

binderth, 2023-02-14:
>  Wie möchtest du es denn genau haben? Wie weit darf der Speicher geleert werden, wenn während des Ladevorgangs die Sonne untergeht?

bis der `threshold` erreicht ist. So wie es heute ist. 3min. im Standard.

VolkerK62, 2023-02-14:
>bis der threshold erreicht ist. So wie es heute ist. 3min. im Standard.

das hat doch aber gar nix mit buffersoc zu tun, bzw. entspricht buffersoc: 100

binderth, 2023-02-14:
Ich rede davon, was ich mir wünsche (und wie `pv` funktioniert, wenn Auto während Überschuss vollständig geladen wird):

* Wähle Modus `pv`
* gehe davon aus, `pv` wird nicht geändert
* warte auf PV-Ladung
* Ladung wird ab Haus-batterie > `bufferSoc` nicht unterbrochen
* Wenn `threshold` erreicht ist, stoppt die Ladung bei unzureichen PV-Erzeugung

ich rede *nicht* davon
* wähle Modus `pv`
* irgendwann entscheidet evcc (intransparent) auf `minPV` zu wechseln
* ...

oder auch nicht von
* wähle Modus `pv`
* stelle zusätzlich Ziel-Laden im Planner ein
* ...

andig, 2023-02-14:
Entweder: beschränken wir uns jetzt hier darauf, *konkrete* Vorschläge mit *konkreten* Ergänzungen oder Gegenvorschlägen zu beantworten, oder ich mache das Issue zu. `bufferSoc` funktioniert heute genau wie er designed wurde, ob das gut oder schlecht ist. Mein Vorschlag das zu ändern steht oben. Ansonsten führt die Diskussion für mich in keine erkennbare Richtung.

Anders gesagt: wenn kein Konsens dann dieses Issue zu, ist allenfalls ein Dokumentationsthema. Konkrete neue Featurerequests bei Bedarf.

Oder: wir kommen wieder auf die Vorschläge zurück.

thebrainkafka, 2023-02-14:
+1 für deinen Vorschlag.
Start ermöglichen und puffern (bei Wolke z.b)

Webwanze, 2023-02-14:
> * Ladung wird ab Haus-batterie > `bufferSoc` nicht unterbrochen

Die Ladung wird also fortgesetzt, bis der soc den buffersoc unterschritten hat? Unabhängig vom threshold?

> * Wenn `threshold` erreicht ist, stoppt die Ladung bei unzureichend PV-Erzeugung
>   _(weil es evcc _nur_ interessiert, DASS eine Batterie zur Unterstützung ab deren Ladestand verwendet weren kann, aber nich _wieviel_ von der Batterie in kW beigetragen werden kann: sprich evcc nimmt bei einer Puffer-Ladung in Kauf, dass Netzbezug zustande kommt, wenn die Batterie nicht vollständig die Mindestladung erreichen kann)_

Angenommen die Hausbatterie bringt die notwendige Leistung nicht und PV ist auch mau, dann wird aus dem Netz gezogen. Logisch. Dann wird der threshold überschritten und nach delay wird die Ladung unterbrochen.
> 
Das widerspricht der Aussage von oben: 
> * Ladung wird ab Haus-batterie > `bufferSoc` nicht unterbrochen

Sorry, aber ich versteh' es echt nicht.
Bei der aktuellen Implementierung ist es so, dass der Hausakku bis buffersoc zum Laden des Autos verwendet werden darf. Dabei wird die Ladeleistung heruntergefahren. Auf Min - in der Regel 6 A. Wo ist hier der Unterschied zu deinem 2. Absatz.

Webwanze, 2023-02-14:
+1 für den Vorschlag.
Start ermöglichen und puffern (bei Wolke z.b)

Sollte der Start erst erfolgen, wenn die Hausbatterie nicht mehr geladen wird, so würde ich das als Bug einordnen.

binderth, 2023-02-14:
+1 für den Vorschlag:
nur puffern (bei Wolke z.b)

+1 für den Vorschlag:
Start aus Batterie durch Schalter ermöglichen (bufferStart: true/false)

VolkerK62, 2023-02-14:
@Webwanze 
>Sollte der Start erst erfolgen, wenn die Hausbatterie nicht mehr geladen wird, so würde ich das als Bug einordnen.

Bist du dir da sicher? Dann passiert doch folgendes:
Batterie < buffersoc und wird geladen. In dem Moment, wo buffersoc überschritten wird, startet die Ladung um dann kurz danach zu stoppen, weil bis buffersoc entladen wurde.

Webwanze, 2023-02-14:
@VolkerK62 
Die Alternative wäre wie oben beschrieben: Die PV quält sich noch ewig und lädt den Hausakku stundenlang mit ein paar Watt. Wenn dann garnichts mehr geht, startet der Ladevorgang aus der Hausbatterie bis buffersoc unterschritten wurde.
Da finde ich die andere Variante besser/logischer.

Webwanze, 2023-02-14:
Mit den Parametern 'disable', 'threshold', 'delay' kann man doch eigentlich den UseCase der Wolken ganz gut abbilden. Es wird für eine gewisse Zeit weitergeladen, auch wenn PV gerade nicht reicht. Ist ein Hausakku vorhanden, dann kommt's halt (teilweise) von dort. Das Wolkenthema kann man doch eigentlich unabhängig von 'buffersoc' sehen, finde ich.

'buffersoc' zusammen mit 'maxbatterydischargepower' würde ich einfach unter 'aux' anrechnen. Fertig.

VolkerK62, 2023-02-14:
>'buffersoc' zusammen mit 'maxbatterydischargepower' würde ich einfach unter 'aux' anrechnen.

den "Trick" mit `aux`finde ich gut.

Webwanze, 2023-02-14:
Am Ende geht es ja um die Berechnung der site. Da ist die Hausbatterie sowieso ein Sonderfall, weil diese gleichzeitig Verbraucher als auch Quelle ist.

soc < prioritysoc: Hausakku wird nicht eingerechnet in site
prioritysoc < soc < buffersoc: aktuelle Ladeleistung Hausakku wird unter site einbezogen.
soc > buffersoc: sowohl Ladeleistung als auch mögliche Entladeleistung wird angerechnet.

VolkerK62, 2023-02-14:
Und mit der Möglichkeit die Entladeleistung festzulegen kann man das Verhalten beeinflussen. Gefällt mir.

thebrainkafka, 2023-02-14:
> Am Ende geht es ja um die Berechnung der site. Da ist die Hausbatterie sowieso ein Sonderfall, weil diese gleichzeitig Verbraucher als auch Quelle ist.
>
> soc < prioritysoc: Hausakku wird nicht [...]

Mit der Möglichkeit die Entladeleistung der Hausbatterie zu definieren. Top

andig, 2023-02-14:
> Start ermöglichen und puffern (bei Wolke z.b)

Also: es bleibt wie heute, der Ladevorgang wir aber nicht mehr automatisch gestartet. Läuft er einmal, so wir die Batterie bis Puffer genutzt. Mit `batteryDischargePower` gingen noch mehr und andere Szenarien; dazu bitte neues Issue mit konkretem Vorschlag.

andig, 2023-02-14:
Closed in https://github.com/evcc-io/evcc/commit/99595b72cdae97f64526052c69e7008d1cb3ff9e

/cc @VolkerK62

closed 2023-02-14 by andig