evcc-io/evcc #6218: Support overriding configuration via environment variables / additional config files

duckfullstop, 2023-02-12:
**Is your feature request related to a problem? Please describe.**
A common problem encountered when running any software in a container is how to handle passing secrets to the container. At present, secrets are stored in the main configuration file, which means that committing the configuration to public repositories (or even private ones) would leak any and all secrets.

At present, thanks to @andig's e220091, there is support for overriding most configuration using the environment (e.g `EVCC_MQTT.PASSWORD` - but it is not possible to override things such as the vehicles entry (i.e you can't do `EVCC_VEHICLES[0].PASSWORD`.

**Describe the solution you'd like**
There's a few possible options for alleviating this:

* Allow overriding of configuration arrays via environment variables (for example, `EVCC_VEHICLE_NAME`, `EVCC_VEHICLE_PASSWORD`, or maybe something like `EVCC_VEHICLE_superexpensivetesla_PASSWORD`) - this would require specific lookups on each key by hand
* Allow special configuration values that causes a lookup from the environment, e.g `password: ENV_VEHICLE_PASSWORD` would check `VEHICLE_PASSWORD`
  * We can't use the `evcc_` space because that's already used for Viper's configuration overloading (as e220091)
* Allow for loading multiple configuration files, e.g `evcc.yaml` and `secrets.yaml`, then merge them to produce a final in-memory configuration that can be used freely
  * e.g secrets.yaml may contain its own Vehicles configuration

All of these methods have their upsides and downsides, hence why I'm raising this as an issue for further discussion instead of going straight to PR.

**Describe alternatives you've considered**
Treating my entire configuration file as a secret isn't ideal, as it means I'm obscuring a large part of my infrastructure from my public repositories.

**Additional context**
Discussed in Slack: https://evccgroup.slack.com/archives/C01321PUJAD/p1676019600376789

StefanSchoof, 2023-02-12:
Related https://github.com/evcc-io/evcc/issues/405

andig, 2023-02-12:
> there is support for overriding most configuration using the environment

is that really true? That would already be great, but I didnt test it yet. Does that really work when reading from the unmarshaled config struct or does it require using the viper functions for reading?

duckfullstop, 2023-02-13:
@andig Yup, try setting `EVCC_SITE.TITLE` in your environment and it'll override whatever's in your config!

andig, 2023-02-14:
Wow, that's better then I thought. Seems I can remove any calls to `viper.Get...` and replace them with accessing the global config like everywhere else. That is really good news as it makes the code more consistent. Always wanted to experiment but never got around to try it.

For this issue the solution should be within Viper, I'd suggest to add an issue there:

    EVCC_VEHICLE_0_PASSWORD

should work for things that are slices.

goebelmeier, 2023-02-14:
Just so I understand, this would mean that we could remove the `--sqlite` command line parameter and just pass the path via a `EVCC_DATABASE_DSN` ENV variable?

```
# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>
```

andig, 2023-02-14:
According to what @duckfullstop said yes. Just give it a try?

StefanSchoof, 2023-02-14:
There are some issues on viper that sound similar: https://github.com/spf13/viper/issues/761 https://github.com/spf13/viper/issues/935

closed 2023-02-16 by andig

andig, 2023-02-17:
> @andig Yup, try setting `EVCC_SITE.TITLE` in your environment and it'll override whatever's in your config (at runtime)!

@duckfullstop OT, but 

    EVCC_DATABASE_DSN=/tmp/foo.db3 evcc

doesn't work for me.

duckfullstop, 2023-02-17:
> > @andig Yup, try setting `EVCC_SITE.TITLE` in your environment and it'll override whatever's in your config (at runtime)!
> 
> 
> 
> @duckfullstop OT, but 
> 
> 
> 
>     EVCC_DATABASE_DSN=/tmp/foo.db3 evcc
> 
> 
> 
> doesn't work for me.
> 
> 

Did you try `EVCC_DATABASE.DSN`? For whatever reason viper wants periods in the middle of environment variables

andig, 2023-02-17:
See https://github.com/evcc-io/evcc/pull/6286. `.` doesn't work art least on mac.

andig, 2023-08-16:
Also not recent comments like https://github.com/evcc-io/evcc/discussions/9363#discussioncomment-6713287

StefanSchoof, 2023-12-06:
Es wurde gerade https://github.com/spf13/viper/pull/1429 gemergt. Möglicherweise lässt sich das Problem jetzt lösen.