evcc-io/evcc #6231: positive dislabe threshold does not work

domerich, 2023-02-13:
### Describe the bug

my inverter is not supported (hybrid, FoxESS H3) so zero power usage means actually the battery is being drained.
so I need to stop PV charging with disable: 100

but evcc does not do anything with this 100 W head room.
currently I am using a virtual meter in vzlogger to add 100W offset but this is an ugly workaround messing up the database with bullshit data... 

### Steps to reproduce

1. set positive disable and see that nothing happens 

### Configuration details

```yaml
(V1.0.6 because here e go chargers are free)
```

### Log details

```text
cannot access
```

### What type of operating system are you running?

Linux

### Version

1.0.6

domerich, 2023-02-13:
> 

hello, do I did not I thought I just needed to do this with the load points. 
as I cannot use my smart meter I thought there is no point in residual value
so what would my config look like? 

thank you

domerich, 2023-02-13:
for the Nachwelt: this is how I get no error

`
site:
  title: My home
  meters:
    grid: my_grid
  residualPower: 100
`

VolkerK62, 2023-02-13:
Keep it at +100.  If the car is charging, it will left 100W for feedin.

domerich, 2023-02-13:
> Keep it at +100. If the car is charging, it will left 100W for feedin.

ok thanks will try tomorrow