evcc-io/evcc #6237: Doku poll verfeinern

thebrainkafka, 2023-02-14:
Was halltet ihr hiervon:

```diff
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
       poll:
         # poll mode defines under which condition the vehicle API is called:
-        #   charging: update vehicle ONLY when charging (this is the recommended default)
-        #   connected: update vehicle when connected (not only charging), interval defines how often
-        #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
+        #   charging: update vehicle when connected and until charge process is done (w.r.t. targetSoc) (this is the recommended default)
+        #   connected: update vehicle when connected even if charge process is done (w.r.t. targetSoc)
+        #   always: always update vehicle regardless of connection state (only supported for single vehicle)
         mode: charging
-        # poll interval defines how often the vehicle API may be polled if NOT charging
+        # poll interval defines how often the vehicle API may be polled under above conditions and if NOT charging
+        # if charging the interval is defined by the cache key on the vehicle definition (default 15m)
         interval: 60m

```

_Originally posted by @nename0 in https://github.com/evcc-io/evcc/discussions/6139#discussioncomment-4904968_

thebrainkafka, 2023-02-14:
Von nename0.

nename0, 2023-02-14:
Gerade gemacht: #6238
😉

thebrainkafka, 2023-02-14:
Perfekt. Danke

closed 2023-02-14 by thebrainkafka