evcc-io/evcc #6239: Experimental: unsichtbare Einträge in Herstellerliste bei Fahrzeug hinzufügen

VolkerK62, 2023-02-14:
In der Herstellerliste sind wohl oberhalb von Aiways noch weiter Auswahlmöglichkeiten die aber nicht angezeigt werden. Je nachdem, welche der "Leerzeilen" man auswählt, ändert sich das Fenster mit den Parametereingaben.
Anscheinend sind nur die Bezeichnungen unsichtbar.

[image]

andig, 2023-02-14:
@VolkerK62 kannst Du mal in der Browserkonsole schauen was da als XHR Request empfangen wird (im Netzwerk Tab)?

VolkerK62, 2023-02-15:
Ich hoffe, ich habe dich richtig verstanden. Wenn nicht, musst du mir über die Straße helfen ;-)

Da sind zwei Einträge (toml und telemetry):

toml:
```
[footer]

[footer.community]
greenEnergy = "Sonne"
greenEnergySub1 = "über evcc geladen"
greenEnergySub2 = "seit Oktober 2022"
greenShare = "Sonnenanteil"
greenShareSub1 = "der Leistung kommt"
greenShareSub2 = "von PV & Hausakku"
power = "Ladeleistung"
powerSub1 = "{activeClients} von {totalClients} Nutzern"
powerSub2 = "laden..."
tabTitle = "Live-Community"

[footer.savings]
footerLong = "{percent}% Sonnenenergie"
footerShort = "{percent}% Sonne"
modalTitle = "Auswertung Ladeenergie"
percentGrid = "{grid} kWh Netz"
percentSelf = "{self} kWh Sonne"
percentTitle = "Sonnenenergie"
priceFeedIn = "{feedInPrice} Einspeisung"
priceGrid = "{gridPrice} Netz"
priceTitle = "Energiepreis"
savingsComparedToGrid = "gegenüber Netzbezug"
savingsTitle = "Ersparnis"
savingsTotalEnergy = "{total} kWh geladen"
since = "seit {since}"
tabTitle = "Meine Daten"

[footer.sponsor]
becomeSponsor = "Sponsor werden"
confetti = "Lust auf Konfetti?"
confettiPromise = "Es gibt auch Sticker und digitales Konfetti"
sticker = "…oder evcc Sticker?"
supportUs = "Unsere Mission: Sonne tanken zum Standard machen. Hilf uns und unterstütze evcc finanziell."
thanks = "Vielen Dank, {sponsor}! Dein Beitrag hilft, evcc weiter zu entwickeln."
titleNoSponsor = "Unterstütze uns"
titleSponsor = "Du bist Unterstützer"

[footer.telemetry]
optIn = "Ich möchte meine Ladedaten teilen."
optInMoreDetails = "Mehr Details {0}."
optInMoreDetailsLink = "hier"
optInSponsorship = "Sponsoring erforderlich."

[footer.version]
availableLong = "Neue Version verfügbar"
modalCancel = "Abbrechen"
modalDownload = "Download"
modalInstalledVersion = "Installierte Version"
modalNoReleaseNotes = "Keine Versionshinweise verfügbar. Mehr Informationen zur neuen Version:"
modalTitle = "Neue Version verfügbar"
modalUpdate = "Installieren"
modalUpdateNow = "Jetzt installieren"
modalUpdateStarted = "Starte die neue Version von evcc..."
modalUpdateStatusStart = "Installation gestartet:"

[header]
about = "Über"
blog = "Blog"
docs = "Dokumentation"
github = "GitHub"
login = "Fahrzeug Logins"
[... 194 lines cut ...]
km = "Km"
label = "Einheit"
mi = "Meilen"

[startupError]
configFile = "Verwendete Konfigurationsdatei:"
configuration = "Konfiguration"
description = "Bitte überprüfe deine Konfigurationsdatei. Sollte dir die Fehlermeldung nicht weiterhelfen, suche in unseren {0} nach einer Lösung."
discussions = "GitHub Diskussionen"
fixAndRestart = "Behebe das Problem und starte den Server neu."
hint = "Hinweis: Ein weiterer Grund könnte ein fehlerhaftes Gerät (Wechselrichter, Zähler, ...) sein. Überprüfe deine Netzwerkverbindungen."
lineError = "In {0} wurde ein Fehler gefunden."
lineErrorLink = "Zeile {0}"
restartButton = "Neu starten"
title = "Fehler beim Starten"

[vehicleSettings]
cancel = "Abbrechen"
template = "Hersteller"
test = "Test"

```
und telemetry:
`{"result":true}
`

VolkerK62, 2023-02-15:
```
{,…}
result
: 
[{Template: "aiways", Products: [{Brand: "Aiways", Description: ""}], Params: [,…], Lang: ""},…]
0
: 
{Template: "aiways", Products: [{Brand: "Aiways", Description: ""}], Params: [,…], Lang: ""}
1
: 
{Template: "etron", Products: [{Brand: "Audi", Description: "e-tron"}], Params: [,…], Lang: ""}
2
: 
{Template: "audi", Products: [{Brand: "Audi", Description: "Andere"}], Params: [,…], Lang: ""}
3
: 
{Template: "bmw", Products: [{Brand: "BMW", Description: ""}], Params: [,…], Lang: ""}
4
: 
{Template: "carwings", Products: [{Brand: "Nissan", Description: "Leaf (pre 2019)"}], Params: [,…],…}
5
: 
{Template: "citroen", Products: [{Brand: "Citroën", Description: ""}], Params: [,…], Lang: ""}
6
: 
{Template: "dacia", Products: [{Brand: "Dacia", Description: ""}], Params: [,…], Lang: ""}
7
: 
{Template: "ds", Products: [{Brand: "DS", Description: ""}], Params: [,…], Lang: ""}
8
: 
{Template: "evnotify", Group: "generic", Products: [{Brand: "", Description: "evNotify"}],…}
9
: 
{Template: "fiat", Products: [{Brand: "Fiat", Description: ""}], Params: [,…], Lang: ""}
10
: 
{Template: "ford", Products: [{Brand: "Ford", Description: ""}], Params: [,…], Lang: ""}
11
: 
{Template: "hyundai", Products: [{Brand: "Hyundai", Description: "Bluelink"}], Params: [,…], Lang: ""}
12
: 
{Template: "jaguar-landrover",…}
13
: 
{Template: "kia", Products: [{Brand: "Kia", Description: "Bluelink"}], Params: [,…], Lang: ""}
14
: 
{Template: "mini", Products: [{Brand: "Mini", Description: ""}], Params: [,…], Lang: ""}
15
: 
{Template: "nissan", Products: [{Brand: "Nissan", Description: "Leaf"}], Params: [,…], Lang: ""}
16
: 
{Template: "niu-e-scooter", Group: "scooter", Products: [{Brand: "NIU", Description: "E-Scooter"}],…}
17
: 
{Template: "offline", Group: "generic", Products: [{Brand: "", Description: "Offline"}], Params: [,…],…}
18
: 
{Template: "opel", Products: [{Brand: "Opel", Description: ""}], Params: [,…], Lang: ""}
19
: 
{Template: "ovms", Group: "generic",…}
20
: 
{Template: "peugeot", Products: [{Brand: "Peugeot", Description: ""}], Params: [,…], Lang: ""}
21
: 
{Template: "porsche", Products: [{Brand: "Porsche", Description: ""}], Params: [,…], Lang: ""}
22
: 
{Template: "renault", Products: [{Brand: "Renault", Description: ""}], Params: [,…], Lang: ""}
23
: 
{Template: "cupra", Products: [{Brand: "Seat", Description: "Cupra Born, My Cupra"}], Params: [,…],…}
24
: 
{Template: "seat", Products: [{Brand: "Seat", Description: "Andere"}], Params: [,…], Lang: ""}
25
: 
{Template: "silence", Group: "scooter", Products: [{Brand: "Silence", Description: "S01/S02"}],…}
26
: 
{Template: "enyaq", Products: [{Brand: "Skoda", Description: "Enyaq"}], Params: [,…], Lang: ""}
27
: 
{Template: "skoda", Products: [{Brand: "Skoda", Description: "Andere"}], Params: [,…], Lang: ""}
28
: 
{Template: "smart", Products: [{Brand: "Smart", Description: "EQ"}], Params: [,…], Lang: ""}
29
: 
{Template: "tesla", Products: [{Brand: "Tesla", Description: ""}], Params: [,…], Lang: ""}
30
: 
{Template: "tronity", Group: "generic", Products: [{Brand: "", Description: "Tronity"}], Params: [,…],…}
31
: 
{Template: "volvo-connected", Products: [{Brand: "Volvo", Description: ""}], Params: [,…], Lang: ""}
32
: 
{Template: "volvo", Products: [{Brand: "Volvo (legacy)", Description: ""}], Params: [,…], Lang: ""}
33
: 
{Template: "id", Products: [{Brand: "Volkswagen", Description: "We Connect ID"}], Params: [,…],…}
34
: 
{Template: "vw", Products: [{Brand: "Volkswagen", Description: "We Connect (legacy)"}], Params: [,…],…}
```

VolkerK62, 2023-02-15:
Die mit `Group: "generic"` tauchen in der angezeigten Liste nicht auf. Das sind die unsichbaren.

VolkerK62, 2023-02-15:
In dem Zusammenhang ist mir noch etwas aufgefallen. Sobald man "Fahrzeug hinzufügen" anklickt, wird im UI auf "Gastfahrzeug" gewechselt und das bleibt auch, wenn ich das Auswahlfenster über "X" schließe.

andig, 2023-02-15:
Ich mache mal zu, das werden wir mit https://github.com/evcc-io/evcc/pull/6199 reparieren wo wir je Product statt je Template vorgehen.

PR #6199 (2023-08-17): Add support for loading and storing device configuration from/to database
Refs https://github.com/evcc-io/evcc/issues/6029

This PR adds database storage for devices. Any device from the database is appended to the respective `configProvider`.

- [x] make devices ordered (order of config file, then order by database id)
- [x] add GET `config/devices/<class>` api getting devices
- [x] add PUT `config/devices/<class>/<id>` api for creating updating device (https://stackoverflow.com/questions/28459418/use-of-put-vs-patch-methods-in-rest-api-real-life-scenarios)
- [x] add POST `config/devices/<cla [... cut]
closed 2023-02-15 by andig