evcc-io/evcc #6250: Current Production is wrong

r-rogge, 2023-02-15:
### Describe the bug

At the moment 1862 Watt will be generated by the pv. Most of the power is going to the battery system of the pv (1630 Watt)

Production will be only shown by 232 Watt.
[image]

[image]

### Steps to reproduce

1. it running instantly
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: xxxxx

meters:
- type: template
  template: huawei-dongle 
  id: 1  
  host: 192.168.xxx.xxx  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- name: grid
  type: template
  template: huawei-dongle-powersensor
  usage: grid      
  # Modbus TCP
  modbus: tcpip
  id: 1
  host: 192.168.xxx.xxx
  port: 502 # Port 
- name: battery
  type: template
  template: huawei-dongle-powersensor
  usage: battery      
  
  # Modbus TCP
  modbus: tcpip
  id: 1
  host: 192.168.xxx.xxx
  port: 502 # Port 

chargers:
- type: template
  template: easee 
  user: xxxxxxxx  
  password: xxxxxx
  charger: xxxxxx 
  name: Carport
- type: template
  template: easee
  user: xxxxxx
  password: xxxxxxx
  charger: xxxxxxx
  name: Stellplatz

loadpoints:
- title: Carport
  charger: Carport
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
- title: Stellplatz
  charger: Stellplatz
  mode: now 
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Mein Zuhause
  meters:
    grid: grid 
    pvs: pv1
    batteries: battery
  residualPower: 100

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: xxxxx # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: xxxxx # [currency]/kWh

vehicles:
  - name: xxxxxxx
    type: template
    template: id
    title: xxxxxx # Wird in der Benutzeroberfläche angezeigt # Optional
    user: xxxxxxx
    password: xxxxxxxxxxxxxx
    vin: Xxxxxxxx
    capacity: 52 # Akku-Kapazität in kWh # Optional
```

### Log details

```text
evcc -log debug
Error: unknown command "debug" for "evcc"
Run 'evcc --help' for usage.
unknown command "debug" for "evcc"
pi@bueppelpi:~ $ evcc --log debug 
[main  ] INFO 2023/02/15 10:09:40 evcc 0.112.5
[main  ] INFO 2023/02/15 10:09:40 using config file: /etc/evcc.yaml
[main  ] INFO 2023/02/15 10:09:40 starting ui and api at :7070
[db    ] INFO 2023/02/15 10:09:40 using sqlite database: /home/pi/.evcc/evcc.db
[main  ] ERROR 2023/02/15 10:09:45 creating vehicle FRI-JR6E failed: cannot create vehicle 'template': cannot create vehicle 'id': terms of service updated- please open app or website and confirm
[lp-1  ] WARN 2023/02/15 10:09:45 locking phase config to 3p for switchable charger
[lp-2  ] WARN 2023/02/15 10:09:45 locking phase config to 3p for switchable charger
[site  ] INFO 2023/02/15 10:09:45 site config:
[site  ] INFO 2023/02/15 10:09:45   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/02/15 10:09:45     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/02/15 10:09:45     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/02/15 10:09:45     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/02/15 10:09:45   vehicles:
[site  ] INFO 2023/02/15 10:09:45     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/02/15 10:09:45 loadpoint 1:
[lp-1  ] INFO 2023/02/15 10:09:45   mode:        pv
[lp-1  ] INFO 2023/02/15 10:09:45   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/02/15 10:09:45   meters:      charge ✓
[lp-1  ] INFO 2023/02/15 10:09:45     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2023/02/15 10:09:45 loadpoint 2:
[lp-2  ] INFO 2023/02/15 10:09:45   mode:        now
[lp-2  ] INFO 2023/02/15 10:09:45   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-2  ] INFO 2023/02/15 10:09:45   meters:      charge ✓
[lp-2  ] INFO 2023/02/15 10:09:45     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/02/15 10:09:45 phase timer inactive
[lp-1  ] DEBUG 2023/02/15 10:09:45 pv timer inactive
[lp-2  ] DEBUG 2023/02/15 10:09:45 phase timer inactive
[lp-2  ] DEBUG 2023/02/15 10:09:45 pv timer inactive
[site  ] DEBUG 2023/02/15 10:09:45 ----
[lp-1  ] DEBUG 2023/02/15 10:09:45 charge power: 0W
[lp-2  ] DEBUG 2023/02/15 10:09:45 charge power: 0W
[server] FATAL 2023/02/15 10:09:45 unlinkat /tmp/evcc: operation not permitted
[main  ] FATAL 2023/02/15 10:09:45 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.112.5

andig, 2023-02-15:
Looks like the `huawei-dongle` template should not use the AC but the DC side?

premultiply, 2023-02-16:
Or use the `huawei-dongle-powersensor` template?

closed 2023-02-16 by premultiply

r-rogge, 2023-02-17:
This template was used!

andig, 2023-02-17:
Re-check pv