evcc-io/evcc #6263: Fehler nach Update auf aktuelle unstable - /usr/bin/evcc: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /usr/bin/evcc)

princemaxwell, 2023-02-15:
### Describe the bug

Nach dem Update der EVCC unstable startet EVCC nicht mehr, Fehler siehe Log Details.

**Letzte funktionierende unstable ist: 0.112.5+1676170038**

### Steps to reproduce

sudo apt --only-upgrade install -y evcc

im unstable repo

### Configuration details

```yaml
irrelevant für diesen Fehler
```

### Log details

```text
Feb 15 19:14:23 evcc systemd[1]: evcc.service: Succeeded.
Feb 15 19:14:23 evcc systemd[1]: Stopped evcc.
Feb 15 19:14:23 evcc systemd[1]: evcc.service: Consumed 4min 29.075s CPU time.
Feb 15 19:15:08 evcc systemd[1]: Started evcc.
Feb 15 19:15:08 evcc evcc[2364]: /usr/bin/evcc: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /usr/bin/evcc)
Feb 15 19:15:08 evcc evcc[2364]: /usr/bin/evcc: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /usr/bin/evcc)
Feb 15 19:15:08 evcc systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Feb 15 19:15:08 evcc systemd[1]: evcc.service: Failed with result 'exit-code'.
Feb 15 19:15:18 evcc systemd[1]: evcc.service: Scheduled restart job, restart counter is at 1.
Feb 15 19:15:19 evcc systemd[1]: Stopped evcc.
Feb 15 19:15:19 evcc systemd[1]: Started evcc.
Feb 15 19:15:19 evcc evcc[2367]: /usr/bin/evcc: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /usr/bin/evcc)
Feb 15 19:15:19 evcc evcc[2367]: /usr/bin/evcc: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /usr/bin/evcc)
Feb 15 19:15:19 evcc systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Feb 15 19:15:19 evcc systemd[1]: evcc.service: Failed with result 'exit-code'.
Feb 15 19:15:29 evcc systemd[1]: evcc.service: Scheduled restart job, restart counter is at 2.
Feb 15 19:15:29 evcc systemd[1]: Stopped evcc.
Feb 15 19:15:29 evcc systemd[1]: Started evcc.
Feb 15 19:15:29 evcc evcc[2370]: /usr/bin/evcc: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /usr/bin/evcc)
Feb 15 19:15:29 evcc evcc[2370]: /usr/bin/evcc: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /usr/bin/evcc)
Feb 15 19:15:29 evcc systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Feb 15 19:15:29 evcc systemd[1]: evcc.service: Failed with result 'exit-code'.
Feb 15 19:15:39 evcc systemd[1]: evcc.service: Scheduled restart job, restart counter is at 3.
```

### What type of operating system are you running?

Linux

### Version

Fehlermeldung, Version ist 0.112.5+1676429402

andig, 2023-02-15:
Probiers bitte mal in 20min mit neuen Nightly. Danke für die Fehlermeldung!

closed 2023-02-15 by andig

princemaxwell, 2023-02-15:
@andig 
Funktioniert perfekt! Danke für's schnelle fixen.