evcc-io/evcc #6264: verzögerter Start wegen Warten auf 3p1p Umschaltung in UI anzeigen

kai-morich, 2023-02-15:
**Kontext:** 
Bei meiner Easee-Wallbox kommt es zu bei ansteigender PV-Leistung zu gelegentlichen Umschaltungen von 16A@1p auf 16A@3p statt 6A@3p. Um das einzugrenzen habe ich einen kleinen Python-Server geschrieben, welcher alternative Wert für pv + grid meter liefert, um damit verschiedene Situationen zu simulieren und 1p3p Umschaltungen zu erzwingen. Das Problem konnte ich noch nicht eingrenzen, aber ein anderes:

**Problem:**
Vorbedingung: es wird nicht geladen, die Wallbox ist im 3p Modus von früherer Ladung.
Situation: Ladung könnte starten, da genug Überschuss
Symptom: Es passiert laut UI einige Minuten nichts, obwohl genug Überschuß für 1p Ladung vorhanden ist
Vorschlag: im UI anzeigen, das auf Umschaltung gewartet wird 

[image]

RichardIstSauer, 2023-02-15:
Gleiches erlebe ich auch bei meiner Easee Box. Eine Anzeige des Timers auf der UI würde sehr viel Sinn machen, da man dann auch versteht warum gerade nicht geladen wird.

andig, 2023-02-16:
Die Easee ist leider ein anhaltender Quell von Ärger: https://github.com/evcc-io/evcc/discussions/4413. Der Featurerequest ist ansonsten absolut nachvollziehbar, hat aber keine hohe Prio.

andig, 2023-02-16:
@naltatis die Timer selbst haben wir ja schon im GUI. Mir ist nicht ganz klar, was hier fehlt. Evtl. nur dass bei Ladefreigabe (aber noch keiner Ladung) auch angezeigt werden soll? Also als "Warten auf Phasenumschaltung"?

andig, 2023-02-24:
@kai-morich also konkret sollten die

    phaseRemaining=<duration> 

und

    phaseAction=scale1p/3p

angezeigt werden.

@naltatis ich denke das Problem liegt in der VehicleStatus.vue:

		phaseTimerActive() {
			return (
				this.charging &&
				this.phaseRemainingInterpolated > 0 &&
				["scale1p", "scale3p"].includes(this.phaseAction)
			);
		},

wir sollten das wohl auch bei Ladefreigabe anzeigen, nicht nur bei `charging`.

closed 2023-02-24 by naltatis

kai-morich, 2023-02-25:
👍 verified