evcc-io/evcc #6278: Messaging

thebrainkafka, 2023-02-17:
### Discussed in https://github.com/evcc-io/evcc/discussions/6277

<div type='discussions-op-text'>

<sup>Originally posted by **thebrainkafka** February 17, 2023</sup>
Hallo zusammen.

Mit folgendem:
`Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration} at  ${vehicleSoc:%.0f}%`

Erhalte ich diesen output:
`Finished charging 0.2kWh in 3m20s at vehicleSoc 89%!(NOVERB)`

Was bedeutet?
!(NOVERB)

Was muss ich anpassen?

Danke
Jogi</div>

Yaml.conf  line 227:
msg: Battery charged to ${vehicleSoc:%.0f}%

Müsste zu geändert werden
msg: Battery charged to ${vehicleSoc}%

closed 2023-02-17 by andig

bene86, 2023-02-17:
Würde ich auch gerne wissen. In meiner Config steht:
`    soc: # vehicle soc update event`
`      title: SoC updated`
`      msg: Battery charged to ${vehicleSoc:%.0f}%`

Die Mail ist dann wie folgt:
`Battery charged to 34%!(NOVERB)`

andig, 2023-02-18:
Ich denke % muss %% sein. Frag mich nur, warum das bei mir geht…

thebrainkafka, 2023-02-18:
Oder ${vehicleSoc}% ?

thebrainkafka, 2023-02-18:
Also gerade getestet.
Mit %% am Ende funktioniert es.