evcc-io/evcc #6283: Dynamischer Phasenwechsel mit Keba KC S10 und EVCC

eldrik, 2023-02-17:
### Discussed in https://github.com/evcc-io/evcc/discussions/5930

<div type='discussions-op-text'>

<sup>Originally posted by **ReLo99** January 28, 2023</sup>
Hallo,
ich habe aktuell eine Keba P30 c-series installiert und steuere die Keba sehr stabil über EVCC.
Gerne möchte ich das PV-Laden optimieren und bei geringer PV-Leistung 1-phasig laden.
Hierfür bietet Keba den Phasenumschalter KeContact S10 an, der mit der P30 zusammenarbeitet.
Hat jemand Erfahrung oder Infos, ob der Phasenumschalter KeContact S10 ebenfalls in das EVCC integriert (inkl. Delay-Paratmeter etc.)  werden kann.
VG ReLo99</div>

Kein direkter Issue sondern eher ein FeatureRequest.

Seit Firmware Version 3.10.42 unterstützt die P30 die externe Ansteuerung eines Phasenumschalters (Keba S10 oder Nachbau) über das X2 Interface der WB.

Es können nunmehr entsprechende Modbus TCP oder UDP Parameter gesetzt werden für die Definition der Source für den Umschaltbefehl sowie das umschalten von 1 auf 3Phasenbetrieb. 

Könnte dies in die evcc Keba Definition mit aufgenommen werden?

UDP
https://www.keba.com/download/x/4a925c4c61/kecontactp30udp_pgen.pdf

x2src für die Quelle des Phasenwechselkommandos
und
x2 für den Phasenwechsel

Modbus TCP
https://www.keba.com/download/x/dea7ae6b84/kecontactp30modbustcp_pgen.pdf

Gesetzt werden muss

Parameter 5050 und 5052

Greetz
Eldrik

andig, 2023-02-17:
Mir ist nicht klar, wie sich rausfinden liesse, ob das S10 eigentlich verbaut ist? Und kann die x2src auch über eine Keba SW gesetzt werden? Dann wäre es ggf eine Idee danach zu schauen ob die passt.

eldrik, 2023-02-17:
könnte man hierfür nicht eine Option in der evcc Config unterbringen? S10 verbaut ja/nein?
Aus dem p30 Fw changelog
[image]

andig, 2023-02-18:
Schöner wäre es ohne config. Gibts Zugriff auf ein Testgerät? Ohne wir sichs nicht umsetzen lassen.

eldrik, 2023-02-21:
> Schöner wäre es ohne config. Gibts Zugriff auf ein Testgerät? Ohne wir sichs nicht umsetzen lassen.

Testgerät = Keba P30c oder den S10 Phasenumschalter?

andig, 2023-02-21:
Umschalter- die KEBA geht ja schon ;)

eldrik, 2023-02-21:
ah ok, aber wofür genau? 
Der Umschalter ist ja nichts weiter als ein Schütz bei dem L2 und L3 durchgeschaltet werden je nachdem ob eine permanent über L1, mit einem Netzteil erzeugte 24V DC Hilfsspannung, durch den X2 Kontakt der KEBA geleitet, am Schütz anliegt.

andig, 2023-02-21:
Anders gesagt: es braucht eine KEBA mit Umschalter!

eldrik, 2023-02-21:
@ReLo99 da du scheinbar Interesse an dem Feature hattest sowie über ein einsatzbereites E-Auto und eine KEBA verfügst, kannst du hier unterstützen?

ReLo99, 2023-02-24:
Aktuell habe ich die keba P30c vom Elektriker auf einphasig "hart" umstellen lassen. Rein theoretisch kann mit 32 A eine Ladestrom von 1,4 kw bis 7,4 kw erreicht werden. Erlaubt sind aber nur 20 A also max. 4,6 kw (Vermeidung einer Netzüberlastung auf einer Phase)
Dennoch kann ich mit der Umstellung auf einphasiges Laden deutlich mehr PV-Strom nutzen, weil das PV-Überschussladen bereits bei 1,4 kw, anstatt mit 4,2 kw bei drei phasigen Laden startet. Die Ladezeit spielt bei mir keine Rolle, sondern die Optimierung des PV-Überschussladens. Im Sommer habe ich, egal wie ich das Laden optimiere, eine, PV-Überschuss, den ich einspeisen muss, d.h. im Sommer kann ich entweder mit mehr Leistung kürzer laden oder mit weniger Leistung länger laden, in beiden Fällen werde ich das Auto i.d.R. mit 100% PV-Strom laden, weil die Tage sonnig und lang sind :-) , egal ob 1-phasig oder 3-phasig.
Wenn allerdings die Ladezeit optimiert werden soll, ist eine Phasenumschaltung im Sommer sinnvoll.
Bei einer dynamischen Phasenumschaltung muss wohl eine Zwangspause von ca. 2 Minuten eingelegt werden, in der der Ladevorgang zum Auto unterbrochen und nach der Pause wieder aufgebaut wird. Ansonsten besteht wohl die Gefahr, dass die Autoladesteuerung bei einem adhoc Phasenwechsel schaden nimmt. Das ist bei einer Kombi Keba S10/P30 bereits berücksichtigt. 
Fazit: Die S10 erscheint mir als eine sehr belastbare Lösung, ist mir aber zu teuer (ca. 250Euro ohne Installation). Dass muss ich erstmal wieder reinholen. Eine reine softwaregestützte Umschaltung erscheint mir risikobehaftet, weil die Zwangspause nicht zu 100% sichergestellt werden kann.
Ich habe mich aus diesen Gründen für einen einphasigen Betrieb der P30c entschieden.

andig, 2023-02-24:
Dann machen wir hier zu. Ohne Testgerät ists nicht umsetzbar.

eldrik, 2023-02-27:
@andig nur für mich zum besseren Verständnis, wofür wird das Testgerät genau benötigt? Kannst du einmal beschreiben, welche Informationen, ohne das vorhanden sein des Gerätes nicht ermittelt werden können?

premultiply, 2023-02-27:
Wir müssen schon mal in der Realität ausprobieren können wie sich das wirklich verhält.

eldrik, 2023-02-27:
ah ok nachvollziehbar und das Verhalten einer Phasenumschaltung "händisch" nachzustellen (digitales Umschaltsignal an die Keba und manuelles umschalten der Phasen, über einen Handschalter/Sicherungsautomaten) ist daher vermutlich auch keine Option? 

Ich kann @ReLo99 verstehen, dass er sich die Box für 270€ nicht anschaffen möchte (zumal sie derzeit auch nicht sofort verfügbar ist).

Ich werde mir die Box nachbauen und die Komponenten dann direkt in die UV setzen wenn ich in den Genuss eines E-Autos komme.
Bis dahin für die interessierten Nachbauer, die Keba S10 Box enthält der Verkaufsgrafik und dem Installationshandbuch nach ein: 

Finder Installationsschütz vom Typ Finder 22.72.0.230.1310
Ein paar 3 und 2 Fach Wago Durchgangsklemmen vom Typ WAGO 2206-1301 2206-1304  2006-1307 2201-1201
Einen B16 Sicherungsautomaten für das 24V Netzteil und ein 
24V Netzteil vom Typ Phoenix Contact STEP3-PS - PHOENIX 1088495 

Alles in allem Materialkosten (ohne Gehäuse) von unter 80€

Bernd2211, 2023-02-27:
Ich habe eine Keba P30 x-Series und werde mir den S10 bestellen, wenn frei am Markt erhältlich. Helfe dann gerne beim Testen.

stuppeliene, 2023-03-14:
Danke für die Aufstellung der Teile für einen möglichen Keba S10 - Nachbau.
Müsste das Schütz nicht die 24V-Spulenversion sein ? Also Finder 22.72.0.024.1370 ?

eldrik, 2023-03-14:
Völlig richtig, Spulenspannung sind 24V habe es korrigiert.

In der S10 sollte verbaut sein ein 22.72.0.024.1310 (32A zwei Schließer mit 24V Spulenspannung)

https://cdn.findernet.com/app/uploads/S22DE.pdf

stuppeliene, 2023-03-15:
Habe mir die Teile nun bestellt. Ab Mitte/Ende April kann ich voraussichtlich eine UV- Installation bereitstellen. Keba P30 x-series und Skoda Enyaq sind vorhanden.

JoergBec, 2023-03-18:
Hallo zusammen.
 Gerade erst gesehen dass es schon ein Thema dazu gibt. Vielleicht hätte ich dann auch die Bauteile einzeln bestellt. Naja sei drum.
 Seit heute habe ich nun die S10 installiert mit einer C Series. Ich könnte nun Änderungen testen
Gerade lädt sie einphasig mit Max 3,5kw obwohl mehr PV Überschuss da ist 

Gruß Jörg

andig, 2023-03-18:
> Dann machen wir hier zu. Ohne Testgerät ists nicht umsetzbar. Gerne hier melden wenn sich daran etwas ändert.

Wir bräuchten dafür weiterhin Zugang.

JoergBec, 2023-03-18:
Ok. Wann und wie? Habe es bisher über Home Assistant laufen…

andig, 2023-03-20:
Bitte Portweiterleitung (oder anderen Weg) per Mail an info@evcc.io

stuppeliene, 2023-03-22:
Bekomme meine Teile leider erst Ende April. Meine Wallbox wäre über VPN  von aussen erreichbar. Die Zugangsdaten und die lokale IP benötigt ihr für den Remotezugriff, richtig?

andig, 2023-03-22:
> Mir ist nicht klar, wie sich rausfinden liesse, ob das S10 eigentlich verbaut ist? Und kann die x2src auch über eine Keba SW gesetzt werden?

Weiss das jemand?

stuppeliene, 2023-03-22:
[image]

stuppeliene, 2023-03-22:
Über z.B. Modbus 1550 kann man den Inhalt von 5050 auslesen (  zumindestens indirekter Zustand, dass S10 verbaut ist)

stuppeliene, 2023-03-22:
Ansonsten kann man überprüfen (nach Umschaltkomando und entsprechender Wartezeit) ob sich die Phasenanzahl 1/3 geändert  hat.

softservicegit, 2023-04-08:
> Finder 22.72.0.024.1310

Hi, hast du eine Qualle für das Schütz gefunden, die auch liefern kann? Ich finde nur die üblichen Verdächtigen, die im Juli oder später was haben wollen.  Was dann wahrscheinlich auch nicht stimmt.

Ich würde mir gerne den Aufpreis der Teile zum fertigen S10 sparen. ...

Andreas

eldrik, 2023-04-08:
Wenn es nur um das Schütz geht, da wird jedes mit den gleichen Eigenschaften funktionieren, ich vermute aber die schlechte Verfügbarkeit des Schützes ist auch ein Grund warum die generelle Verfügbarkeit der Box derzeit nicht so berauschend ist.

stuppeliene, 2023-04-08:
Ich habe das Schütz vor etwa 3 Wochen bei Voelkner bestellt. Soll laut Bestellübersicht am 11.4. geliefert werden

Stephan

softservicegit, 2023-04-08:
> Ich habe das Schütz vor etwa 3 Wochen bei Voelkner bestellt. Soll laut Bestellübersicht am 11.4. geliefert werden

Aktuell sagen die 19.7. ach ja ...

krk-elektro, 2023-04-12:
> Ich habe eine Keba P30 x-Series und werde mir den S10 bestellen, wenn frei am Markt erhältlich. Helfe dann gerne beim Testen.

Habe bei Kleinanzeigen solch eine Box inseriert, und habe genügend Material auf Lager um noch ein paar davon zu „bauen“.

mucki12, 2023-04-12:
> Habe bei Kleinanzeigen solch eine Box inseriert, und habe genügend Material auf Lager um noch ein paar davon zu „bauen“.
> 
> Bekomme ggfs. kurzfristig auch eine Keba P30 DE440, dann kann ich auch ein Test-Setting bereitstellen. Allerdings kann mein Auto nur 1phasig laden.

Ist das die Anzeige, die die Eigenlösung für 209 Euro anbietet (wurde hier nicht von 80 Euro Materialkosten gesprochen?) und wo bereits auf eine Unterstützung durch evcc hingewiesen wird?
Also muss @andig gar nicht mehr testen und es läuft so oder so mit evcc? Das sind ja tolle Neuigkeiten!

krk-elektro, 2023-04-13:
> Ist das die Anzeige, die die Eigenlösung für 209 Euro anbietet (wurde hier nicht von 80 Euro Materialkosten gesprochen?) und wo bereits auf eine Unterstützung durch evcc hingewiesen wird? Also muss @andig gar nicht mehr testen und es läuft so oder so mit evcc? Das sind ja tolle Neuigkeiten!

Wenn Deine Zeit nichts kostet, kannst Du solch eine Umschaltbox ja gern für 80€ anbieten ;) 
Ist ja auch nur für diejenigen gedacht, die sowas nicht mal eben schnell selber zusammenstellen.
Durch die nun verfügbare Hardware wird die Implementierung in evcc vermutlich keine Ewigkeit mehr drauern.

@andig Können wir den Test mit einem Auto machen, welches nur 1p laden kann ?
Ansonsten frage ich meinen Nachbarn ob er seinen Tesla mal gratis laden möchte.
Zugriff per Anydesk oder Portfreigabe kann ich einrichten.
VG Benjamin

andig, 2023-04-13:
Umschalten kannst Du ja immer.

> Wenn ich es richtig verstanden habe, kann man den S10 Parameter (einmalig) entweder über Modbus (5050) oder udp (x2src) einstellen. Dieser Parameter kann auch im Einstellmenü der KebaApp (oder über die webui) eingestellt werden.

Hast Du das S10 jetzt schon eingerichtet? Ist irgendwo auslesbar, ob das vorhanden und eingerichtet ist?

> Zugriff per Anydesk oder Portfreigabe kann ich einrichten.

Da fehlt mir leider die Idee. Es scheint, als würden wir UDP nicht über Portfreigaben transportiert bekommen. Wenn das nicht gelingt wäre die Alternative, dass ich KEBA komplett auf Modbus umschreiben müsste.

mucki12, 2023-04-13:
> Wenn Deine Zeit nichts kostet, kannst Du solch eine Umschaltbox ja gern für 80€ anbieten ;)

Nein nein. Ist ja ok wenn die fertige Lösung funktioniert. Wollte eigentlich insbesondere wissen, welche Unterschiede zwischen den hier diskutierten Einzelkomponenten, deiner Fertiglösung und der KEBA S10 bestehen.

Irritiert hatte mich eigentlich nur, dass in der Anzeige ja klar auf eine Funktionalität mit evcc hingewiesen wird. Ich hatte diese Diskussion hier eher so verstanden, dass hier noch Tests nötig wären. Da ich gerade auf der Wallbox Suche bin, ist die Bestätigung für die Nutzung mit evcc natürlich schon mal von Bedeutung.

krk-elektro, 2023-04-14:
> Umschalten kannst Du ja immer.
>
>
> Hast Du das S10 jetzt schon eingerichtet? Ist irgendwo auslesbar, ob das vorhanden und eingerichtet ist?
>
>
> Da fehlt mir leider die Idee. Es scheint, als würden wir [...]

S10 baue ich heute ein, meine Keba WB kommt Montag.
Melde mich bei Dir sobald ich die WB angeschlossen habe.
Kann Dir dann auch einen VPN Tunnel bereitstellen, dann bist Du zum testen quasi bei mir Lokal im LAN.

andig, 2023-04-15:
Ist bekannt, ob alle KEBAs ModbusTCP beherrschen oder sind wir von UDP abhängig?

mucki12, 2023-04-15:
Nach meiner Recherche sollten sie Modbus TCP beherrschen.
Zitat: KeContact P30 c-series with firmware version 3.10.16 or higher, or KeContact P30 x-series with software version 1.11 or higher

Doku:
https://www.keba.com/download/x/dea7ae6b84/kecontactp30modbustcp_pgen.pdf

andig, 2023-04-15:
PR ist bereit zum Test. X2 Source muss auf UDP (=4) stehen.

mucki12, 2023-04-15:
Erst einmal vielen Dank für die schnelle Umsetzung zum Test. Leider kann ich wie schon geschrieben nicht wirklich weiterhelfen. Aber grundsätzlich lässt mich diese Entwicklung doch ein wenig hoffen und eine jetzige Bestellung einer Keba P30x wäre doch nicht die schlechteste Wahl, oder?
Sorry für das OT, aber ich bin wirklich immer noch auf der Suche nach einer verlässlichen Lösung. Und @krk-elektro falls das zunächst falsch rüberkam - ich wäre damit vielleicht auch in der Zukunft an einer technischen Alternative zur Phasenumschaltung interessiert...

tBounce7, 2023-04-16:
Hallo zusammen, kurze Info: bei Vattenfall gibts aktuell ne x-Serie für 599€..
Ein Bekannter von mir nutzt die Keba Dienstwagen Version und die Software von clever PV. Bei ihm geht das phasenwechseln ohne die s-10. habt ihr ne Idee warum?

andig, 2023-04-16:
Könnte jemand eine Keba mit ModbusTCP für info@evcc.io per Portweiterleitung o.ä. verfügbarmachen? Egal ob mit Phasenumschaltung, oder ohne.

mucki12, 2023-04-16:
Habe die Keba gestern bestellt. Wenn sich ansonsten niemand meldet kann ich das nach der Lieferung und Installation gerne machen - kann aber noch nicht genau sagen wann das sein wird.

JosefRick, 2023-04-16:
Ich hab ne Keba verfügbar und mach sie dir verfügbar.
Die Software ist schon auf 3.10.42, allerdings fehlt für die tatsächliche Phasenumschaltung noch das entsprechende Schütz. Da weiß ich nicht wann ich das endlich bekomme. Könnte die Umschaltung also nur testen indem ich kontrolliere ob der entsprechende Kontakt in der Wallbox geschalten wird der das Schütz ansteuert.

andig, 2023-04-16:
Implementierung ist testbereit, hat aber zwei dicke Macken:

- enable/disable über 5014 zu setzen und über 1000 != 5 auszulesen scheint nicht zu funktionieren
- 3p (5052 == 1) wieder auf 1p zurück zu schalten ist mir nicht gelungen- evtl. wehrt sich da intern die Box mit einem Timer

Jetzt sind wir auf Eure Ideen angewiesen...

premultiply, 2023-04-16:
Hast du vor dem Phasenumschalten gestoppt?
Vielleicht geht das nur in einem bestimmten Zustand?

ufux, 2023-04-17:
DSW1.2 on? (UDP off)

ufux, 2023-04-17:
> PR ist bereit zum Test. X2 Source muss auf UDP (=4) stehen.

Wenn Du per TCP auf die Box zugreifst, kann man das doch auch per modbus register 5050 setzen oder die Box muss per Web Interface als X2src eben TCP (Modbus) gesetzt bekommen haben. Wenn X2src auf UDP steht, kann doch ein Umschalten auf 3p (5052 == 1) nicht funktionieren ... my 2ct...

mucki12, 2023-04-18:
Ich musste leider feststellen, dass ich leider doch nicht mehr als 16A laden könnte. Somit wäre die Keba bei mir wohl doch raus, da dieses bei einphasig lediglich 3,7KW wären. Die Entwicklung mit der Phasenumschaltung hört sich hier zwar sehr spannend an, aber vielleicht müsste ich mich wohl doch auf die Go e konzentrieren und die Keba zurückschicken.

mucki12, 2023-04-18:
> Ein Bekannter von mir nutzt die Keba Dienstwagen Version und die Software von clever PV. Bei ihm geht das phasenwechseln ohne die s-10. habt ihr ne Idee warum?

Könntest du deinen Bekannten noch mal fragen ob die wirklich einen Phasenwechsel ohne S10 oder anderweitiger eigener Komponenten macht? Kann ich irgendwie gar nicht so recht glauben…

tBounce7, 2023-04-18:
> Könntest du deinen Bekannten noch mal fragen ob die wirklich einen Phasenwechsel ohne S10 oder anderweitiger eigener Komponenten macht? Kann ich irgendwie gar nicht so recht glauben…

Da haste recht, er hat sich geirrt

tBounce7, 2023-04-18:
> Ich musste leider feststellen, dass ich leider doch nicht mehr als 16A laden könnte. Somit wäre die Keba bei mir wohl doch raus, da dieses bei einphasig lediglich 3,7KW wären. Die Entwicklung mit [...]

hab zwar gegoogelt, aber zur Sicherheit: gibts von Go e ne geeichte Version? Würdest du nur wegen einphasig laden bevorzugen, bei gleichen Schnittstellen?

mucki12, 2023-04-18:
> hab zwar gegoogelt, aber zur Sicherheit: gibts von Go e ne geeichte Version? Würdest du nur wegen einphasig laden bevorzugen, bei gleichen Schnittstellen?

Glaube nicht, aber ich benötige auch keinen geeichten Zähler. Mein Interesse liegt nur an der maximalen Nutzung von PV Strom für das Laden und da ich für ein Leistung >16A doch einige Leitungen tauschen müsste, bin ich auf eine Phasenumschaltung angewiesen. Könnte ja mit der Keba eventuell auch bald funktionieren, dann aber natürlich zu einem höheren Preis.

tBounce7, 2023-04-18:
Den Vattenfall Deal zur x-Serie kennste oder? Vielleicht wäre das was für dich

mucki12, 2023-04-18:
Ja, die P30x kommt auch heute bei mir an :-) 
In Verbindung mit den Kosten für einen S10 und einen Leitungstausch vom Garagenverteiler zur bzw. zu den Wallboxen haben mich aber noch einmal ins Grübeln gebracht. Die Go e kostet 550, wird an die Wand gehängt und würde anschließend die Anforderung wohl auch erfüllen - zumal da die Phasenumschaltung ja eh schon funktioniert.

motdule, 2023-04-18:
Ich klinke mich hier mal mit ein, da ich auch versuche mit einer Keba P30 X die Phasenumschaltung umzusetzen. Ich besitze die Keba erst seit einigen Tagen und habe bisher noch wenig Erfahrung mit der Steuerung der Keba. Eine S10 Phasenumschaltung aus Einzelteilen habe ich mir aber auch gebaut und ist installiert. Bisher bin ich gescheitert das PV Laden über die Keba eigene Regelung mit einem Fronius Smart Meter TS65A umzusetzen. Die Keba startet zwar kurz das Laden, stoppt aber nach 1 Minute und pausiert die Ladung. Ich bin noch nicht dahintergestiegen und aus den Logs noch nicht 100% schlau geworden warum die Ladung pausiert wird. Ich vermute aber das die Keba die Werte vom Fronius Smartmeter über den Symo nicht einwandfrei lesen kann und deshalb die Ladung nach gewisser Zeit unterbricht. Die Phasenumschaltung funktioniert aber zumindest, da zu Beginn der Ladung entweder einphasig oder dreiphasig begonnen wird. Wie ich das ausgelöst habe weiß ich nicht mehr genau, evtl über die harten A Werte Min / Max Grenzem im Einstellungsmenü Ladeverbund oder über eine Testweise anbindung an Openwb. Testfahrzeug ist ein ID4 der mit der OpenWB Series 2 die Phasenumschaltung einwandfrei macht. Ich bin gerne bereit beim Testing mitzuhelfen.

andig, 2023-04-21:
> Implementierung ist testbereit, hat aber zwei dicke Macken:
>
> * enable/disable über 5014 zu setzen und über 1000 != 5 auszulesen scheint nicht zu funktionieren
> * 3p (5052 == 1) wieder auf 1p zurück [...]

Falls es keine Ideen gibt können wir hier zu machen. Mir fällt dazu nix mehr ein :/

mucki12, 2023-04-21:
Sorry nur interessehalber (schicke sie wohl zurück), aber läuft die zukünftige Integration der KEBA dann trotzdem über Modbus, oder bleibt es bei udp?

stuppeliene, 2023-04-21:
Ich schicke mal einen Link aus dem Loxone-Forum. Vielleicht ist das hilfreich. Ab Beitrag 268 ff. wird es m. E. nach interessant.
Offensichtlich macht die Keba die ganze Umschaltung inkl. der notwendigen Umschaltverzögerungen selbstständig, nur abhängig von der Leistungs-(Strom) Sollwertvorgabe. Das würde ja auch passen weil zum Kebasystem ja normalerweise ein extra Energiezähler (E10) zur Erfaasung des PV-Überschusses gehört. Wahrscheinlich muss diese Schnittstelle emuliert werden...

https://www.loxforum.com/forum/hardware-zubeh%C3%B6r-sensorik/77402-keba-wallbox/page18

JosefRick, 2023-04-22:
> @JosefRick An der Wallbox DIP SW1.2 = on? (UDP off)

Sorry das ich erst so spät Antworte, aber ich bin derzeit unterwegs und hab nicht nachschauen können. Hab mir aber jetzt den DIP SW Stand durchgeben lassen. Er war auf 1.2 off und 1.3 on, so wie es in der Keba Anleitung zur Phasenumschaltung steht. Ich probiers aber gern nochmal aus wenn ich wieder daheim bin und versuch mich an einer Lösung.

[59 more replies]