evcc-io/evcc #6300: fehlendes Ladelog und zurückgesetzte Auswertung Ladeenergie

StevieC121176, 2023-02-18:
### Describe the bug

seit der Version `evcc version 0.112.5 (2080452a)` fehlt bei mir das komplette Ladelog und die Auswertung der Ladeenergie ist auf 0 gesetzt.

```
[main  ] INFO 2023/02/18 12:31:39 evcc 0.112.5 (2080452a)
[main  ] INFO 2023/02/18 12:31:39 using config file: /etc/evcc.yaml
[db    ] INFO 2023/02/18 12:31:40 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2023/02/18 12:31:40 connecting evcc-609958330 at tcp://192.168.178.77:1883
```

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.112.5 (2080452a)

andig, 2023-02-18:
> using sqlite database: /home/pi/.evcc/evcc.db

Ich vermute das ist die falsche DB. Hast Du noch mehr davon? Verwendet ein ältere Version eine andere DB?

StevieC121176, 2023-02-18:
Nein ich habe seit Anfang an nur diese eine DB.
Installiert ist EVCC auf einem raspberryPi. gestartet wird über den Systemdienst
`sudo systemctl enable evcc ` 
oder bei updates über `sudo systemctl stop evcc` bzw. `sudo systemctl start evcc`  

tritt wie gesagt erst seit heute morgen nach dem update auf die nightly  version 0.112.5 (https://github.com/evcc-io/evcc/commit/2080452ae43ff7b8f4f4291128dbfff1219266dd)

andig, 2023-02-18:
Ich vermute wenn Du in die DB rein schaust gibts keine Ladevorgänge?

pauxus, 2023-02-18:
@StevieC121176 : Was für ein Linux hast du denn? 

Zeig mal den Inhalt von `/etc/systemd/system/multi-user.target.wants/evcc.service` falls vorhanden... Ich wüsste nicht, warum er auf einmal den User Pi nehmen sollte

Adhoc würde ich mal vermuten, dass das nicht mit dem Deb Paket installiert wurde, sondern manuell...

StevieC121176, 2023-02-18:
unter /home/pi/ gibt es bei mir gar kein Verzeichnis /.evcc
und auf das /root Verzeichnis komme ich nicht drauf

StevieC121176, 2023-02-18:
@pauxus 

Betriebssystem ist raspberrypi OS lite 32bit. ohne Desktop. basierend auf Debian bullseye

Hier der Inhalt der Datei:

```
# evcc.service
#

[Unit]
Description=evcc
Requires=network-online.target
After=syslog.target network.target network-online.target
Wants=network-online.target
StartLimitIntervalSec=10
StartLimitBurst=10

[Service]
ExecStart=/usr/bin/evcc
Environment="EVCC_DATABASE_DSN=/var/lib/evcc/evcc.db"
Restart=always
RestartSec=10

User=evcc
Group=evcc

[Install]
WantedBy=multi-user.target 
```

andig, 2023-02-18:
> unter /home/pi/ gibt es bei mir gar kein Verzeichnis /.evcc

    ls -la /home/pi

VolkerK62, 2023-02-18:
Es kommt darauf an, wie man fragt.
Wenn ich `evcc meter `abfrage, bekomme ich als Antwort `using sqlite database: /home/pi/.evcc/evcc.db`.
Frage ich `sudo evcc meter` ab, bekomme ich als Antort `using sqlite database: /root/.evcc/evcc.db`

StevieC121176, 2023-02-18:
oh sorry, versteckte Verzeichnisse waren ausgeblendet, hier der Inhalt. Wurde am 28.01.23 das letzte mal geändert

```
SQLite format 3   @                                                                     .c
¯ b c«b                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                Å~ÉOtablesessionssessionsCREATE TABLE `sessions` (`id` integer,`created` datetime,`finished` datetime,`loadpoint` text,`identifier` text,`vehicle` text,`odometer` real,`meter_start_kwh` real,`meter_end_kwh` real,`charged_kwh` real,PRIMARY KEY (`id`))bÅtablesettingssettingsCREATE TABLE `settings` (`key` text,`value` text,PRIMARY KEY (`key`))/C indexsqlite_autoindex_settings_1settings       
```
SQLite format 3   @                                                                     .c
¯ b c«b                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                               [... cut ...]                                                                                       Å~ÉOtablesessionssessionsCREATE TABLE `sessions` (`id` integer,`created` datetime,`finished` datetime,`loadpoint` text,`identifier` text,`vehicle` text,`odometer` real,`meter_start_kwh` real,`meter_end_kwh` real,`charged_kwh` real,PRIMARY KEY (`id`))bÅtablesettingssettingsCREATE TABLE `settings` (`key` text,`value` text,PRIMARY KEY (`key`))/C indexsqlite_autoindex_settings_1settings       

pauxus, 2023-02-18:
Sicher, dass du den evcc zuletzt wirklich über systemctl gestartet hast? das sieht aus, als hättest du ihn einfach mit "evcc" gestartet, dann würde er die db im Userverzeichnis des angemeldeten Users suchen.

pauxus, 2023-02-18:
@VolkerK62 Klar, das nimmt immer das Verzeichnis des aktuellen Nutzers, aber in dem Fall wird die DB ja garnicht genutzt.

@andig Vielleicht sollte man die Angabe der Datenbank für diese Kommando aus der Ausgabe rausnehmen, das verwirrt eher, als dass es hilft.

pauxus, 2023-02-18:
Die Logausgabe im Original Pos sieht mir auch nicht aus wie eine "journalctl" Ausgabe, da müssten von Journalctl noch die Zeitstempel und Host und User davor stehen.

andig, 2023-02-18:
> Vielleicht sollte man die Angabe der Datenbank für diese Kommando aus der Ausgabe rausnehmen, das verwirrt eher, als dass es hilft.

Wieso? Das ist genau die auf die wir zugreifen. Welche Verwirrung?