evcc-io/evcc #6317: Not copying /var/lib/evcc/.evcc/evcc.db to /var/lib/evcc/evcc.db, since there is already a database there

andig, 2023-02-19:
Fehler beim Einspielen des nightly:

Changelogs werden gelesen... Fertig
(Lese Datenbank ... 120991 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../evcc_0.112.5+1676818169_armhf.deb ...
usermod: Keine Änderungen
Not copying /var/lib/evcc/.evcc/evcc.db to /var/lib/evcc/evcc.db, since there is already a database there
Either delete one of the databases or run 'touch /var/lib/evcc/.evcc/.copiedToEvccUser' to keep both.
then restart installation.
dpkg: Fehler beim Bearbeiten des Archivs /var/cache/apt/archives/evcc_0.112.5+1676818169_armhf.deb (--unpack):
 »neues evcc-Skript des Paketes pre-installation«-Unterprozess gab den Fehlerwert 1 zurück
Fehler traten auf beim Bearbeiten von:
 /var/cache/apt/archives/evcc_0.112.5+1676818169_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

_Originally posted by @hillwig in https://github.com/evcc-io/evcc/discussions/6305#discussioncomment-5025496_

andig, 2023-02-19:
@pauxus könntest Du da nochmal rein schauen?

pauxus, 2023-02-19:
Das ist eigentlich works as designed. Die dB wurde schonmal kopiert, deshalb gibt es jetzt zwei, aber keinen Marker. Zur Sicherheit wird dann abgebrochen. Feststellen, welches die richtige ist (Dateigröße) und dann die andere löschen oder umbenennen. 
Ist vermutlich durch das hin und her mit der Umgebungsvariable passiert. 

Reicht die Fehlermeldung nicht?

andig, 2023-02-19:
D.h. beim "normalen" Upgrade passiert das nicht? Dann wär das sicher gut genug.

pauxus, 2023-02-19:
Genau, das ist nur ein safeguard, damit nicht versehentlich Daten überschrieben werden. Normalerweise kopiert er genau einmal vom alten User zur neuen Datenbank Location und markiert dann mit einem Markerfile bei der alten, dass bereits kopiert wurde. 

Wenn an beiden Stellen eine Datenbank liegt, aber die alte als schor kopiert markiert ist, macht er gar nichts. Aber wenn am Zielort schon eine Datenbank liegt, ohne das die alte markiert wurde, lasse ich zur Sicherheit abbrechen. 

Sollte eigentlich die Ausnahme sein, insbesondere bei regulären releases sollte hier kein Problem auftreten. 

Wobei es ganz interessant zu wissen wäre, wie genau das hier dazu gekommen ist.

pauxus, 2023-02-19:
Ok, auch habe es nochmal nachgestellt. Das Problem ist wie vermutet der Fehler mit der Umgebungsvariable.

Jetzt gibt es eine DB unter /var/lib/evcc und wegen des Fehler unter /var/lib/evcc/.evcc und eine alte unter /root/, die korrekt als kopiert markiert wurde. Die neue, unnötige und leere kann halt nicht mehr kopiert werde, weil eben richtige kopierte schon da ist. Die gute Nachricht: die unter /var/lib/evcc/.evcc kann ohne Probleme gelöscht werden.

pauxus, 2023-02-19:
Es reicht ein `sudo rm /var/lib/evcc/.evcc/evcc.db`, dann die Installation neu starten.

andig, 2023-02-19:
Top, vielen Dank! Ich lass das Issue mal sichtbar falls es doch vermehrt auftreten sollte.

StevieC121176, 2023-02-19:
Ja super vielen Dank, hat funktioniert. Alles wieder vorhanden.

hillwig, 2023-02-19:
Helft mir bitte nochmal kurz... Befehl für eine Neuinstallation ist...?

closed 2023-02-21 by andig