evcc-io/evcc #6326: Shelly bleibt aus; mincurrent: 0.07 = max charge current: 0A

VolkerK62, 2023-02-20:
Bug oder Feature? 
Bei `max charge current: 0A` wird der Shelly wohl nicht eingeschaltet.
Mit `mincurrent: 1` funktioniert es.
 

Ich vermute, dass ist der Grund
>lp-1  ] DEBUG 2023/02/20 12:07:48 max charge current: 0A

und das der Übeltäter 
>  mincurrent: 0.07		# 0.07A * 220 V = 15,4 W

Änder das mal auf `mincurrent: 1`

_Originally posted by @VolkerK62 in https://github.com/evcc-io/evcc/discussions/6323#discussioncomment-5052673_

andig, 2023-02-20:
Und was ist hier der vermeintliche Fehler?

VolkerK62, 2023-02-20:
Das bei dem sehr kleinen `mincurrent` der Shelly nicht eingeschaltet wird. Erhöht man `mincurrent` auf `1`, dann funktioniert das Einschalten.
Meine Vermutung ist, dass das mit `max charge current: 0A` zu tun hat.

```
[site  ] DEBUG 2023/02/20 12:07:36 ----
[lp-1  ] DEBUG 2023/02/20 12:07:37 charge power: 0W
[lp-2  ] DEBUG 2023/02/20 12:07:37 charge power: 0W
[site  ] DEBUG 2023/02/20 12:07:38 grid power: -39W
[site  ] DEBUG 2023/02/20 12:07:38 pv power: 39W
[site  ] DEBUG 2023/02/20 12:07:38 site power: -39W
[lp-1  ] DEBUG 2023/02/20 12:07:38 charger status: B
[lp-1  ] DEBUG 2023/02/20 12:07:38 pv charge current: 0.17A = 0A + 0.17A (-39W @ 1p)
[lp-1  ] DEBUG 2023/02/20 12:07:38 site power -39W <= -20W enable threshold
[lp-1  ] DEBUG 2023/02/20 12:07:38 pv enable timer start: 10s
[lp-1  ] DEBUG 2023/02/20 12:07:38 pv enable in 10s
[site  ] DEBUG 2023/02/20 12:07:41 ----
[lp-1  ] DEBUG 2023/02/20 12:07:42 charge power: 0W
[lp-2  ] DEBUG 2023/02/20 12:07:42 charge power: 0W
[site  ] DEBUG 2023/02/20 12:07:42 grid power: -55W
[site  ] DEBUG 2023/02/20 12:07:42 pv power: 55W
[site  ] DEBUG 2023/02/20 12:07:42 site power: -55W
[lp-2  ] DEBUG 2023/02/20 12:07:42 charger status: B
[site  ] DEBUG 2023/02/20 12:07:46 ----
[lp-1  ] DEBUG 2023/02/20 12:07:47 charge power: 0W
[lp-2  ] DEBUG 2023/02/20 12:07:47 charge power: 0W
[site  ] DEBUG 2023/02/20 12:07:47 grid power: -54W
[site  ] DEBUG 2023/02/20 12:07:47 pv power: 54W
[site  ] DEBUG 2023/02/20 12:07:47 site power: -54W
[lp-1  ] DEBUG 2023/02/20 12:07:48 charger status: B
[lp-1  ] DEBUG 2023/02/20 12:07:48 pv charge current: 0.235A = 0A + 0.235A (-54W @ 1p)
[lp-1  ] DEBUG 2023/02/20 12:07:48 site power -54W <= -20W enable threshold
[lp-1  ] DEBUG 2023/02/20 12:07:48 pv enable in 0s
[lp-1  ] DEBUG 2023/02/20 12:07:48 pv enable timer elapsed
[lp-1  ] DEBUG 2023/02/20 12:07:48 max charge current: 0A
[site  ] DEBUG 2023/02/20 12:07:51 ----
[lp-1  ] DEBUG 2023/02/20 12:07:52 charge power: 0W
[lp-2  ] DEBUG 2023/02/20 12:07:52 charge power: 0W
[site  ] DEBUG 2023/02/20 12:07:52 grid power: -87W
[site  ] DEBUG 2023/02/20 12:07:52 pv power: 87W
[site  ] DEBUG 2023/02/20 12:07:52 site power: -87W
[lp-2  ] DEBUG 2023/02/20 12:07:52 charger status: B
[site  ] DEBUG 2023/02/20 12:07:56 ----
[lp-1  ] DEBUG 2023/02/20 12:07:57 charge power: 0W
[lp-2  ] DEBUG 2023/02/20 12:07:57 charge power: 0W
[site  ] DEBUG 2023/02/20 12:07:57 grid power: -113W
[site  ] DEBUG 2023/02/20 12:07:57 pv power: 113W
[site  ] DEBUG 2023/02/20 12:07:57 site power: -113W
[lp-1  ] DEBUG 2023/02/20 12:07:58 charger status: B
[lp-1  ] DEBUG 2023/02/20 12:07:58 pv charge current: 0.491A = 0A + 0.491A (-113W @ 1p)
[lp-1  ] DEBUG 2023/02/20 12:07:58 site power -113W <= -20W enable threshold
[lp-1  ] DEBUG 2023/02/20 12:07:58 pv enable in 0s
[lp-1  ] DEBUG 2023/02/20 12:07:58 pv enable timer elapsed
[lp-1  ] DEBUG 2023/02/20 12:07:58 max charge current: 0A
[site  ] DEBUG 2023/02/20 12:08:01 ----
[lp-1  ] DEBUG 2023/02/20 12:08:02 charge power: 0W
[lp-2  ] DEBUG 2023/02/20 12:08:02 charge power: 0W
[site  ] DEBUG 2023/02/20 12:08:03 grid power: -175W
[site  ] DEBUG 2023/02/20 12:08:03 pv power: 175W
[site  ] DEBUG 2023/02/20 12:08:03 site power: -175W
[lp-2  ] DEBUG 2023/02/20 12:08:03 charger status: B
[site  ] DEBUG 2023/02/20 12:08:06 ----
[lp-1  ] DEBUG 2023/02/20 12:08:07 charge power: 0W
[lp-2  ] DEBUG 2023/02/20 12:08:07 charge power: 0W
[site  ] DEBUG 2023/02/20 12:08:08 grid power: -175W
[site  ] DEBUG 2023/02/20 12:08:08 pv power: 175W
[site  ] DEBUG 2023/02/20 12:08:08 site power: -175W
[lp-1  ] DEBUG 2023/02/20 12:08:09 charger status: B
[lp-1  ] DEBUG 2023/02/20 12:08:09 pv charge current: 0.761A = 0A + 0.761A (-175W @ 1p)
[lp-1  ] DEBUG 2023/02/20 12:08:09 site power -175W <= -20W enable threshold
[lp-1  ] DEBUG 2023/02/20 12:08:09 pv enable in 0s
[lp-1  ] DEBUG 2023/02/20 12:08:09 pv enable timer elapsed
[lp-1  ] DEBUG 2023/02/20 12:08:09 max charge current: 0A
[site  ] DEBUG 2023/02/20 12:08:11 ----
```

thierolm, 2023-02-20:
@VolkerK62 , m.E. ist die Config von @GhostPV falsch (siehe meinen Kommentar in #6323  ) ... bevor ihr weiter forscht sollte das korrigiert werden ...

Auch ist das m.E. kein Shelly spezifisches Problem!

VolkerK62, 2023-02-20:
@thierolm Hab deinen Kommentar gelesen. 
Mal gespannt, ob das der Grund für das Nicht-Einschalten war. 
Und weil ich wissbegierig bin, musst du mir das dann erklären ;-)

andig, 2023-02-21:
Scheint erstmal kein Issue zu sein. Wenns ein konkretes gibt gerne melden.

closed 2023-02-21 by andig