evcc-io/evcc #6327: MQTT - multiple connections with same clientid (if configured)

klein0r, 2023-02-20:
### Describe the bug

I've configured mqtt with a custom client id (config see below)

This leads to errors in my mosquitto log (see below).

It looks like evcc creates multiple connections with the same (configured) clientid (instead of using just 1 connection)?!

### Steps to reproduce

1. Configure mqtt
2. Set a custom client id
3. Use mqtt plugin at multiple locations

### Configuration details

```yaml
mqtt:
  broker: 172.16.0.5:1883
  topic: evcc
  clientid: evcc
  user: evcc
  password: 1234567890

meters:
  - name: pv
    type: custom
    power: # power (W)
      source: mqtt
      topic: home/energy/electricity/photovoltaic/generator/power
    energy: # optional energy (kWh)
      source: mqtt
      topic: home/energy/electricity/photovoltaic/generator/energy
      scale: 0.001
    currents:
      - source: mqtt
        topic: home/energy/electricity/photovoltaic/current/l1
      - source: mqtt
        topic: home/energy/electricity/photovoltaic/current/l2
      - source: mqtt
        topic: home/energy/electricity/photovoltaic/current/l3
  - name: charge
    type: custom
    power: # power (W)
      source: mqtt
      topic: home/energy/electricity/wallbox/power
    energy: # optional energy (kWh)
      source: mqtt
      topic: home/energy/electricity/wallbox/energy
      scale: 0.001
    currents:
      - source: mqtt
        topic: home/energy/electricity/wallbox/current/l1
      - source: mqtt
        topic: home/energy/electricity/wallbox/current/l2
      - source: mqtt
        topic: home/energy/electricity/wallbox/current/l3
  - name: grid
    type: custom
    power: # power (W)
      source: mqtt
      topic: home/energy/electricity/meter/power
      scale: -1
    energy: # optional energy (kWh)
      source: mqtt
      topic: home/energy/electricity/photovoltaic/generator/energy
      scale: 0.001
    currents:
      - source: mqtt
        topic: home/energy/electricity/meter/current/l1
      - source: mqtt
        topic: home/energy/electricity/meter/current/l2
      - source: mqtt
        topic: home/energy/electricity/meter/current/l3
```

### Log details

```text
mqtt broker / mosquitto log:

1676896287: New client connected from 172.16.0.251:33898 as evcc (p2, c1, k30, u'evcc').
1676896287: New connection from 172.16.0.53:54988 on port 1883.
1676896287: Client evcc already connected, closing old connection.
1676896287: New client connected from 172.16.0.53:54988 as evcc (p2, c1, k30, u'evcc').
1676896287: New connection from 172.16.0.251:33899 on port 1883.
1676896287: Client evcc already connected, closing old connection.
1676896287: New client connected from 172.16.0.251:33899 as evcc (p2, c1, k30, u'evcc').
1676896287: New connection from 172.16.0.53:54994 on port 1883.
1676896287: Client evcc already connected, closing old connection.
1676896287: New client connected from 172.16.0.53:54994 as evcc (p2, c1, k30, u'evcc').
1676896287: New connection from 172.16.0.251:33900 on port 1883.
1676896287: Client evcc already connected, closing old connection.
1676896287: New client connected from 172.16.0.251:33900 as evcc (p2, c1, k30, u'evcc').
1676896287: New connection from 172.16.0.53:54996 on port 1883.
1676896287: Client evcc already connected, closing old connection.
1676896287: New client connected from 172.16.0.53:54996 as evcc (p2, c1, k30, u'evcc').
1676896288: New connection from 172.16.0.251:33901 on port 1883.
1676896288: Client evcc already connected, closing old connection.
1676896288: New client connected from 172.16.0.251:33901 as evcc (p2, c1, k30, u'evcc').
1676896288: New connection from 172.16.0.53:55012 on port 1883.
1676896288: Client evcc already connected, closing old connection.
1676896288: New client connected from 172.16.0.53:55012 as evcc (p2, c1, k30, u'evcc').
....
```
```

### What type of operating system are you running?

Linux

### Version

0.112.5

andig, 2023-02-20:
Thats strange. It should reuse the connection. Closing since evcc logfile is missing.

klein0r, 2023-02-20:
```
[main  ] INFO 2023/02/20 13:19:45 evcc 0.112.5
[main  ] INFO 2023/02/20 13:19:45 using config file: /etc/evcc.yaml
[main  ] INFO 2023/02/20 13:19:45 starting ui and api at :7070
[db    ] INFO 2023/02/20 13:19:45 using sqlite database: /home/mkleine/.evcc/evcc.db
[mqtt  ] INFO 2023/02/20 13:19:45 connecting evcc at tcp://172.16.0.5:1883
[mqtt  ] ERROR 2023/02/20 13:19:45 tcp://172.16.0.5:1883 connection lost: read tcp 172.16.0.53:55244->172.16.0.5:1883: read: connection reset by peer
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: not currently connected and ResumeSubs not set
[mqtt  ] ERROR 2023/02/20 13:19:45 error: not currently connected and ResumeSubs not set
[mqtt  ] ERROR 2023/02/20 13:19:45 error: not currently connected and ResumeSubs not set
[mqtt  ] ERROR 2023/02/20 13:19:45 error: not currently connected and ResumeSubs not set
[mqtt  ] ERROR 2023/02/20 13:19:45 error: not currently connected and ResumeSubs not set
[mqtt  ] ERROR 2023/02/20 13:19:45 error: not currently connected and ResumeSubs not set
[mqtt  ] ERROR 2023/02/20 13:19:45 error: not currently connected and ResumeSubs not set
[mqtt  ] ERROR 2023/02/20 13:19:45 error: not currently connected and ResumeSubs not set
[mqtt  ] ERROR 2023/02/20 13:19:45 error: not currently connected and ResumeSubs not set
[mqtt  ] ERROR 2023/02/20 13:19:45 error: not currently connected and ResumeSubs not set
[mqtt  ] ERROR 2023/02/20 13:19:45 error: not currently connected and ResumeSubs not set
[mqtt  ] ERROR 2023/02/20 13:19:45 error: not currently connected and ResumeSubs not set
[mqtt  ] ERROR 2023/02/20 13:19:45 tcp://172.16.0.5:1883 connection lost: read tcp 172.16.0.53:55258->172.16.0.5:1883: read: connection reset by peer
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 tcp://172.16.0.5:1883 connection lost: read tcp 172.16.0.53:55270->172.16.0.5:1883: read: connection reset by peer
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 tcp://172.16.0.5:1883 connection lost: read tcp 172.16.0.53:55272->172.16.0.5:1883: read: connection reset by peer
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[... cut ...]
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: connection lost before Subscribe completed
[mqtt  ] ERROR 2023/02/20 13:19:45 error: not currently connected and ResumeSubs not set
[mqtt  ] ERROR 2023/02/20 13:19:45 error: not currently connected and ResumeSubs not set
...
```

andig, 2023-02-20:
Passiert das auch wenn Du die `clientid` weg lässt?

klein0r, 2023-02-20:
Nein, ohne die clientId ist alles gut

andig, 2023-02-20:
Sehr gut, dann hab ich ein Testszenario (und Du einen workaround ;).

klein0r, 2023-02-20:
Oh man - ich bin blind... Es lief ein Docker-Container zum Test (längst vergessen, dass es das System gibt) mit evcc. Wenn man das Log lesen würde, dann sieht man auch zwei unterschiedliche IPs, welche sich um die Client-ID streiten...

> 1676896287: New client connected from 172.16.0.251:33898 as evcc (p2, c1, k30, u'evcc').
> 1676896287: New connection from 172.16.0.53:54988 on port 1883.
> 1676896287: Client evcc already connected, [...]

Ich gehe dann mal lieber ins Bett 😞

closed 2023-02-20 by klein0r