evcc-io/evcc #6335: "Invalid phase wiring" bzgl. Phasenströmen ignorieren

premultiply, 2023-02-21:
Nachdem nun offensichtlich verstärkt Fahrzeuge auf dem Markt sind die im 3p-Modus beliebige Phasenkonfigurationen (z.T. auch dynamisch) nutzen sollten wir überlegen diese Warnung runterzustufen (Debug?) oder ganz auszubauen.

Somit im Endeffekt:
Strom nur auf L1: 1p-Modus
alle anderen Kombinationen: 3p-Modus

Hi there, I just noticed that I too get a lot of "invalid phase wiring" messages when charging a Kia Niro EV using my SonnenCharger.
Reading through the discussions I checked that the phases are connected properly(ref #6108, #3065). What I found, however, is that the car seems to use 2 phases only - and phases 2 and 3 at that - when charging when the load is low:

[image]

Since it seems to be working fine I'm not too concerned. Just wanted to let you know and it didn't seem important enough to open an issue ;)

_Originally posted by @RustyDust in https://github.com/evcc-io/evcc/discussions/6330_

andig, 2023-02-21:
@DerAndereAndi hatte das für VW auch schonmal in einer Discussion. Sollten wir ausbauen!

@premultiply würdest Du das wieder in die Hand nehmen?

DerAndereAndi, 2023-02-21:
Das war beim Taycan, wenn er >90% SoC erreicht hatte und dann nur noch auf Phase 2 geladen hatte.

andig, 2023-02-21:
Bleibt die Frage, ob wir das i.S. der PV-Ladung als 1p oder 3p werten sollten? Mein Vorschlag wäre, dass wir hier _nur_ die Meldung deaktivieren, die sonstige Logik aber unangetastet lassen. Das heisst im Klartext aber auch, dass wir keine Fehlerzustände mehr erkennen können da Ladung auf jeder beliebigen Phase zulässig wäre.

premultiply, 2023-02-21:
Das ist halt die Frage.
Z.B. bzgl. Lastmanagement ist das perspektivisch durchaus ein Problem wenn 1p erkannt würde aber auf L2 oder L3 geladen wird. Ich würde daher eher dazu tendieren mittelfristig die Logik wie in der GUI zu machen.

Aber erstmal nur die Warnung raus.
Oder auf Debug-Level setzen?

andig, 2023-02-21:
Raus- das ist ja ein ganz normaler Anwendungsfall. Das Lastmanagement muss m.E. ohnehin auf die echten Phasenströme achten, aber das können wir dann dort diskutieren.

premultiply, 2023-02-21:
Umgesetzt mit f5abb20781c8443511818cc00a29bf67027d1d43

closed 2023-02-21 by premultiply