evcc-io/evcc #6354: Warn/Error: kleiner Makel im UI

VolkerK62, 2023-02-22:
Ich habe ein "offline" vehicle. Da kommt beim Start von evcc der Hinweis `vehicle offline does not support automatic detection`.
Dieser wird jetzt auch einmalig als Msg verschickt. Soweit alles okay.

Die Warnung erscheint auch im UI. Wenn man die Meldung entfernt, bleibt sie auch entfernt, solange das Browserfenster mit dem UI im Vordergrund bleibt. Wechselt man z.B. auf einen anderen Browser-Tab oder minimiert den Browser, dann ist die Warnung im UI wieder da.

andig, 2023-02-22:
Ja, das liegt an https://github.com/evcc-io/evcc/commit/ff7ae369678d55679e3869bc2e03dfee0ea9921a#diff-ab967ab1a2f3a1b769106eeb7bfe892ef0e81d1d27811fa15be08e6749feee1fL138, da lass ich mir was einfallen dass die Meldung

VolkerK62, 2023-02-22:
Bei Push kommt die Meldung nur einmal. Einmal im UI wäre ja auch nicht tragisch (aber das stelle ich mir wahrscheinlich zu einfach vor)

VolkerK62, 2023-02-22:
Ja, ok. Hab ich heute gemerkt. 
Mein use case hat sich auch dadurch erledigt, dass DaheimLaden Modbus ab morgen implementiert ist.

andig, 2023-02-22:
Dann machen wir erstmal zu. Problem verstanden, aber grad keine Lösungsidee…

closed 2023-02-24 by andig