evcc-io/evcc #6362: Probleme mit Vehicle SoC am Ioniq 5 (v 0.112.5)

frog60, 2023-02-22:
### Describe the bug

Habe heute evcc installiert.

Der SOC des IONIQ kam jetzt eher zufällig einmal sporadisch, aber nicht zuverlässig.

[image]

"pv 2 power" könnte vom WR kommen oder doch vom IONIQ? 

Bekomme folgende zwei Fehler (Integration mit RCT-Power + GO-eCharger):

[site  ] ERROR 2023/02/22 14:07:07 **pv 2 power: unexpected status: 400**
[site  ] DEBUG 2023/02/22 14:07:09 pv power: 3658W
[site  ] DEBUG 2023/02/22 14:07:09 battery soc: 82%
[site  ] DEBUG 2023/02/22 14:07:09 battery power: -281W
[site  ] DEBUG 2023/02/22 14:07:09 grid power: -1W
[site  ] DEBUG 2023/02/22 14:07:09 site power: -282W
[lp-1  ] DEBUG 2023/02/22 14:07:10 charge currents: [11.5 0 0]A
[lp-1  ] DEBUG 2023/02/22 14:07:10 detected active phases: 1p
[lp-1  ] DEBUG 2023/02/22 14:07:10 charger status: C
[lp-1  ] DEBUG 2023/02/22 14:07:10 next soc poll remaining time: 59m49s
[lp-1  ] ERROR 2023/02/22 14:07:10 **vehicle soc: unexpected status: 408**

### Steps to reproduce

1. keine SOC Anzeige auf dem evcc-Screen
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: IONIQ_5
    type: template
    template: hyundai
    title: "IONIQ 5" # Wird in der Benutzeroberfläche angezeigt # Optional
    user: xxx
    password: xxx
    #vin: # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 72 # Akkukapazität in kWh # Optional
    language: en # 'de' für Deutsch und 'en' für Englisch # Optional
```

### Log details

```text
[site  ] ERROR 2023/02/22 14:07:07 pv 2 power: unexpected status: 400
[site  ] DEBUG 2023/02/22 14:07:09 pv power: 3658W
[site  ] DEBUG 2023/02/22 14:07:09 battery soc: 82%
[site  ] DEBUG 2023/02/22 14:07:09 battery power: -281W
[site  ] DEBUG 2023/02/22 14:07:09 grid power: -1W
[site  ] DEBUG 2023/02/22 14:07:09 site power: -282W
[lp-1  ] DEBUG 2023/02/22 14:07:10 charge currents: [11.5 0 0]A
[lp-1  ] DEBUG 2023/02/22 14:07:10 detected active phases: 1p
[lp-1  ] DEBUG 2023/02/22 14:07:10 charger status: C
[lp-1  ] DEBUG 2023/02/22 14:07:10 next soc poll remaining time: 59m49s
[lp-1  ] ERROR 2023/02/22 14:07:10 vehicle soc: unexpected status: 408
```

### What type of operating system are you running?

Docker container

### Version

0.112.5

VolkerK62, 2023-02-22:
Ist keine Fehler bei evcc.
Hatte ich heute und die letzten Tage auch.
Bei mir hat das Auto manchmal schlechten Empfang und dann ist auch Bluelink nicht immer zuverlässig.

panzerdev, 2023-02-22:
Hatte ich auch die letzten Tage Probleme bei meinem.

Spannend ist: der Abruf des gesamten Status funktioniert aber des partiellen nicht. Ist in anderen Implementierungen genauso (bluelinky)

Scheint Server Probleme zu geben

czam70, 2023-02-22:
Ich hänge mich hier mal dran.
Wir haben einen Kia EV6 und einen Ioniq 5. 
Mir ist klar dass die Meldungen "vehicle soc: unexpected status: 408" und bei mir noch "" vehicle soc: unexpected status: 503" an einem schlechten Empfang liegen (bei mir in der Garage ist da in 90% der Fälle cnihts zu wollen) oder eben an der Kia/Hyundai Serverinfrastruktur.

Daher Frage: kann man die Meldungen im EVCC GUI (rotes Dreieck oben rechts) irgendwie unterdrücken? Mir ist ja klar, an was es liegt und wenn mal Empfang ist, geht ja auch alles... Will da deshlab nicht immer diese Meldungen haben...

tobiasbayer, 2023-02-22:
Vielleicht könnte man es weniger aufdringlich machen, indem man nur den SoC anders einfärbt oder so, wenn der letzte erfolgreiche Abruf schon länger zurückliegt?

czam70, 2023-02-22:
Ja, ist vielleicht auch eine Idee. 
Aber auf jeden Fall sollte man die Option haben, alle "vehicle"-bezogenen Meldungen im GUI unterdrücken zu können.
Einschalten bei Ersteinrichtung oder gewünschten Troubleshooting, ausschalten wenn es prinzipel läuft, aber aufgrund einer grundsätzlich als unstabil zu bewertenden Mobilfunkverbindung mit Fehlern zu rechnen ist.

Warum Autos keine WLAN-Clients sein können ist mir übrigens schon lange ein Rätsel... WLAN-Empfang habe ich in der Grage natürlich ;-)

VolkerK62, 2023-02-22:
>Warum Autos keine WLAN-Clients sein können ist mir übrigens schon lange ein Rätsel

Weil trotz 70kWh im Akku, die 12V Batterie den Geist aufgibt ;-)

panzerdev, 2023-02-22:
> > Warum Autos keine WLAN-Clients sein können ist mir übrigens schon lange ein Rätsel
> 
> Weil trotz 70kWh im Akku, die 12V Batterie den Geist aufgibt ;-)

Jedes Auto hat einen DC - DC Wandler der an geht wenn der 12V Akku zu wenig Spannung hat. VW zb. gibt dir die Option das öfter zu machen wenn du mit dem Reichweitenverlusst fein bist. Also technisch kein Problem den 12V Akku nachzuladen.