evcc-io/evcc #6367: UI: add option to enable/disable error msg for vehicle

VolkerK62, 2023-02-22:
Ja, ist vielleicht auch eine Idee. 
Aber auf jeden Fall sollte man die Option haben, alle "vehicle"-bezogenen Meldungen im GUI unterdrücken zu können.
Einschalten bei Ersteinrichtung oder gewünschten Troubleshooting, ausschalten wenn es prinzipel läuft, aber aufgrund einer grundsätzlich als unstabil zu bewertenden Mobilfunkverbindung mit Fehlern zu rechnen ist.

_Originally posted by @czam70 in https://github.com/evcc-io/evcc/issues/6362#issuecomment-1440237816_

andig, 2023-02-22:
Das gibts bereits. Einfach Logging für das Fahrzeug auf "fatal" statt "error" stellen.

closed 2023-02-22 by andig

VolkerK62, 2023-02-22:
noch besser.
Für die Doku: `fatal` ist da nicht aufgeführt. Gibt es noch mehr level? In welcher Datei find ich die?

VolkerK62, 2023-02-22:
Hat sich erledigt. Habs gefunden.