evcc-io/evcc #6370: FritzBox configured as "battery" - scenario "Powerstation"

frog60, 2023-02-22:
**Is your feature request related to a problem? Please describe.**
No, but I have the requirement, that I would like to control a Powerstation (DC Output -> Inverter -> AC Output) by a Smart Plug (FritzDect 200/210). In the night this works as an additional battery (spring/autumn) and extents the house battery. It could be a cheap way extending a so called "Balkonkraftwerk" with a "multi purpose" Powerstation.

**Describe the solution you'd like**
configure "usage" as "battery" - SOC is not available, but at least the "house consumption" can be consolidated/corrected (battery power will be added).

andig, 2023-02-22:
Du kannst das heute schon machen- einfach direkt konfigurieren anstatt die Templates zu nutzen. Vor dem Hintergrund würde ich sagen wontfix. Scenario zu individuell.

closed 2023-02-22 by andig

frog60, 2023-02-23:
> "direkt konfigurieren" @andig : Kannst du mir hier einen Hinweis geben, wie man das dann umsetzt?