evcc-io/evcc #6385: Ergänzung von Klima im generischen Template

menkej, 2023-02-23:
Ergänzung im "manuellen Auto" Klima aktiv/inaktiv zurückgeben zu können. Ggf. auch mit der Möglichkeit das zu steuern.

Aus der Diskussion:
Die Plugins können aktuell nur 1 Wert zurück geben, Klima braucht aber 3. Die Klima-SSt. ist mit 3 Temperaturen (innen/aussen/ziel) ohnehin ziemlich overengineered da die meisten Fahrzeuge das eh nicht liefern können. Wenn ihr fein seid dass wir das auf Klima aktiv/inaktiv vereinfachen dann würde das gehen.äre dann vmtl ebenfalls möglich- da würde ich direkt um PR bitten.

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/5559#discussioncomment-5085667_

premultiply, 2023-02-23:
Bin ich fein mit.

Weiter:
Ggf. könnte man auch überlegen ein "Jetzt Vorklimatisieren"-API als Kommando zum Fahrzeug zu ergänzen.

Anwendungsfall: Vorklimatisiertes Fahrzeug zur Zielzeit (Abfahtszeit).

mdkeil, 2023-02-23:
Es sind zahlreiche Car-Templates vorhanden, wo auch die Klimatisierung integriert ist. Ich denke ein Großteil der Nutzer wird damit happy sein. Da die meisten Hersteller-APIs aber beschränkt sind, was die Anzahl der Abfragen angeht, haben Nutzer die eine Hausautomation (Home Assitant, ioBroker, FHEM, you name it) einsetzen und natürlich auch die Fahrzeugdaten visualisiert haben wollen ein Problem mit der dann doppelten Datenabfrage.. Um die Beschränkung zum umgehen, setze ich z.B. für Stelantis-Fahrzeuge den [PSA-Car-Controller](https://github.com/flobz/psa_car_controller) ein, der quasi als Datenproxy agiert. Ähnliches gibt es auch für [KIA/Hyundai](https://github.com/Hyundai-Kia-Connect/kia_uvo) und bestimmt auch für zahlreiche andere Fahrzeuge. Um am Ende nicht für jeden "CarProxy" ein separates Template zu erstellen ist meiner Ansicht nach der gangbare Weg das generische Car-Template zu erweitern (in diesem Fall um die Klimatisierung on/off) um auch hier ein vollumfängliche Steuerung der WB zu ermöglichen.

andig, 2023-02-23:
…das beantwortet aber auch in keiner Weise die Frage ob alle mit der Vereinfachung einverstanden sind. Genau darum gehts hier denn das wäre die Voraussetzung. Wäre super beim Thema zu bleiben.

mdkeil, 2023-02-23:
sorry.. ich wollte nur einen Hintergrund aufzeigen ;) -- Generell habe ich mir aber die Frage gestellt, in welchem Szenario die Temperaturwerte überhaupt genutzt werden sollten.. denn für die Freigabe der WB reicht ja der Zustand..

MarkusGH, 2023-02-23:
Sehe keine Nachteile das so zu vereinfachen. Wenn das irgendwann die Option ermöglicht das Fahrzeug per evcc UI zu klimatisieren umso besser...

thebrainkafka, 2023-02-23:
Sorry aber was ist denn der IST Zustand?

andig, 2023-02-25:
Scheint als gäbe es keine Einwände.

harrykausl, 2023-02-27:
Toll wäre natürlich, wenn man dann das Vorklimatisieren auch über die api starten könnte.

closed 2023-03-01 by andig