evcc-io/evcc #6399: FritzDect Template Config: unexpected status: 400 for AINs having leading zero

thierolm, 2023-02-23:
### Describe the bug

FritzDect Meter and Chrager templates are not supporting devices having a leading zero in AIN:

The following error is raising: 
```
thierolm@LAPTOP-KQMMVN3H:~/git-repos/evcc$ ./evcc -c ./evcc_fritzdecttest.yaml meter
[main  ] INFO 2023/02/23 21:29:02 evcc 0.112.5 (2080452a)
[main  ] INFO 2023/02/23 21:29:02 using config file: ./evcc_fritzdecttest.yaml
[db    ] INFO 2023/02/23 21:29:02 using sqlite database: /home/thierolm/.evcc/evcc.db
Power:  unexpected status: 400
Energy: unexpected status: 400
```

FritzDect template config leads to the following request:
```
https://fritz.box/webservices/homeautoswitch.lua?ain=%270876XXXXXXX%27&sid=ba50431d28684bbe&switchcmd=getswitchpower
```
==> AIN value is enclosed by %27, which leads to an error 400 ...

The evcc fritzdect 'raw' configuration with ain enclosing single quotes is working, while creating the following correct request:
```
https://fritz.box/webservices/homeautoswitch.lua?ain=0876XXXXXXX&sid=ad3267db69439ee1&switchcmd=getswitchpower
```

### Steps to reproduce

1. Configure a fritzdect meter or charger via the fritzdect template config with an AIN having a leading zero
2. Test meter or charger with `evcc meter` or `evcc charger`

### Configuration details

```yaml
# Template config leading to an error
- name: pv2
  type: template
  usage: pv
  template: fritzdect
  uri: https://fritz.box
  user: xxxx
  password: xxxx
  ain: "087610XXXXXX" # switch actor identification number without blanks (see AIN number on switch sticker)
```

### Log details

```text
thierolm@LAPTOP-KQMMVN3H:~/git-repos/evcc$ ./evcc -c ./evcc_fritzdecttest.yaml meter
[main  ] INFO 2023/02/23 21:29:02 evcc 0.112.5 (2080452a)
[main  ] INFO 2023/02/23 21:29:02 using config file: ./evcc_fritzdecttest.yaml
[db    ] INFO 2023/02/23 21:29:02 using sqlite database: /home/thierolm/.evcc/evcc.db
Power:  unexpected status: 400
Energy: unexpected status: 400
...
[fritzdect] TRACE 2023/02/23 21:29:26 GET https://fritz.box/webservices/homeautoswitch.lua?ain=%270876XXXXXXX%27&sid=ba50431d28684bbe&switchcmd=getswitchpower
```

### What type of operating system are you running?

Linux

### Version

0.112.5

thierolm, 2023-02-23:
Just to add the working 'raw' configuration, which has to be defined with enclosing single quotes to work:
```yaml
- name: pv1
  type: fritzdect
  uri: https://fritz.box
  user: xxxx
  password: xxxxx
  ain: '087610030810' # switch actor identification number without blanks (see AIN number on switch sticker)
```

andig, 2023-02-23:
Gilt das für jede AIN in Quotes oder nur mit führender 0?

thierolm, 2023-02-23:
Ohne führende 0 können bei der 'raw' Config die Quotes weggelassen werden.

andig, 2023-02-23:
Das war aber nixht die Frage ;)

thierolm, 2023-02-23:
Siehe Slack-Chat ...

andig, 2023-02-24:
Behoben: https://github.com/evcc-io/evcc/commit/560ba8973468a8023aad38e32839def212f0829d. Du hattest den Automatismus für korrektes Quoting nochmal gedoppelt...

closed 2023-02-24 by andig

thierolm, 2023-02-26:
Ah, ok! :-)