evcc-io/evcc #6400: Move SDM72D-M meter to own template

premultiply, 2023-02-23:
Der Grund ist eigentlich ganz einfach:
Das Template ist technisch nur für den einphasigen SDM220/230 gedacht (dort ist die Leistung auf Register 13) und funktioniert somit mit dem dreiphasigen SDM72D-M nicht (richtig).
Dafür braucht es ein neues Template.

_Originally posted by @premultiply in https://github.com/evcc-io/evcc/discussions/6051#discussioncomment-5094491_

andig, 2023-02-24:
Wäre die Lösung nicht ganz einfach, den `SDM72D-M` mit im SDM630 Template zu erwähnen?

premultiply, 2023-02-24:
Nein, das geht nur mit dem SDM72DMv2.
Der SDM72D-M ist was ganz anderes.

andig, 2023-02-24:
Es gibt DM und D-M? Was für ein Chaos. Dann sollte der 72D-M in mbmd rein? Ich schau mal...

premultiply, 2023-02-24:
Es gibt
SDM72D-M
und
SDM72DMv2

Ist alles schon in mbmd.
Ich baue nachher nur noch ein extra Template für den SDM72D-M und den aus dem SDM220/230-Template aus.

PR #6418 (2023-02-26): Add more modbus meters
- Moved Eastron SDM72D-M (not V2) to own template (Fixes #6400)
- Added SDM120-Modbus
- Added all other well-known mbmd 3p modbus meters
- Removed incompatible usages (inverted signs and wrong energy counter registers)
- Added phase powers where applicable for grid meter usage
- Added phase voltages where applicable for charg meter usage
- Fixed small naming issues etc.
closed 2023-02-26 by andig