evcc-io/evcc #6407: Zielladen: Ladestart auf Slotanfang setzen

VolkerK62, 2023-02-24:
Letzte Nacht ist mir noch eine kleine Verbesserungsmöglichkeit aufgefallen.
Günstigste Tibberladezeit war heute von 5 bis 6 Uhr. Zielladen war von mir mit 8 Uhr angegeben! Zielsoc stand auf 90%!
evcc hat sich dann, wohl abhängig vom Zielsoc und der Zielladezeit als Ladestartzeit 5:22 Uhr errechnet!
Und das ist nun genau das, was ich nicht verstehe bzw. als Verbesserung vorschlage!

Geladen wurden ca. 6kwh in 57 Minuten, also ca. bis 6:17 Uhr, aber warum? Ab 6 Uhr war der Strompreis bereits etwas teurer als zwischen 5 und 6 Uhr. Warum wurde nicht Punkt 5 Uhr gestartet und so die volle Stunde zum günstigsten Tarif voll ausgenutzt?
Warum erst der Start 22 Minuten nach 5 Uhr?

Also mein Vorschlag wäre beim Tibber (o. ä.) -laden exakt zur günstigen Zeit zu starten, um das Maximum der günstigen Ladestunde auch so zu nutzen!

Keine Ahnung, ob das so umsetzbar wäre, macht aber m. M. nach mehr Sinn.

_Originally posted by @steve0564 in https://github.com/evcc-io/evcc/discussions/6072_

andig, 2023-02-24:
Aktuell wontfix bis wir eine bessere Idee zur Steuerung haben. Diskussion dazu läuft in Slack.

closed 2023-02-24 by andig

andig, 2023-02-24:
Siehe https://github.com/evcc-io/evcc/discussions/6235