evcc-io/evcc #6423: UI: unterschiedliche Einstellungsmöglichkeiten je nach Browser

VolkerK62, 2023-02-25:
### Describe the bug

Ich bekomme je nach Browser unterschiedliche Anzeigen.
Im Chrome habe ich Zugriff auf die Carport Einstellungen (Symbol bei den Lademodi)
Im Firefox (und auf dem Smartphone) wird dieses Symbol nicht angezeigt.

Chrome:
[image]

Firefox:
[image]

### Steps to reproduce

Ich hatte dieses Feature in der vorherigen Version aktiviert und nutze normalerweise Chrome Browser.

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.113.0

VolkerK62, 2023-02-25:
Nachdem ich den Cache vom Chrome gelöscht habe, ist die Einstellungsmöglichkeit weg. Insofern Issue gelöst.

Wo ist die Option hin? 
Gab es die, weil experimentell, nur im Nighly?

andig, 2023-02-25:
Unter Safari Desktop ist alles da.

VolkerK62, 2023-02-25:
Hast du Mal den Cache geleert? Danach war es bei mir weg.

andig, 2023-02-25:
Ich leere nie den Cache- das ist Voodoo. Force-reload und es ist weiter da, ebenso neues Incognitofenster.

VolkerK62, 2023-02-25:
Ich kann nur sagen, was bei mir passiert ist.

andig, 2023-02-25:
Du könntest mal in der Entwicklerkonsole (Desktop) schauen ob es da ein Element überhaupt gibt: 
[image]

int5749, 2023-02-25:
Ich habe gerade Version 113.1 installiert und in einem neuen in-private Fenster fehlen die Einstellungen
[image]

VolkerK62, 2023-02-25:
Das wird dort ausgegeben:

`<div class="mb-3 d-flex align-items-center" data-v-55867613=""><div class="mode-group border d-inline-flex flex-grow-1" role="group" data-v-fd0f9db5="" data-v-55867613=""><button type="button" class="btn flex-grow-1 flex-shrink-1" data-v-fd0f9db5="">Aus</button><button type="button" class="btn flex-grow-1 flex-shrink-1 active" data-v-fd0f9db5="">PV</button><button type="button" class="btn flex-grow-1 flex-shrink-1" data-v-fd0f9db5="">Min+PV</button><button type="button" class="btn flex-grow-1 flex-shrink-1" data-v-fd0f9db5="">Schnell</button></div></div>`

hambest, 2023-02-25:
Bei mir fehlen auch die Einstellungen. Safari 16.3

hambest, 2023-02-25:
Auch auf Microsoft Edge wird der Knopf nicht angezeigt.

naltatis, 2023-02-26:
Ok, große Verwirrung hier. Ich glaube das ist kein technisches Probleme über das hier diskutiert wird. Ich probiere mal zu lösen 😄

In dem Loadpoint Settingsdialog befinden sich:
- 1p/3p Konfiguration (wenn der Charger das unterstützt)
- experimentelle Konfigurationsoptionen wie min/maxcurrent, minsoc (wenn ich die in eurem Browser(!) aktiviert habt)

[image]

Unterstützt euer Charger kein 1p/3p oder die Experimental (nur nightly) sind aus wird das Konfigurationsicon nicht angezeigt.

Bitte checkt mal ob das bei euch zutrifft.

int5749, 2023-02-26:
Klassisches Jain, würde ich sagen ;-)

Ja, es geht um ein Feature, welches über den Switch Experimentals aktiviert wurde.
Nein, da vermutet wurde, dass das Feature zur Einstellung es in den Stable geschafft hatte.

Etwas verwirrend war, das es bei einigen noch funktioniert hatte (inkl. mir) => Browser cache sei dank.
Funktionieren heißt in dem Fall nicht nur sehen, sondern eben funktional.

Wenn nun geklärt ist, dass es dieses Feature noch nicht ins Release geschafft hat, sollte dies Issue zumindest geschlossen sein.

Bleibt die Frage (bzw, schließt sich an) was braucht es evtl. noch um das Feature aktiv zu nehmen?

hambest, 2023-02-26:
Da ich ja Evcc  auf einem Mac laufen habe, kann ich die nightly nicht laden. Den Schalter Experimental gibt es bei mir nicht.

naltatis, 2023-02-26:
> Nein, da vermutet wurde, dass das Feature zur Einstellung es in den Stable geschafft hatte.

Um welches Feature geht es hier denn was es ins nightly geschafft haben soll? Vermutlich hab ich dann was überlesen.

int5749, 2023-02-26:
> > Nein, da vermutet wurde, dass das Feature zur Einstellung es in den Stable geschafft hatte.
> 
> Um welches Feature geht es hier denn was es ins nightly geschafft haben soll? Vermutlich hab ich dann was überlesen.

[Hepp ;-)](https://github.com/evcc-io/evcc/discussions/6165)

int5749, 2023-02-26:
> 

Das war mit ein Grund, warum ich dies auf einen Linux Rechner umgezogen habe => Flexibilität und Stabilität
Ich hatte früher alles unter Windows 10 laufen, da kam neben der fehlenden Flexibilität noch die ständigen Reboots dazu.
Ein Windows Server wäre much to much overhead gewesen. Also habe ich mir einen kleinen PC gekauft, Proxmox installliert und auf dem laufen nun diverse Linux Container. Für EVCC einer mit der stable (manuell installiert) und einer mit der Nightly. Bei Bedarf aktualisiere ich die Nightly, ziehe die evcc "executable" per FileZilla und ab auf's stable. Läuft wie die Katze von Schmitz ;-)

Aber das führt hier im Issue zu sehr off-topic. In Deinem Fall heißt es dann leider warten, bis es released wurde und im offiziellen Build ist :-/

andig, 2023-02-26:
Bitte nochmal konkret: WAS fehlt jetzt das es vorher gab?

andig, 2023-02-26:
Ich hatte verstanden: der Settingsknopf, siehe Eingangspost

andig, 2023-02-26:
@int5749 deine Kommentare haben nix mit den Screenshots oben zu tun?

andig, 2023-02-26:
@naltatis das sollte ohne experimental gehen https://github.com/evcc-io/evcc/issues/6423#issuecomment-1445076355 ?

int5749, 2023-02-26:
> Bitte nochmal konkret: WAS fehlt jetzt das es vorher gab?

@andig Eigentlich nichts!! Es wurde vermutet, das eine Funktion aus dem Nightly (diese wurde [hier](https://github.com/evcc-io/evcc/discussions/6165) gewünscht) aus dem Nightly Build in die Version 113.x übernommen wurde.

> Ich hatte verstanden: der Settingsknopf, siehe Eingangspost

Den kann man ja "nur" über das Nightly (experimentell) aktivieren. 

> @int5749 deine Kommentare haben nix mit den Screenshots oben zu tun?

Ich meine doch, denn es geht um die o.g. Funktion, welche nur über die "Regler" aus dem Nightly aktivierbar sind.

hambest, 2023-02-26:
Ich hatte ja diesen Wunsch #6165 (comment) geäußert und es wäre schön wenn man dies ohne Nightly oder Experimentell in der offiziellen Version machen könnte.

VolkerK62, 2023-02-26:
Da ich das Ding eröffnet habe ...

In der 0.113.0 gab es in den Einstellungen keine Möglichkeit "experimentell" zu aktivieren. Der Schalter war einfach nicht da. Deshalb bin ich davon ausgegangen, dass es nicht mehr experimentell ist, sondern fester Bestandteil. Dann passierte das, was oben im Eingang beschrieben ist.

Aktuell in der Version 0.113.2 (742b2bea) ist die Aktivierungsmöglichkeit für "experimentell" in den Einstellungen wieder gegeben.
Alles ist gut.

int5749, 2023-02-26:
Sorry, aber evtl. bin ich zu ....? Ich habe gerade ein neues in-private Fenster geöffnet und ..
[image]

VolkerK62, 2023-02-26:
ok. du hast Release 0.113.2   Ich habe das aktuelle Nightly.

andig, 2023-02-26:
> In der 0.113.0 gab es in den Einstellungen keine Möglichkeit "experimentell" zu aktivieren. Der Schalter war einfach nicht da. 

Welcher Schalter? Wenn wir das jetzt nicht konkret bekommen mache ich zu. Der aus Deinem Eingangspost?

VolkerK62, 2023-02-26:
siehe Bild von int5749 von vor 5 Minuten. Der "Schalter" für experimentell, der sich unterhalb von Telemetrie befinden sollte.

andig, 2023-02-26:
Ich gebs auf. Bitte neues Issue. Was Du schreibst hat nix mit der Eingansbeschreibung zu tun.

closed 2023-02-26 by andig

VolkerK62, 2023-02-26:
Den **"Experimentell"**  meine ich. Der war weg und ist jetzt wieder da. Warum auch immer.
[image]

naltatis, 2023-02-26:
Ich hab ne Vermutung wo die Verwirrung herkommt:
Der "Experimentell 🧪" Schalter ist nur in Nightly Builds, aber niemals im Release zu sehen. Das ist anders als bei den meisten anderen Funktionen die "erst im Nightly dann fest ab dem nächsten Release dabei sind".

Die Funktionen hinter dem Experimentell Schalter sollen, zumindest in der jetzigen Form, nicht in den Release-Versionen (also von breitem Publikum) verwendet werden.