evcc-io/evcc #6429: RCT-Power: Integration Ertrag eines Fremdwechselrichters

andig, 2023-02-25:
### Discussed in https://github.com/evcc-io/evcc/discussions/6427

<div type='discussions-op-text'>

<sup>Originally posted by **JAK0721** February 25, 2023</sup>
Hallo, 

ich habe meine RCT-Power Anlage um ein Balkonkraftwerk erweitert. Der RCT-Power Wechselrichter bietet die Möglichkeit den Ertrag eines Fremdwechselrichters über eine S0 Schnittstelle einzulesen. Dieser zusätzlich Ertrag bekommt man unter dem "G" Symbol angezeigt. 
Besteht ggf. die Möglichkeit diesen Ertrag auf den "normalen" PV Ertrag drauf zu rechnen, s.d. man evcc die Gesamtleistung anzeigt. Negative Auswirkungen bzgl. dem Überschussladen konnte ich nicht feststellen, wenn evcc keine Kenntnis von diesem Ertrag hat. Wahrscheinlich eher ein nice-to-have Feature.

Danke schon mal vorab!
Viele Grüße
[image]

</div>

andig, 2023-02-25:
@Pitillo wir addieren heute schon GenA und GenB auf:

		a, err := m.conn.QueryFloat32(rct.SolarGenAPowerW)
		if err != nil {
			return 0, err
		}
		b, err := m.conn.QueryFloat32(rct.SolarGenBPowerW)
		return float64(a + b), err

Funktioniert das bei Dir nicht? Was sagt 

    evcc meter --log trace

im Vergleich zu Deiner RCT App?

/cc @mlnoga vllt. hast Du noch eine Idee?

Pitillo, 2023-02-25:
@andig ich denke du hast hier den falschen getaggt. Ich glaube das sollte ein @JAK0721 werden :D

andig, 2023-02-25:
Oh sorry- da ist das Autocomplete auf Deinen native Namen angesprungen :O

JAK0721, 2023-02-25:
Hi, es geht mir nicht um die Addition von String A und B, welche ja an den Wechselrichter angeschlossen werde, sondern um eine Addition der Leistung eines externen Wechselrichters, dessen Leistung über die S0 Schnittstelle erfasst wird und in der App unter dem Symbol "G" angezeigt wird. (Eingang B ist bei mir nicht angeschlossen)

andig, 2023-02-25:
Das Modul bietet aktuell folgende Werte:

	// power
	//
	SolarGenAPowerW  Identifier = 0xB5317B78 // float32
	SolarGenBPowerW  Identifier = 0xAA9AA253 // float32
	BatteryPowerW    Identifier = 0x400f015b // float32, positive = discharge, negative = charge
	InverterACPowerW Identifier = 0xDB2D69AE // float32
	RealPowerW       Identifier = 0x4E49AEC5 // float32
	TotalGridPowerW  Identifier = 0x91617C58 // float32, positive = taken from grid, negative = feed into grid
	BatterySoC       Identifier = 0x959930BF // float32, range 0 ... 1

JAK0721, 2023-02-25:
Unter https://rctclient.readthedocs.io/en/latest/inverter_registry.html habe ich 
0xE96F1844 | FLOAT | FLOAT |   | io_board.s0_external_power
entdeckt. Leider ohne Beschreibung. Klinkt aber passend.

andig, 2023-02-25:
Wartet auf https://github.com/mlnoga/rct/pull/9

mlnoga, 2023-02-27:
In der mir vorliegenden V1.13 der RCT Power Serial Communication Protocol Spezifikation findet sich dieser Wert auch, leider ebenfalls ohne Beschreibung. Ich nehme den PR mal an, da er bestehende Funktionalität nicht beeinflusst. 

@JAK0721, bitte teste ob er das Gewünschte leistet. Hast Du bereits dokumentiert, wie Du das Balkonkraftwerk integriert hast? Das wäre ggf. für andere Nutzer interessant.

andig, 2023-02-27:
Closed in 78e51198552e91f13f99b3323918ae048bfaec84

closed 2023-02-27 by andig

JAK0721, 2023-03-01:
@mlnoga & @andig,

besten Dank für Eure Mühe meine Anfrage so prompt umgesetzt zu haben. Funktioniert perfekt.

Für die Integration eines Fremdwechselrichters braucht es einen separaten Stromzähler mit einer S0 Schnittstelle. Details hat RCT eigentlich gut in https://www.rct-power.com/de/download-bereich-de.html?file=files/Download-Bereich/Download%20Bereich%20DE/8_Technische%20Dokumentation/TD4_Vernetzung%20von%20mehreren%20Invertern%20%C3%BCber%20S0-Schnittstelle_Web19V1DE.pdf beschrieben.
Allerdings hat sich RCT vermutlich nicht exakt an die EN 62053-31 gehalten. Mein Eltako Stromzähler funktioniert nicht ohne Umwege. Ich musste einen Optokoppler dazwischen schalten. Bei Bedarf kann ich gern auf die Details eingehen.

mlnoga, 2023-03-03:
Prima, freut mich.

torsten1106, 2023-05-01:
> @mlnoga & @andig,
>
> besten Dank für Eure Mühe meine Anfrage so prompt umgesetzt zu haben. Funktioniert perfekt.
>
> Für die Integration eines Fremdwechselrichters braucht es einen separaten Stromzähler [...]

Hallo, kannst du ein Foto von der Konfiguration in der RCT App hochladen? Bei meiner Integration eines Eltako mit S/0 Schnittstelle komme ich zu keiner erfolgreichen Anzeige unter G.

JAK0721, 2023-05-02:
@torsten1106 , hier die Einstellungen
[image]
[image]

Verkabelt muss es genauso wie in der Anleitung beschrieben, mit der Brücke zw. dem 5V Pin und Pin 1.

Viel Erfolg

kschaumann, 2023-05-23:
@JAK0721 sorry für das hervorholen des issues, aber ich beziehe mich auf

> Bei Bedarf kann ich gern auf die Details eingehen.

Magst Du hier mal kurz beschreiben, wie Deine Schaltung aussieht? Ich habe mich an den Entwurf von [wiki.freesol.de](https://wiki.freesol.de/index.php?title=Aufbauanleitungen#Stromz.C3.A4hler_mit_belegtem_S0-Ausgang) gehalten. Ich bekomme aus diesem auch ein Signal, was mein  [PulseCounter](https://www.stall.biz/project/pulsecounter-2-komfortabel-verbraeuche-von-strom-wasser-gas-und-solar-messen/) auch auswerten kann. Allerdings der RCT-Wechselrichter nicht.

JAK0721, 2023-05-23:
[image]

kschaumann, 2023-05-23:
@JAK0721 Vielen Dank, ok, meine Schaltung hat jeweils einen Vorwiderstand vor den Optokopplern und steuert eben alle drei an, aber sonst unterscheidet sie sich quasi nicht. Ich hab allerdings einen CNY17-3 keinen 4N35 verwendet. Kann aber eigentlich bei so einem trivialen Signal keinen Unterschied machen, meine ich.

JAK0721, 2023-05-24:
@kschaumann, hier eine Messung der Signale mit einem Ozi. Wenn der Optokoppler auf der RCT Seite durchschaltet, bricht die Spannung von 5V auf 1,57V ein. 

[image]

kschaumann, 2023-05-24:
@JAK0721 Wow, danke für den ScreenShot vom Oszi! Ich habe leider derzeit keines zur Hand, arbeite da aber an einer Lösung ;)

timoglaess, 2025-01-01:
Danke dafür. Funktioniert.
Wenn ich nun Pin 2 für einen weiteren S0 Zähler nutzen möchte, kann ich ja nicht einfach ebenfalls mit 5V verbinden, da ja dann 5V/ Pin1 und Pin2 kurzgeschlossen wären??
Müsste ich hier dann eine Diode dazwischen setzen, damit Pin2 nicht Pin 1 mit beeinflusst?

JAK0721, 2025-01-02:
@timoglaess , bin kein Elektroniker und kenne den internen Aufbau des RCT WR nicht. Am besten Du stellst Deine Frage direkt an den RCT Support. Viel Erfolg.