evcc-io/evcc #6433: Fehler im commit E3DC: add external generators

maikheinrich, 2023-02-25:
### Describe the bug

Seit der Version 0.113 wird im E3DC Template wird jetzt automatisch der zusätzliche Einspeisen ebenfalls abgefragt.
Leider werden die beiden Werte vom internen PV und vom zusätzlicher Einspeiser nicht addiert sondern subtrahiert.

Hier muss in in der Datei templates/definition/meter/e3dc.yaml nach der Zeile 35 "scale: -1" eingefügt werden. Dann ist es so wie beabsichtigt.

LG Maik

### Steps to reproduce

update von 0.112.x auf 0.113.0

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.113.0

andig, 2023-02-25:
Du meinst https://github.com/evcc-io/evcc/pull/6428?

maikheinrich, 2023-02-25:
ja

andig, 2023-02-25:
Ich glaube die Verwirrung kommt daher, dass der zweite Wert ein anderes Vorzeichen zu haben scheint und daher genau *nicht* addiert werden darf.

maikheinrich, 2023-02-25:
Hallo sandig,

E3DC betrachtet den Zusatzzähler wie die Batterie. Minuswerte liefern Energie (Erzeuger) in das System und Pluswerte (Verbraucher) verbrauchen die Energie.

Daher muss mit "scale: -1" das Vorzeichen getauscht werden. Damit der Wert positiv ist und addiert werden kann.

LG Maik

andig, 2023-02-25:
Closed in https://github.com/evcc-io/evcc/commit/123e69afc9465fb80aa8c81922396a507f7a842d