evcc-io/evcc #6439: Added two Fronius inverters. -> "Unexpected token '}', "{"":}" is not valid JSON [{"":}]"

buliwyf42, 2023-02-25:
### Describe the bug

Today i added my PV setup.

Sometimes i get the following error in the Webinterface:
Unexpected token '}', "{"":}" is not valid JSON [{"":}]

### Steps to reproduce

Today i added my PV setup.

Sometimes i get the following error in the Webinterface:
Unexpected token '}', "{"":}" is not valid JSON [{"":}]

### Configuration details

```yaml
nterval: 10s # control cycle interval

tariffs:
    currency: EUR
    grid:
        type: fixed
        price: 0.2369
    feedin:
        type: fixed
        price: 0.075

meters:
  - name: my_fronius_smartmeter
    type: template
    template: fronius-gen24
    usage: grid
    host: 192.168.215.156
  - name: my_fronius_pv1
    type: template
    template: fronius-gen24
    usage: pv
    host: 192.168.215.156
  - name: my_fronius_pv2
    type: template
    template: sunspec-inverter
    host: 192.168.215.190
  - name: my_byd_battery
    type: template
    template: fronius-gen24
    usage: battery
    host: 192.168.215.156

chargers:
  #- type: template
  #template: go-e-v3
  #host: 192.168.211.204
  #name: wallbox1
- type: template
  template: go-e-v3
  host: 192.168.211.33
  name: wallbox2

loadpoints:
- title: go-e rechts
  charger: wallbox2
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  enable:
    threshold: 0
    delay: 1m
  disable:
    threshold: 0
    delay: 1m

site:
  title: xxxx
  meters:
    grid: my_fronius_smartmeter
    pv:
      - my_fronius_pv1
      - my_fronius_pv2
    battery: my_byd_battery
  bufferSoc: 95
  prioritySoc: 99
```

### Log details

```text
There is nothing relevant in the logfile
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.113.1

VolkerK62, 2023-02-25:
Ich konnte es vorhin mal provozieren. 
Es wurde geladen und ich habe auf "aus" geklickt. In dem Moment erschien das rote Dreieck mit dem Fehler.

VolkerK62, 2023-02-25:
kann #6440 damit zusammenhängen? Tauchen nämlich gleichzeitig auf.

andig, 2023-02-25:
Gut möglich. Da das Ding keinen Namen hat ist es leider superschwer rauszufinden, wo das her kommt :(

buliwyf42, 2023-02-25:
> kann #6440 damit zusammenhängen? Tauchen nämlich gleichzeitig auf.

Ja, so einen Eintrag sehe ich auch. 

0xc0017f0ba0

thebrainkafka, 2023-02-25:
Hab ich auch mit go-e Gemini .
Kommt direkt mit Start Ladevorgang

VolkerK62, 2023-02-25:
Ich kann nicht mehr sicher nachvollziehen, wann es das erste Mal aufgetreten ist. Aufgefallen ist es mir vor ein paar Tagen, als es die Problem bei Bluelink gab und ständig die Fehlermeldungen kamen. 
Aktuell hängt es wohl mit Start und/oder Ende der Ladung zusammen.

buliwyf42, 2023-02-25:
Hängt es eventuell mit der Telegramm integration zusammen. Erhaltet aktuell auch folgenden Fehler:

`[telegram] ERROR 2023/02/25 17:12:26 Conflict: terminated by other getUpdates request; make sure that only one bot instance is running
[telegram] ERROR 2023/02/25 17:12:26 Failed to get updates, retrying in 3 seconds...`

andig, 2023-02-25:
Bitte mal in 15min das nightly probieren. Ich hab einen Debugschritt mit eingebaut der einen Fehler in Log und UI ausgeben sollte falls wir die Ursache vorher erkennen.

andig, 2023-02-25:
UI-Fehler ist weg und ich sehe

    [server] ERROR 2023/02/25 18:01:55 invalid key/val for {Loadpoint:<nil> Key: Val:0xc001229560}, please report to https://github.com/evcc-io/evcc/issues/6439

Logging nochmal erhöht

VolkerK62, 2023-02-25:
ganz frisches Nightly

`Feb 25 18:17:19 raspberrypi evcc[14445]: [server] ERROR 2023/02/25 18:17:19 invalid key/val for {Loadpoint:<nil> Key: Val:0x593edc0} 0x593edc0, please report to https://github.com/evcc-io/evcc/issues/6439
`

thebrainkafka, 2023-02-25:
[image]
[image]

thebrainkafka, 2023-02-25:
Gleicher Ladevorgang.
Am Laptop Browser obiger Fehler. Am Handy der untenstehenden.
113.1

andig, 2023-02-25:
Nightly nochmals aktualisiert

VolkerK62, 2023-02-25:
```
Feb 25 20:19:41 raspberrypi evcc[16612]: [server] ERROR 2023/02/25 20:19:41 invalid key/val for {Loadpoint:<nil> Key: Val:0x5741f80} (util.flush)(0x5741f80), please report to https://github.com/evcc-io/evcc/issues/6439
```

andig, 2023-02-25:
VOLKER- Du bist mein Held! Danke, jetzt weiss ich wo es her kommt. Der Wert darf tatsächlich nie im Cache landen.

andig, 2023-02-25:
Release in 15min

closed 2023-02-25 by andig