evcc-io/evcc #6442: Bei mir funktioniert es unter Debian auch nicht mehr ab der 0.113, mit der 0.112.5 alles ok.

pauxus, 2023-02-25:
Bei mir funktioniert es unter Debian auch nicht mehr ab der 0.113, mit der 0.112.5 alles ok.

EVCC lässt sicht nicht über systemctl starten, ich nutze den EVCC Useraccount nicht zur Anmeldung.
EVCC lässt sich im Moment nur direkt über die Konsole starten.

Hier die Befehle die Du sehen wolltest:

root@server:/home/steffen# systemctl start evcc
root@server:/home/steffen# sudo ps aux | grep evcc
root        2665  0.0  0.0   6264   640 pts/0    S+   15:56   0:00 grep evcc
root@server:/home/steffen# evcc
[main  ] INFO 2023/02/25 15:56:20 evcc 0.113.1
[main  ] INFO 2023/02/25 15:56:20 using config file: /etc/evcc.yaml
[main  ] INFO 2023/02/25 15:56:20 starting ui and api at :80
[db    ] INFO 2023/02/25 15:56:20 using sqlite database: /root/.evcc/evcc.db
root@server:/home/steffen# getent passwd evcc
evcc:x:106:112:evcc runtime user,,,:/var/lib/evcc:/bin/false
root@server:/home/steffen#

Grüße Steffen

_Originally posted by @Steffen294 in https://github.com/evcc-io/evcc/discussions/6419#discussioncomment-5108706_

Steffen294, 2023-02-25:
root@server:/home/steffen# sudo systemctl status evcc
● evcc.service - evcc
     Loaded: loaded (/lib/systemd/system/evcc.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) since Sat 2023-02-25 16:20:20 CET; 3ms ago
    Process: 3959 ExecStart=/usr/bin/evcc (code=exited, status=0/SUCCESS)
   Main PID: 3959 (code=exited, status=0/SUCCESS)
        CPU: 197ms
root@server:/home/steffen# 

Das passierte wie gesagt nach dem Update von 112.5 auf 113

Ich habe dann EVCC komplett gelöscht mir allen Ordnern, Datenbanken, EVCC user etc. und dann nochmal installiert.
Nix zu machen. Der Dienst will nicht starten.

Bei der Eingabe von systemctl start evcc dauert es ca.10 Sekunden, wir dann aber ohne Meldung beendet und EVCC läuft nicht.

Hier noch der Inhalt von evcc.service

# evcc.service
#

[Unit]
Description=evcc
Requires=network-online.target
After=syslog.target network.target network-online.target
Wants=network-online.target
StartLimitIntervalSec=10
StartLimitBurst=10

[Service]
ExecStart=/usr/bin/evcc
Environment="EVCC_DATABASE_DSN=/var/lib/evcc/evcc.db"
Restart=always
RestartSec=10

User=evcc
Group=evcc

[Install]
WantedBy=multi-user.target

maikheinrich, 2023-02-25:
Hast Du den user evcc unter Debian angelegt ?

mit der Version 0.112.5 läuft das evcc (noch) unter dem user root. Und mit der version 0.113 wird dann der user evcc verwendet.

LG Maik

Steffen294, 2023-02-25:
> Hast Du den user evcc unter Debian angelegt ?
> 
> mit der Version 0.112.5 läuft das evcc (noch) unter dem user root. Und mit der version 0.113 wird dann der user evcc verwendet.
> 
> LG Maik

Hi Maik,

der User EVCC wurde bei dem Update auf 0.113 automatisch angelegt.

In der /etc/passwd steht nur folgender Eintrag.
evcc:x:106:112:evcc runtime user,,,:/var/lib/evcc:/bin/false

Soll ich vor dem Update den user evcc selbst von hand anlegen?

VG Steffen

maikheinrich, 2023-02-25:
Kann es sein, daß  der Port von 7070 auf 80 in der evcc.yaml geändert wurde.
Ports unter 1024 darf nur root öffnen.

Steffen294, 2023-02-25:
Was soll ich sagen?

Ja war auf Port 80, zurück auf 7070 und es funktioniert.
Dann lasse ich es auf 7070 stört mich nicht.

DANKE Maik !!!

andig, 2023-02-25:
@pauxus was machen wir damit? Blöd ist ja, dass das für den Anwender nicht vorhersehbar ist. Ich wundere mich allerdings, warum es im Log keine bessere Fehlermeldung gibt.

maikheinrich, 2023-02-25:
Hi andig,

doch steht in /var/log/daemon

Feb 25 17:16:59 evcc systemd[1]: Started evcc.
Feb 25 17:16:59 evcc evcc[9333]: [main  ] INFO 2023/02/25 17:16:59 evcc 0.113.0
Feb 25 17:16:59 evcc evcc[9333]: [main  ] INFO 2023/02/25 17:16:59 using config file: /etc/evcc.yaml
Feb 25 17:16:59 evcc evcc[9333]: [main  ] FATAL 2023/02/25 17:16:59 listen tcp :80: bind: permission denied
Feb 25 17:16:59 evcc systemd[1]: evcc.service: Succeeded.
Feb 25 17:17:10 evcc systemd[1]: evcc.service: Scheduled restart job, restart counter is at 50.
Feb 25 17:17:10 evcc systemd[1]: Stopped evcc.

Grüße

Maik

pauxus, 2023-02-25:
Naja, vermutlich wird journactl schon was zeigen, ich teste das bei Gelegenheit mal. 

Vermutlich könnte man aber in die Doku einen Satz mit aufnehmen, dass Port 80 oder 443 keine gute Idee ist. 

Ich kann natürlich in das Preinstall Skript noch einen Test aufnehmen, aber das halte ich eigentlich für zu speziell. Außerdem würde das ja nur beim Update greifen. 

Umgekehrt könnte man aber vielleicht in evcc eine Warnung bei Port <= 1024 einbauen.

maikheinrich, 2023-02-25:
Die Ports unter 1024 darf nur der User root öffnen. Eventuell könnt Ihr den Startdienst modifizieren sodaß dieser als root starten und DANN die root rechte abgibt und als User evcc weiter läuft. (Der Apache Webserver macht es so).

Ansonsten kann der User der evcc auf Port 80 nutzen möchte,  einen Reverse Proxy aufzusetzen, der  http://localhost:80 zu http://localhost:7070 umleitet. 

Grüße Maik

Steffen294, 2023-02-25:
Vielleicht ein kurzer Hinweis direkt in der der evcc.yaml, den Port nicht zu ändern bzw. größer 1024.

Was mir noch aufgefallen ist, wenn jemand EVCC neu installiert und seine vorhanden Datenbank in den neuen Pfad kopiert:
/var/lib/evcc/evcc.db

Müssen unbedingt die Besitzrechte auf den user evcc geändert werden. 
Ansonst kann nicht in die Datenbank geschrieben werden. Ein entsprechende Fehlermeldung kommt dann auch im UI.

Hier der Befehl dazu:
chown evcc:evcc /var/lib/evcc/evcc.db

andig, 2023-02-25:
> Die Ports unter 1024 darf nur der User root öffnen. Eventuell könnt Ihr den Startdienst modifizieren sodaß dieser als root starten und DANN die root rechte abgibt und als User evcc weiter läuft. (Der Apache Webserver macht es so).

Gerne wenn mir jemand erklären kann wir ich das in Go bewerkstellige

maikheinrich, 2023-02-25:
> > Die Ports unter 1024 darf nur der User root öffnen. Eventuell könnt Ihr den Startdienst modifizieren sodaß dieser als root starten und DANN die root rechte abgibt und als User evcc weiter läuft. (Der Apache Webserver macht es so).
> 
> Gerne wenn mir jemand erklären kann wir ich das in Go bewerkstellige. Dann hätten wir es aber auch gleich weiter unter root laufen lassen können ;)

Hi andig,

eventuell hilft dir dieser Artikel -> https://mgdm.net/weblog/systemd/

Grüße

Maik

maikheinrich, 2023-02-25:
Oder dieser -> https://www.sobyte.net/post/2022-01/systemd-socket-activation/

ThinkEV, 2023-02-26:
@andig Die Lösung ist gar nicht so schwer:

/lib/systemd/system/evcc.service muss nach [Service] um diese Zeile ergänzt werden:

**AmbientCapabilities=CAP_NET_BIND_SERVICE**

Danach startet er auch auf Port 80 als User evcc.

premultiply, 2023-02-26:
@ThinkEV PR welcome

ThinkEV, 2023-02-26:
already submitted :)

andig, 2023-02-26:
…und nur eine Zeile. Nice!

ThinkEV, 2023-02-26:
you are welcome :)
fixed by https://github.com/evcc-io/evcc/pull/6474
issue can be closed.

andig, 2023-02-26:
Wir lassen nochmal auf damit es leichter zu finden ist. Vielen Dank!

Steffen294, 2023-02-26:
Danke euch! 👍

closed 2023-03-08 by andig