evcc-io/evcc #6444: Daheimladen (Modbus Template) Wallbox startet immer mit minimaler Leistung bei Einstellung schnell

StevieC121176, 2023-02-25:
### Describe the bug

Ich habe die Daheimladen Touch jetzt als Modbus Version integriert. Es gibt da einige Auffälligkeiten bei mir.

Starte ich das Laden mit "Schnell" wird immer mit `mincurrent` gestartet. Erst wenn ich `min und maxcurrent `per ui Ladepunkt Einstellung ändere, geht er dann auch bis 16A hoch.

Eine weitere Eigenheit, wenn das Laden beendet ist, bleibt der Modus immer auf "Schnell" stehen. 

Im Log gibt es am Anfang und Ende der Ladung immer Fehler (charger out of sync...) sh. Foto

[image]

[image]

### Steps to reproduce

.

### Configuration details

```yaml
- type: template
  template: daheimladen-mb 
  host: 192.168.178.74 
  port: 502  
  name: wallbox9
```

### Log details

```text
pi@raspberrypi:~ $ evcc charger
[main  ] INFO 2023/02/25 16:19:58 evcc 0.112.5 (f7e21b4c)
[main  ] INFO 2023/02/25 16:19:58 using config file: /etc/evcc.yaml
[db    ] INFO 2023/02/25 16:19:58 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2023/02/25 16:19:58 connecting evcc-983099523 at tcp://192.168.178.77:1883
wallbox10
---------
Power:         0W
Energy:        0.0kWh
Charge status: B
Enabled:       false

wallbox9
--------
Power:          7100W
Energy:         1770.0kWh
Current L1..L3: 15.8A 15.9A 0A
Voltage L1..L3: 225V 228V 239V
Charge status:  C
Enabled:        true

pi@raspberrypi:~ $
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.112.5 (f7e21b4c)

VolkerK62, 2023-02-25:
Ich nutze Version 0.113.0 und habe die DaheimLaden Smart. 
Ich habe es nachgestellt. Bei mir funktioniert alles und es gibt keine Fehler.

StevieC121176, 2023-02-25:
Ich habe die Firmware Version 1.11 für die Touch. Evtl. kann es auch damit zusammen hängen.

stelle ich die Wallbox auf WLAN bekomme ich noch mehr Fehler und die Wallbox lässt sich dann garnicht mehr steurern über EVCC

[image]