evcc-io/evcc #6479: evcc 0.113.2+1677380262: template not found: eastron-sdm72v2_630

ThinkEV, 2023-02-26:
### Describe the bug

FATAL 2023/02/26 23:43:27 cannot create meter 'charge5': cannot create meter 'template': template not found: eastron-sdm72v2_630

Wie lautet der korrekte Name des Templates für das SDM72DM-v2?
Ich habe gesehen, dass @premultiply Änderungen an den Templates vorgenommen hat, aber irgendwie klappt es noch nicht...

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
- type: template
  #template: eastron-sdm72_220_230
  template: eastron-sdm72v2_630
  id: 2
  device: /dev/ttyUSB0
  baudrate: 9600
  comset: 8N1
  usage: charge
  modbus: rs485serial
  name: charge5
```

### Log details

```text
FATAL 2023/02/26 23:43:27 cannot create meter 'charge5': cannot create meter 'template': template not found: eastron-sdm72v2_630
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.113.2 (742b2bea)

premultiply, 2023-02-26:
`template: eastron`

ThinkEV, 2023-02-26:
Es klappt nicht mit eastron
FATAL 2023/02/26 23:52:23 cannot create meter 'charge5': cannot create meter 'template': invalid key: usage
und auch nicht it eastron-sdm72
FATAL 2023/02/26 23:54:15 cannot create meter 'charge5': cannot create meter 'template': template not found: eastron-sdm72

Es geht nur mit eastron-sdm72_220_230

premultiply, 2023-02-26:
In deiner Version ist die Template-Änderung dann noch nicht drin.
Die kommt ja erst heute nacht ins Nightly...

closed 2023-02-26 by premultiply