evcc-io/evcc #6495: Fahrzeug wird nicht mehr erkannt

TTXaser2, 2023-02-27:
Hallo,

seit dem Update auf die aktuelle Version 0.113.1 kriege ich für meinen VW ID3 weder den Namen noch den aktuellen Ladestand angezeigt, stattdessen folgenden Fehler:

`vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'id': terms of service updated- please open app or website and confirm`

Laden funktioniert zwar, allerdings wird der Ladestand immer mit 100% angezeigt. An der Config hab ich eigentlich nix geändert, vor dem Update lief EVCC reibungslos. 

Kann mir jemand helfen?

goebelmeier, 2023-02-27:
Probiere ein Mal bitte über die Webseite zu gehen. Frag' mich nicht wieso, aber manchmal gibt's ToS in der App zu bestätigen, manchmal auf der Webseite, manchmal an beiden Orten.

Maergsche, 2023-02-27:
Über die Webseite myvolkswagen bekomme ich die Werte angezeigt:

Ich habe auch seit ca. 14:30 Medungen im evcc WebInterface:

vehicle soc: Get \"https://mobileapi.apps.emea.vwapps.io/vehicles/WVWZZZE1ZLP00000x/selectivestatus?jobs=all\": unexpected status: 404
vor 5 Minuten
vehicle odometer: Get \"https://mobileapi.apps.emea.vwapps.io/vehicles/WVWZZZE1ZLP00000x/selectivestatus?jobs=all\": unexpected status: 404
vor 9 Minuten
vehicle range: Get \"https://mobileapi.apps.emea.vwapps.io/vehicles/WVWZZZE1ZLP00000x/selectivestatus?jobs=all\": unexpected status: 404
vor 9 Minuten
vehicle soc limit: Get \"https://mobileapi.apps.emea.vwapps.io/vehicles/WVWZZZE1ZLP00000x/selectivestatus?jobs=all\": unexpected status: 404
vor 9 Minuten
vehicle soc: Get \"https://mobileapi.apps.emea.vwapps.io/vehicles/WVWZZZE1ZLP00000x/selectivestatus?jobs=all\": unexpected status: 404 (ignored by estimator)
vor 1 Stunde
invalid message template for soc: wanted: [vehicleSoC], got: [tariffGrid vehicles climater savingsSelfConsumptionPercent chargeTotalImport minCurrent vehicleCapacity savingsSelfConsumptionCharged chargeDuration siteTitle vehicleOdometer mode pvPower prioritySoc tariffEffectivePrice savingsGridCharged loadpoint phasesEnabled targetSoc savingsTotalCharged savingsEffectivePrice savingsSince connected batteryPower version battery homePower chargeCurrent chargerIcon chargeRemainingEnergy chargeVoltages vehicleFeatureOffline gridPower enabled pvConfigured pv maxCurrent phasesConfigured vehicleRange auxPower targetTime vehicleTargetSoc batteryCapacity vehiclePresent savingsAmount auth batterySoc charging targetEnergy pvRemaining chargeRemainingDuration vehicleSoc gridConfigured minSoc chargeCurrents chargedEnergy pvAction chargerFeatureIntegratedDevice batteryConfigured phaseAction vehicleIcon sponsor title tariffFeedIn chargePower currency connectedDuration phasesActive bufferSoc sponsorValid vehicleDetectionActive vehicleTitle residualPower phaseRemaining planActive greenShare]

Flauschi83, 2023-02-27:
same für audi etron:

`[main  ] ERROR 2023/02/27 15:37:10 creating vehicle ev1 failed: cannot create vehicle 'template': cannot create vehicle 'etron': unexpected status: 404
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'etron': unexpected status: 404
Capacity: 0.0kWh
`

merlinste, 2023-02-27:
Same here.
Da hat VW sicherlich an der API was geändert / kaputt gemacht / dekativiert...

`[lp-1  ] ERROR 2023/02/27 15:55:25 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'id': form not found`

thebrainkafka, 2023-02-27:
Same here 
Volkswagen ID.4

p7lejo, 2023-02-27:
Kann ich bestätigen ID.3 (egal ob mit template "id" oder "vw"), 
```
pi@rapi2 ~ $ evcc vehicle -l debug
[main  ] INFO 2023/02/27 17:45:11 evcc 0.113.2
[main  ] INFO 2023/02/27 17:45:11 using config file: /home/pi/evcc.yaml
[db    ] INFO 2023/02/27 17:45:12 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2023/02/27 17:45:12 connecting evcc at tcp://localhost:1883
[mqtt  ] DEBUG 2023/02/27 17:45:12 tcp://localhost:1883 connected
[eebus ] INFO 2023/02/27 17:45:12 Local SKI:  7247dcee8526f8234763d670777xxx
[eebus ] DEBUG 2023/02/27 17:45:12 mdns: announce
[eebus ] DEBUG 2023/02/27 17:45:12 mdns: using avahi
[eebus ] DEBUG 2023/02/27 17:45:13 starting websocket server on :4712
[main  ] ERROR 2023/02/27 17:45:13 creating vehicle ev4 failed: cannot create vehicle 'template': cannot create vehicle 'id': form not found

**ev4
---
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'id': form not found
Capacity: 0.0kWh**
```

Auffällig war, dass in der App "we connect id" auch erst stand "Fahrzeug synchronisiert vor einem Tag". Dort geht nun aber seit (mindestens) 17 Uhr aber alles wieder normal.

thebrainkafka, 2023-02-27:
Tronity synchronisiert auch nicht aktuell.

RTTTC, 2023-02-27:
All good with Enyaq.

aBua-MUC, 2023-02-27:
Same with Q4 etron
`today at 20:24:40[lp-1  ] ERROR 2023/02/27 20:24:40 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'etron': unexpected status: 404`

ropgenoo, 2023-02-27:
Gleiches Problem hier beim ID.3.
Was mir aufgefallen war, ich musste heute Mittage in der VW ID Connect APP die Zugangsdaten neu eingeben...

TTXaser2, 2023-02-27:
> Probiere ein Mal bitte über die Webseite zu gehen. Frag' mich nicht wieso, aber manchmal gibt's ToS in der App zu bestätigen, manchmal auf der Webseite, manchmal an beiden Orten.

TTXaser2, 2023-02-27:
Sorry für den Close, hab mich verklickt...

AES-256-GCM, 2023-02-27:
@TTXaser2 Regarding your initial problem: I had that terms of conditions thingy some days ago, too. It worked again, after logging in on the website and accepting it.

To me it seams `[lp-1  ] ERROR 2023/02/27 22:52:13 vehicle soc: vehicle not available: cannot create vehicle 'template': 
cannot create vehicle 'id': form not found` is another issue which came up today. I guess VW did some API changes. At least applications which are using another WeConnect API library are having problems as well, like https://github.com/tillsteinbach/VWsFriend/issues/435.

TTXaser2, 2023-02-27:
> Probiere ein Mal bitte über die Webseite zu gehen. Frag' mich nicht wieso, aber manchmal gibt's ToS in der App zu bestätigen, manchmal auf der Webseite, manchmal an beiden Orten.

Danke für den Hinweis - in der Tat gab's auf der WebSite neue ToS, denen ich auch zustimmen konnte/musste. Seitdem ist der von mir oben in der GUI gemeldete Fehler verschwunden, leider wird das Fahrezug immer noch nicht erkannt, ebenso nicht Ladestand etc.

AES-256-GCM, 2023-02-27:
@TTXaser2 Jep, I guess so. Maybe you should close this issue since your initial problem is solved an I will create a new issue addressing the second problem.

thebrainkafka, 2023-02-27:
Das scheint der neue login link zu sein.
Aus einem anderen Forum entnommen (meinID Forum)

https://identity.vwgroup.io/oidc/v1 https://emea.bff.cariad.digital/user-login/v1/authorize

thierolm, 2023-02-27:
> Same with Q4 etron `today at 20:24:40[lp-1 ] ERROR 2023/02/27 20:24:40 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'etron': unexpected status: 404`

Ich bekomme auch den Audi E-Tron Fehler:
```bash
...
[etron ] TRACE 2023/02/27 18:26:15 POST https://idkproxy-service.apps.emea.vwapps.io/v1/emea/token
[etron ] TRACE 2023/02/27 18:26:16 client_id=xxx7049ad26%40apps_vw-dilab_com&code=xxxx&code_verifier=b2VFRmJTQlBuclJJeDF2YU9YU0lVTzdxaXY1bVN6UVM&grant_type=authorization_code&redirect_uri=myaudi%3A%2F%2F%2F&response_type=token+id_token
--
404 Not Found: Requested route ('idkproxy-service.apps.emea.vwapps.io') does not exist.
[main  ] ERROR 2023/02/27 18:26:16 creating vehicle audi failed: cannot create vehicle 'template': cannot create vehicle 'etron': unexpected status: 404

audi
----
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'etron': unexpected status: 404
```

Es sieht nach Änderungen in der Audi Connect API aus. Die betroffene URL gibt auch via curl die entsprechende Ausgabe zurück. (Ich habe alle Permissions auf der MyAudi Webseite angeklickt):
```bash
thierolm@LAPTOP-KQMMVN3H:~$ curl -X POST https://idkproxy-service.apps.emea.vwapps.io/v1/emea/token
404 Not Found: Requested route ('idkproxy-service.apps.emea.vwapps.io') does not exist.
```

oligom1, 2023-02-28:
Same with my ID3 + ID4 Sinne yesterday

21Kingshill, 2023-02-28:
> Das scheint der neue login link zu sein. Aus einem anderen Forum entnommen (meinID Forum)
> https://emea.bff.cariad.digital/user-login/v1/authorize

Wenn sich der Link wie hier genannt geändert hat, wäre das ja die Erklärung. Der Vollständigkeit halber hier noch die Fehlermeldung für Audi aus der EVCC GUI:

climater: Get \"https://mobileapi.apps.emea.vwapps.io/vehicles/WAUZZZXXXXXXXXXXX/selectivestatus?jobs=all\": unexpected status: 404

andig, 2023-02-28:
VW ist behoben, Audi bleibt offen, dafür ist noch keine Lösung bekannt.

closed 2023-02-28 by andig

thebrainkafka, 2023-02-28:
Danke.
Heißt es gibt ein neues Release?

Flauschi83, 2023-02-28:
> VW ist behoben, Audi bleibt offen, dafür ist noch keine Lösung bekannt. Skoda/Seat funktioniert soweit bekannt weiter.

kann ich Dich mit Zugangsdaten unterstützen oder hängts noch wo anders?

andig, 2023-02-28:
Es ist keine Lösung bekannt. Da hilft nur tracen der App- das muss jemand tun und die Erkenntnisse veröffentlichen.

VacalTarka, 2023-02-28:
Habe gerade vor 1h die Installation mit Sungrow/Easee und "Audi Andere" (Q5 PHEV) vorgenommen und kam mit dem Fehler 404 auch nicht weiter.
Nun ja, der Fehler scheint hier (mal wieder) bei Audi zu liegen.
Wenn ich irgendetwas beitragen kann, bitte melden: (App tracen = App verfolgen?)

jhejmann, 2023-02-28:
Habe es gestern mit Postman probiert (mein erster API trace versuch) bei myAudi bekam ich nur Fehler. Vermute er hat das Root Zertifikat nicht animmt, bei Senec z.B. hat https: REST Calls tracen wunderbar geklappt. 
Wenn mir einer einen Hint zum Tracen gibt, schaue ich mir das später an....

sahomm, 2023-02-28:
Bei meinem Hybrid Audi leider auch keine Verbindung möglich.

Hier mal der gesamte Auszug des Debug:
```
[main  ] INFO 2023/02/28 17:44:58 evcc 0.113.2
[main  ] INFO 2023/02/28 17:44:58 using config file: /etc/evcc.yaml
[db    ] INFO 2023/02/28 17:44:58 using sqlite database: /root/.evcc/evcc.db
[eebus ] INFO 2023/02/28 17:44:58 Local SKI:  d42cedxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
[eebus ] DEBUG 2023/02/28 17:44:58 mdns: announce
[eebus ] DEBUG 2023/02/28 17:44:58 mdns: using avahi
[eebus ] DEBUG 2023/02/28 17:44:58 starting websocket server on :4712
[eebus ] DEBUG 2023/02/28 17:44:58 error during websocket server starting: listen tcp :4712: bind: address already in use
[main  ] ERROR 2023/02/28 17:45:00 creating vehicle ev2 failed: cannot create vehicle 'template': cannot create vehicle 'audi': unexpected status: 404
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'audi': unexpected status: 404
Capacity: 0.0kWh
[eebus ] DEBUG 2023/02/28 17:45:03 mdns: stop announcement
```

Mich macht die Fehlermeldung das die Adresse in Nutzung sei stutzig.

thierolm, 2023-02-28:
Lass das mal mit `evcc -l trace` laufen.

sahomm, 2023-02-28:
Hier die Ausgabe des Trace als TXT, da die Ausgabe sehr lang ist:

[Trace EVCC Audi.txt](https://github.com/evcc-io/evcc/files/10853057/Trace.EVCC.Audi.txt)

andig, 2023-02-28:
Bitte off-topic vermeiden. Bzgl. Audi und MITM wird es siw nur funktionieren, wenn man das Certificate Pinning z.B. aus der Android App entfernt. Leider nicht mein Skill...

Ham60elu, 2023-03-15:
HA Integration:
Auch mein Fiat 500e wird seit dem Update auf 0.113.2 (und auch bei den weiteren Versionen auf die aktuelle 0.114.1) nur noch als "unavailable" angezeigt. Die Anmeldung in der Fiat-App funktioniert mit den eingetragenen Credentials. 
Mit dem Update auf die 0.113.2 funtionierten bei mir auch die Einträge in der secrets.yaml (sponsortoken, user, password, vim und pin) für evcc nicht mehr (=evcc startete nicht mehr). Erst nachdem ich diese wieder orginal in die evcc.yaml eingetragen habe, funktionierte evcc wieder.

andig, 2023-03-15:
@Ham60elu hat nix mit VAG zu tun

merlinste, 2023-03-16:
Ich bekomme seit heute folgende Fehlermeldung:

`vehicle soc: Get \"https://emea.bff.cariad.digital/vehicle/v1/vehicles/WVGZZZXXX/selectivestatus?jobs=all\": unexpected status: 400`

App und Webseite funktionieren.

andig, 2023-03-16:
Geht bei mir. Temporäres Problem? Sonst bitte neues Thema und Zugangsdaten an info@evcc.io

andig, 2023-03-16:
Siehe auch https://openwb.de/forum/viewtopic.php?p=82377#p82377 - evtl. wieder neue Terms & Conditions?

merlinste, 2023-03-16:
Danke dir! Nach einem evcc-Neustart läuft es jetzt wieder! Hatte auch keine neuen T&Cs zu akzeptieren. Naja, VW eben :)

IonasElate, 2023-03-29:
Hallo

Ich habe das gleiche Problem …

`
evcc         | [main  ] ERROR 2023/03/29 11:39:54 creating vehicle ev4 failed: cannot create vehicle 'template': cannot create vehicle 'id': missing state
evcc         | [site  ] INFO 2023/03/29 11:39:54 site config:
evcc         | [site  ] INFO 2023/03/29 11:39:54   meters:      grid ✓ pv ✓ battery ✓
evcc         | [site  ] INFO 2023/03/29 11:39:54     grid:      power ✓ energy ✗ currents ✗
evcc         | [site  ] INFO 2023/03/29 11:39:54     pv 1:      power ✓ energy ✗ currents ✗
evcc         | [site  ] INFO 2023/03/29 11:39:54     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
evcc         | [site  ] INFO 2023/03/29 11:39:54   vehicles:
evcc         | [site  ] INFO 2023/03/29 11:39:54     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
evcc         | [lp-1  ] INFO 2023/03/29 11:39:54 loadpoint 1:
evcc         | [lp-1  ] INFO 2023/03/29 11:39:54   mode:        pv
evcc         | [lp-1  ] INFO 2023/03/29 11:39:54   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
evcc         | [lp-1  ] INFO 2023/03/29 11:39:54   meters:      charge ✓
evcc         | [lp-1  ] INFO 2023/03/29 11:39:54     charge:    power ✓ energy ✓ currents ✓
evcc         | [lp-1  ] INFO 2023/03/29 11:39:54 vehicle updated: unknown -> unavailable
evcc         | [lp-1  ] INFO 2023/03/29 11:39:55 car connected
evcc         | [lp-1  ] ERROR 2023/03/29 11:39:55 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'id': missing state`

Habe auf das neueste Nightly-Build gewechselt aber immer noch den gleichen Fehler.
Ist ein eGolf - es hat bei mir noch nie funktioniert mit dem Auto - ich habe es letzte Woche erst als Gebrauchtwagen gekauft.
Service Plus hab ich gekauft und in der App und auf der Website sehe ich alle Daten …

Meine Config:

`vehicles:
- name: ev4
  type: template
  template: id
  title: eGolf
  user: myemail@mel.de
  password: 123456789
  vin: WVWZZZsdfasefasefasef
  capacity: 32 
  mode: pv  # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll
  minSoC: 0
  phases: 2
  targetSoC: 80
  minCurrent: 6
  maxCurrent: 16  `

StefanSchoof, 2023-03-29:
Ich hatte die Meldung auch. Nachdem ich auf mich auf My Volkswagen angemeldet habe und die Meldung zur personalisierten Werbung weggeklickt habe, geht es wieder.

IonasElate, 2023-03-29:
Hallo
So eine Nachricht kommt bei mir gar nicht …
[image]
Ist das so richtig?

Lex318, 2023-12-09:
Bei mir hat es jetzt auch geholfen, die Mitteilung zur personalisierten Werbung (myvolkswagen.net) im Browser einmal zu bestätigen und dann EVCC neuzustarten. ID4 wieder online.