evcc-io/evcc #6499: DaheimLaden Modbus invalid phase wiring

VolkerK62, 2023-02-27:
### Describe the bug

Gelegentlich, aber dann immer im Zusammenhang mit dem Start der Ladung bekomme ich die einmalige Warnung
`invalid phase wiring between charge meter and charger`.

Dadurch wechseln dann auch kurzfristig die `detected phases` von `1` auf `3`. Das hat aber gefühlt keine Auswirkungen, da es nur ein Intervall betrifft.  

Siehe Log: 11:25:42 und 12:59:12

Die Wallbox ist per LAN mit einem Fritz Powerline Adapter verbunden.

### Steps to reproduce

Kann leider nicht provoziert werden, tritt nicht immer beim Ladestart auf. 

### Configuration details

```yaml
chargers:
- name: wallbox
  type: template
  template: daheimladen-mb
  host: 192.168.178.26
  port: 502
```

### Log details

```text
Feb 27 11:25:22 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:22 charger enable
Feb 27 11:25:22 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:22 wake-up timer: start
Feb 27 11:25:31 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:31 ----
Feb 27 11:25:31 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:31 charge power: 0W
Feb 27 11:25:31 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:31 pv power: 6854W
Feb 27 11:25:31 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:31 battery soc: 44%
Feb 27 11:25:31 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:31 battery power: -1183W
Feb 27 11:25:32 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:32 grid power: -808W
Feb 27 11:25:32 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:32 grid powers: [1223 -113 -1918]W
Feb 27 11:25:32 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:32 grid currents: [5.36 -1.29 -8.21]A
Feb 27 11:25:32 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:32 aux power 0: 0W
Feb 27 11:25:32 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:32 site power: -1892W
Feb 27 11:25:32 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:32 charge voltages: [236 0 0]V
Feb 27 11:25:32 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:32 detected connected phases: 1p
Feb 27 11:25:32 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:32 charge currents: [0 0 0]A
Feb 27 11:25:32 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:32 charger status: B
Feb 27 11:25:32 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:32 next soc poll remaining time: 57m49s
Feb 27 11:25:32 raspberrypi evcc[583]: [lp-1  ] WARN 2023/02/27 11:25:32 charger out of sync: expected enabled, got disabled
Feb 27 11:25:32 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:32 pv charge current: 8.22A = 0A + 8.22A (-1892W @ 1p)
Feb 27 11:25:32 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:32 pv timer reset
Feb 27 11:25:32 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:32 pv timer inactive
Feb 27 11:25:32 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:32 max charge current: 8A
Feb 27 11:25:41 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:41 ----
Feb 27 11:25:41 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:41 charge power: 0W
Feb 27 11:25:41 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:41 pv power: 6846W
Feb 27 11:25:41 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:41 battery soc: 44%
Feb 27 11:25:41 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:41 battery power: -1191W
Feb 27 11:25:41 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:41 grid power: -830W
Feb 27 11:25:42 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:42 grid powers: [1212 -131 -1914]W
Feb 27 11:25:42 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:42 grid currents: [5.31 -1.34 -8.18]A
Feb 27 11:25:42 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:42 aux power 0: 0W
Feb 27 11:25:42 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 11:25:42 site power: -1921W
Feb 27 11:25:42 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:42 charge voltages: [236 83.2 235]V
Feb 27 11:25:42 raspberrypi evcc[583]: [lp-1  ] WARN 2023/02/27 11:25:42 invalid phase wiring between charge meter and charger
Feb 27 11:25:42 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:42 detected connected phases: 3p
Feb 27 11:25:42 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:42 charge currents: [0 0 0]A
Feb 27 11:25:42 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:42 charger status: C
Feb 27 11:25:42 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:42 charge start energy: 173kWh
Feb 27 11:25:42 raspberrypi evcc[583]: [lp-1  ] INFO 2023/02/27 11:25:42 start charging ->
Feb 27 11:25:51 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:51 vehicle status: C
Feb 27 11:25:51 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 11:25:51 vehicle soc: 80%
[... 63 lines cut ...]
Feb 27 12:59:20 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 12:59:20 max charge current: 10A
Feb 27 12:59:21 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 12:59:21 ----
Feb 27 12:59:21 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 12:59:21 charge power: 2100W
Feb 27 12:59:21 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 12:59:21 pv power: 6813W
Feb 27 12:59:21 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 12:59:21 battery soc: 55%
Feb 27 12:59:21 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 12:59:21 battery power: -1165W
Feb 27 12:59:21 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 12:59:21 grid power: -2481W
Feb 27 12:59:21 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 12:59:21 grid powers: [-911 -2140 571]W
Feb 27 12:59:22 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 12:59:22 grid currents: [-4.13 -9.27 2.24]A
Feb 27 12:59:22 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 12:59:22 aux power 0: 0W
Feb 27 12:59:22 raspberrypi evcc[583]: [site  ] DEBUG 2023/02/27 12:59:22 site power: -3546W
Feb 27 12:59:22 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 12:59:22 charge voltages: [230 0 0]V
Feb 27 12:59:22 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 12:59:22 detected connected phases: 1p
Feb 27 12:59:22 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 12:59:22 charge currents: [9.4 0 0]A
Feb 27 12:59:22 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 12:59:22 detected active phases: 1p
Feb 27 12:59:22 raspberrypi evcc[583]: [lp-1  ] DEBUG 2023/02/27 12:59:22 charger status: C
```

### What type of operating system are you running?

Linux

### Version

0.113.2 (742b2bea)

premultiply, 2023-02-27:
Keine Idee was wir dagegen tun können wenn die Box diese Messwerte liefert.

premultiply, 2023-02-27:
@stephanschwebe Irgendeine Idee was da passiert?
Die Box ist nur einphasig angeschlossen.

andig, 2023-03-08:
ping @stephanschwebe

stephanschwebe, 2023-03-08:
@andig spoke with Volker by phone. Problem comes occasionally, but not in a manner to be reproduced. We will distribute to  Volker a new firmware next week, run some tests ...

VolkerK62, 2023-03-08:
ich mach hier mal zu. Werde es im Auge behalten.

closed 2023-03-08 by VolkerK62

andig, 2023-03-09:
@stephanschwebe Herausragender Support- vielen Dank!