evcc-io/evcc #6503: vehicle soc: vehicle not available: cannot create vehicle 'template':  cannot create vehicle 'id': form not found

AES-256-GCM, 2023-02-27:
### Describe the bug

Since today in the afternoon I get these error messages when connecting my ID.4:
```
[lp-1  ] INFO 2023/02/27 22:52:13 car connected                                                          
[lp-1  ] ERROR 2023/02/27 22:52:13 vehicle soc: vehicle not available: cannot create vehicle 'template': 
cannot create vehicle 'id': form not found                                                               
[lp-1  ] ERROR 2023/02/27 22:57:28 vehicle soc: vehicle not available: cannot create vehicle 'template': 
cannot create vehicle 'id': form not found                                                               
[lp-1  ] ERROR 2023/02/27 23:02:43 vehicle soc: vehicle not available: cannot create vehicle 'template': 
cannot create vehicle 'id': form not found                                                               
[lp-1  ] ERROR 2023/02/27 23:07:58 vehicle soc: vehicle not available: cannot create vehicle 'template': 
cannot create vehicle 'id': form not found
```

The current SOC isn't updated in the Web UI and my ID.4 is shown as `unavailable`.

### Steps to reproduce

1. Connect ID.4 with Wallbox and open evcc log

### Configuration details

```yaml
vehicles:
    - name: id4
      title: ID.4
      type: template
      template: id
      user: <redacted>
      password: <redacted>
      vin: WV<redacted>
      capacity: 77
      cache: 5m
      mode: pv
      minSoC: 30
      targetSoC: 80
```

### Log details

```text
[main  ] INFO 2023/02/27 23:16:48 evcc 0.113.2                                                           
[main  ] INFO 2023/02/27 23:16:48 using config file: /etc/evcc.yaml                                      
[main  ] INFO 2023/02/27 23:16:48 starting ui and api at :7070                                           
[db    ] INFO 2023/02/27 23:16:48 using sqlite database: /root/.evcc/evcc.db                             
[main  ] ERROR 2023/02/27 23:16:49 creating vehicle id4 failed: cannot create vehicle 'template': cannot 
create vehicle 'id': form not found                                                                      
[lp-1  ] WARN 2023/02/27 23:16:49 poll mode '{connected 5m0s}' may deplete your battery or lead to API mi
suse. USE AT YOUR OWN RISK.                                                                              
[lp-1  ] WARN 2023/02/27 23:16:49 poll interval '5m0s' is lower than 1h0m0s and may deplete your battery 
or lead to API misuse. USE AT YOUR OWN RISK.                                                             
[lp-1  ] WARN 2023/02/27 23:16:49 phases not configured, assuming 3p                                     
[site  ] INFO 2023/02/27 23:16:49 site config:                                                           
[site  ] INFO 2023/02/27 23:16:49   meters:      grid ✓ pv ✓ battery ✗                                   
[site  ] INFO 2023/02/27 23:16:49     grid:      power ✓ energy ✓ currents ✗                             
[site  ] INFO 2023/02/27 23:16:49     pv 1:      power ✓ energy ✓ currents ✗                             
[site  ] INFO 2023/02/27 23:16:49   vehicles:                                                            
[site  ] INFO 2023/02/27 23:16:49     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗            
[lp-1  ] INFO 2023/02/27 23:16:49 loadpoint 1:                                                           
[lp-1  ] INFO 2023/02/27 23:16:49   mode:        pv                                                      
[lp-1  ] INFO 2023/02/27 23:16:49   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗           
[lp-1  ] INFO 2023/02/27 23:16:49   meters:      charge ✓                                                
[lp-1  ] INFO 2023/02/27 23:16:49     charge:    power ✓ energy ✓ currents ✓                             
[lp-1  ] DEBUG 2023/02/27 23:16:49 phase timer inactive                                                  
[lp-1  ] DEBUG 2023/02/27 23:16:49 pv timer inactive                                                     
[lp-1  ] INFO 2023/02/27 23:16:49 vehicle updated: unknown -> unavailable                                
[site  ] DEBUG 2023/02/27 23:16:49 ----                                                                  
[lp-1  ] DEBUG 2023/02/27 23:16:49 charge power: 2W                                                      
[site  ] DEBUG 2023/02/27 23:16:49 pv power: 0W                                                          
[site  ] DEBUG 2023/02/27 23:16:49 grid power: 311W                                                      
[site  ] DEBUG 2023/02/27 23:16:49 site power: -1789W                                                    
[lp-1  ] DEBUG 2023/02/27 23:16:49 charge currents: [0.03 0 0]A                                          
[lp-1  ] DEBUG 2023/02/27 23:16:49 charger status: B                                                     
[lp-1  ] DEBUG 2023/02/27 23:16:49 charge start energy: 1.36e+03kWh                                      
[lp-1  ] INFO 2023/02/27 23:16:49 car connected                                                          
[lp-1  ] DEBUG 2023/02/27 23:16:49 pv timer elapse                                                       
[lp-1  ] DEBUG 2023/02/27 23:16:49 pv timer inactive                                                     
[lp-1  ] ERROR 2023/02/27 23:16:49 vehicle soc: vehicle not available: cannot create vehicle 'template': 
cannot create vehicle 'id': form not found                                                               
[lp-1  ] DEBUG 2023/02/27 23:16:49 pv charge current: 2.59A = 0A + 2.59A (-1789W @ 3p)                   
[lp-1  ] DEBUG 2023/02/27 23:16:49 pv enable timer reset                                                 
[lp-1  ] DEBUG 2023/02/27 23:16:49 pv timer inactive                                                     
[site  ] DEBUG 2023/02/27 23:17:24 ----                                                                  
[lp-1  ] DEBUG 2023/02/27 23:17:24 charge power: 2W                                                      
[site  ] DEBUG 2023/02/27 23:17:24 pv power: 0W                                                          
[site  ] DEBUG 2023/02/27 23:17:24 grid power: 311W                                                      
[site  ] DEBUG 2023/02/27 23:17:24 site power: -1789W                                                    
[lp-1  ] DEBUG 2023/02/27 23:17:24 charge currents: [0.03 0 0]A                                          
[lp-1  ] DEBUG 2023/02/27 23:17:24 charger status: B                                                     
[lp-1  ] DEBUG 2023/02/27 23:17:24 next soc poll remaining time: 4m25s                                   
[lp-1  ] DEBUG 2023/02/27 23:17:24 pv charge current: 2.59A = 0A + 2.59A (-1789W @ 3p)
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.113.2

AES-256-GCM, 2023-02-27:
Seems other users are having this issue as well, see https://github.com/evcc-io/evcc/issues/6495

thebrainkafka, 2023-02-27:
Das scheint der neue login link zu sein.
Aus einem anderen Forum entnommen (meinID Forum)

https://identity.vwgroup.io/oidc/v1 https://emea.bff.cariad.digital/user-login/v1/authorize

closed 2023-02-28 by andig