evcc-io/evcc #6522: Audi IDK Proxy Fehler

andig, 2023-02-28:
Same as https://github.com/TA2k/ioBroker.vw-connect/issues/236, https://github.com/arjenvrh/audi_connect_ha/issues/168

basilfx, 2023-02-28:
I've been able to make it work (for my own custom implementation, non-evcc). I'll share some details:

- AUDI app resolves https://content.app.my.audi.com/service/mobileapp/configurations/market/DE/de?v=4.15.0
- Look for the value of `idkLoginServiceConfigurationURLProduction`. It currently is https://emea.bff.cariad.digital/login/v1/idk/openid-configuration. It was https://idkproxy-service.apps.emea.vwapps.io/v1/emea/token. I was able to sent the same request.
- AAZS Proxy token URL changed to https://emea.bff.cariad.digital/login/v1/audi/token. It was https://aazsproxy-service.apps.emea.vwapps.io/token. I was able to sent the same request.

I've been using the QMAuth details from [here](https://github.com/arjenvrh/audi_connect_ha/commit/314dab7fda775bc3814dc36f2b44edf3f73600c8).

andig, 2023-02-28:
Great that, that should help! Any hint how you‘ve actually accomplished this feat?

andig, 2023-02-28:
Closed in https://github.com/evcc-io/evcc/commit/5bdc578187b9cfe5a197cddff242906a03515db9. MUCH APPRECIATED; thank you so much!

closed 2023-02-28 by andig

sveba, 2023-03-01:
@andig can you pls create a release with this fix. Thanx

andig, 2023-03-01:
We release when ready ;)

andig, 2023-03-01:
https://github.com/evcc-io/evcc/releases/tag/0.114.0. To speed up the process and save efforts: it would be really nice to wait for the release. As usual, we do release when ready. Thank you.