evcc-io/evcc #6531: Tibber Pulse - grid meter: timeout nach DSL Reconnect

dom881, 2023-02-28:
### Describe the bug

Nach einem DSL Reconnect werden keine Daten mehr vom Tibber in evcc empfangen.
Der angezeigt Netzbezug bleibt konstant beim letzten Wert, der vor dem DSL-Reconnect 
empfangen wurde.

### Steps to reproduce

1. Tibber Pulse als grid Meter in yaml konfiguriert
2. Neustart evcc
3. Werte werden in evcc korrekt vom Tibber Pulse angezeigt
4. FritzBox führt einen DSL Re-connect durch - Dauer ca. 3min
5. grid meter: timeout - Fehler in evcc 
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: xxx

interval: 10s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554

telemetry: false

meters:
- type: template
  template: tibber-pulse
  usage: grid
  token: xxx
  name: grid1
- type: template
  template: sma-inverter 
  usage: pv  
  host: xxx  
  password: xxx 
  name: pv2

chargers:
- type: template
  template: easee 
  user: xxx
  password: xxx  
  charger: xxx
  name: wallbox4

vehicles:
- type: template
  template: opel 
  title: Opel Corsa-e  
  user: xxx
  password: xxx
  capacity: 50  
  phases: 3  
  icon: car  
  cache: 5m  
  mode: MinPV  
  minSoc: 35  
  targetSoc: 85  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev3

loadpoints:
- title: Garage
  charger: wallbox4
  vehicle: ev3
  mode: minpv
  soc:
    poll:
      mode: connected
      interval: 5m
    estimate: true
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

tariffs:
   grid:
    # either
    type: tibber
    cheap: 0.2 # [currency]/kWh
    token: "xxx" # access token
   feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.083 # [currency]/kWh

site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv2
```

### Log details

```text
[main  ] INFO 2023/02/28 23:05:52 evcc 0.113.2
[main  ] INFO 2023/02/28 23:05:52 using config file: /etc/evcc.yaml
[main  ] INFO 2023/02/28 23:05:52 starting ui and api at :7070
[db    ] INFO 2023/02/28 23:05:58 using sqlite database: /home/pi/.evcc/evcc.db
[main  ] FATAL 2023/02/28 23:06:03 cannot create meter 'grid1': cannot create meter 'template': cannot create meter 'tibber-pulse': Post "https://api.tibber.com/v1-beta/gql": dial tcp: lookup api.tibber.com: i/o timeout
[main  ] FATAL 2023/02/28 23:06:03 will attempt restart in: 5m0s
[server] FATAL 2023/02/28 23:06:03 unlinkat /tmp/evcc: operation not permitted
```

### What type of operating system are you running?

Linux

### Version

0.113.2

naltatis, 2023-03-17:
Ich beobachte hier das gleiche Problem. Lässt sich bei einer lokal laufenden evcc Instanz auch gut nachstellen. Nach dem ersten Start läuft alles super. Trennt man den Rechner kurz vom Netz wird die Verbindung nicht wie erwartet neu aufgebaut. Stattdessen wir ab dem Zeitpunkt nur noch das hier gemeldet:

```
[site  ] ERROR 2023/03/17 09:14:52 grid meter: timeout
```

Logging ist leider nicht hilfreich. Ich habe die ganze Anwendung auf `trace` laufen lassen. `pulse` Einträge tauche nur auf wenn erfolgreich Daten empfangen wurden. Keine Fehlermeldungen.

andig, 2023-03-17:
@naltatis magst Du mal hier https://github.com/hasura/go-graphql-client/issues/64 kommentieren oder ein neues Issue aufmachen?

naltatis, 2023-03-17:
@andig Sieht super aus. Mit dem aktuellen Master Stand bekomme ich nen sauberen Reconnect.

closed 2023-03-17 by andig