evcc-io/evcc #6532: Tesla Wallconnector V3 zeigt keine "Leistung" an

fynnsh, 2023-02-28:
### Describe the bug

Hallo,

ich habe einen Tesla WC3 und diesen in evcc eingebunden. Leider zeigt dieser beim Aufladen von Fremdfahrzeugen keine Leistung an sondern nur die geladenen KW und die Dauer. Richtig ist das die Wallbox selbst über die json API keine direkten Watt ausgibt. Es wäre aber sehr cool wenn ihr das noch hinzufügen könntet. 

Die Leistung lässt sich mit den Daten aus der API folgendermaßen berechnen.
`(currentA_a + currentB_a + currentC_a) * grid_v`

### Steps to reproduce

TeslaWallbox einbinden
[image]

### Configuration details

```yaml
chargers:
- type: template
  template: twc3
  host: tesla-wallconnector-003
  name: wallbox2
```

### Log details

```text
[main  ] INFO 2023/02/28 23:34:27 evcc 0.113.2
```

### What type of operating system are you running?

Linux

andig, 2023-03-01:
@premultiply könnten wir machen, wäre dann aber Scheinleistung. Das tut vmtl. mehr weh als das es hilft?

fynnsh, 2023-03-01:
Im Grunde kann das Ergebnis etwas verfälscht werden je nachdem ob auf einer Phase weniger Spannung ist. Aber im praktischen Beispiel war die Berechnung immer genau.

[image]

andig, 2023-03-01:
Liegt vmtl. daran, dass Dein Auto keine Scheinleistung produziert. Bei der Zoe sieht das anders aus.

premultiply, 2023-03-01:
Kann man machen.
Ist z.B. bei ABL nicht anders. Da werden nur Ströme gemessen und 230V angenommen.

fynnsh, 2023-03-01:
@andig  der Ladevorgang von 19 bis 0 Uhr ist tatsächlich ein Zoe der von 10 bis 14 ein Dacia Spring

closed 2023-03-02 by andig