evcc-io/evcc #6533: cannot create charger

arteck, 2023-03-01:
### Describe the bug

nach installation von 0.113.2 startet evcc nicht mehr mit der fehlermeldung
```
 evcc -c /etc/evcc.yaml
[main  ] INFO 2023/03/01 07:22:41 evcc 0.112.0
[main  ] INFO 2023/03/01 07:22:41 using config file: /etc/evcc.yaml
[main  ] INFO 2023/03/01 07:22:41 starting ui and api at :7070
[main  ] FATAL 2023/03/01 07:22:41 cannot create charger 'wallbox5': cannot create charger 'template': cannot create charger 'go-e': sponsorship required, see https://github.com/evcc-io/evcc#sponsorship
[main  ] FATAL 2023/03/01 07:22:41 will attempt restart in: 5m0s
```

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070

network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error
  db: error

# unique installation id
plant: c5b0b9xx

interval: 10s # control cycle interval

sponsortoken: eyJhbGciOiJIxx
telemetry: false

meters:
- type: template
  template: solarwatt 
  usage: grid  
  host: 192.168.1.252  
  name: grid1
- type: template
  template: solarwatt 
  usage: pv  
  host: 192.168.1.252  
  name: pv2
- type: template
  template: solarwatt 
  usage: pv  
  host: 192.168.1.252  
  name: pv3

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.1.3  
  name: wallbox5

vehicles:
- type: template
  template: kia 
  title: kia  
  user: xxx  
  password: 'xxx'  
  vin: Uxx
  capacity: 13.8 
  phases: 1
  language: de  
  name: sportage

loadpoints:
- title: Garage
  charger: wallbox5
  mode: minpv
  phases: 3
  vehicle: sportage
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pvs:
    - pv2
    - pv3
```

### Log details

```text
[main  ] INFO 2023/03/01 07:31:11 evcc 0.113.2
[main  ] INFO 2023/03/01 07:31:11 using config file: /etc/evcc.yaml
[main  ] INFO 2023/03/01 07:31:11 starting ui and api at :7070
[main  ] FATAL 2023/03/01 07:31:11 cannot create charger 'wallbox5': cannot create charger 'template': cannot create charger 'go-e': sponsorship required, see https://github.com/evcc-io/evcc#sponsorship
[main  ] FATAL 2023/03/01 07:31:11 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.113.2

arteck, 2023-03-01:
alle Geräte sind erreichbar..

andig, 2023-03-01:
Stell dir bitte mal ein neues Token aus- ich vermute, Deins ist abgelaufen. Falls das nicht hilfe bitte info@evcc.io kontaktieren.

closed 2023-03-01 by andig

arteck, 2023-03-01:
kann ich nicht bin angeblich kein sponsor