evcc-io/evcc #6545: Tibber Börsenstromdaten wurden nicht übernommen

steve0564, 2023-03-01:
### Describe the bug

Die Tibber Börsenstromdaten des nächsten Tages (Erscheinzeit: ab 13 Uhr) wurden heute nicht automatisiert übernommen.
Mit evcc tariff wurden die Daten für morgen jedoch angezeigt.
Erst nach einem Restart wurden die Daten dann, nachdem eine Zielladezeit neu konfiguriert wurde, übernommen.

### Steps to reproduce

1. keine, da nur einmal / Tag möglich
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ***

interval: 10s # control cycle interval

sponsortoken: ***
# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.178.57  
  port: 502  
  name: grid1
- type: template
  template: victron-energy 
  usage: pv  
  host: 192.168.178.57  
  port: 502  
  name: pv2
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.178.57  
  port: 502  
  name: battery3

chargers:
- type: template
  template: easee 
  user: ***
  password: ***
  charger: E***
  name: wallbox5
- type: template
  template: fritzdect 
  uri: https://192.168.178.2  
  user: ***
  password: ***
  ain: 11***
  standbypower: 15  
[... 283 lines cut ...]
Mär 01 13:59:03 raspberrypi evcc[5290]: [site  ] DEBUG 2023/03/01 13:59:03 battery soc: 95%
Mär 01 13:59:03 raspberrypi evcc[5290]: [site  ] DEBUG 2023/03/01 13:59:03 battery power: 0W
Mär 01 13:59:03 raspberrypi evcc[5290]: [site  ] DEBUG 2023/03/01 13:59:03 grid power: 313W
Mär 01 13:59:03 raspberrypi evcc[5290]: [site  ] DEBUG 2023/03/01 13:59:03 site power: 213W
Mär 01 13:59:03 raspberrypi evcc[5290]: [lp-1  ] DEBUG 2023/03/01 13:59:03 charge currents: [0.009 0.008 0.008]A
Mär 01 13:59:03 raspberrypi evcc[5290]: [lp-1  ] DEBUG 2023/03/01 13:59:03 charger status: B
Mär 01 13:59:03 raspberrypi evcc[5290]: [lp-1  ] DEBUG 2023/03/01 13:59:03 planned 3h27m50s until 2023-03-02 05:10:00 +0100 CET at 7360W: total plan duration: 3h27m50s, avg cost: 0.287
Mär 01 13:59:03 raspberrypi evcc[5290]: [lp-1  ] DEBUG 2023/03/01 13:59:03 plan too short- ignoring remaining 3h27m50s
Mär 01 13:59:03 raspberrypi evcc[5290]: [lp-1  ] DEBUG 2023/03/01 13:59:03 available power -213W < 2760W min 2p threshold
Mär 01 13:59:03 raspberrypi evcc[5290]: [lp-1  ] DEBUG 2023/03/01 13:59:03 phase scale1p in 4m0s
Mär 01 13:59:03 raspberrypi evcc[5290]: [lp-1  ] DEBUG 2023/03/01 13:59:03 pv charge current: 0A = 0A + -0.463A (213W @ 2p)
Mär 01 13:59:03 raspberrypi evcc[5290]: [lp-1  ] DEBUG 2023/03/01 13:59:03 climater active: false, target temp: 23.5°C, outside temp: 0.0°C
Mär 01 13:59:13 raspberrypi evcc[5290]: [site  ] DEBUG 2023/03/01 13:59:13 ----
Mär 01 13:59:13 raspberrypi evcc[5290]: [lp-1  ] DEBUG 2023/03/01 13:59:13 charge power: 0W
Mär 01 13:59:13 raspberrypi evcc[5290]: [lp-2  ] DEBUG 2023/03/01 13:59:13 charge power: 0W
Mär 01 13:59:13 raspberrypi evcc[5290]: [site  ] DEBUG 2023/03/01 13:59:13 pv power: 192W
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.113.2 (5bdc5781)

steve0564, 2023-03-01:
Tibber Service? Wie logge ich den?
Ich hatte das manuell festgestellt, dass die Daten nicht übernommen wurden.

andig, 2023-03-01:
Logging siehe Doku- ich mache zu bis es neue Erkenntnisse gibt die das Problem nachvollziehbar machen.

closed 2023-03-01 by andig

steve0564, 2023-03-09:
nicht mehr nachvollziehbar!