evcc-io/evcc #6550: charged energy is not correct displayed in charging history

TMoersch, 2023-03-01:
### Describe the bug

Since the last update the  charged energy is not longer correct displayed in charging history. During the charging session at at the end of the charging session the shown value of kWh charged is correct, only in the history is shown 0.1 kwH??
SW Version v0.113.2
Vestel EVC04
..reboot did'nt help
Thank you

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

# data via modbusproxy  192.168.2.xyz:502

log: info
levels:
  cache: error

# unique installation id
plant: 122b825f8e761bfd0c2424b768455521d011e96b1ffe8fe441b08f548e38f794

interval: 10s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3NjgxMjMzNDAsImlhdCI6MTY3MzUxNTM0MCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6IlRNb2Vyc2NoIn0.x2kN90rhdpFszuaEchGMx6WThrjIi-7vsNVzWn6QTys

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.2.xyz
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.2.xyz
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.2.xyz
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3

chargers:
- type: template
  template: eon-vbox
  id: 255
  host: 192.168.2.xxx
  port: 502
  modbus: tcpip
  name: wallbox5

vehicles:
- type: template
  template: hyundai
  title: IONIQ5
  user: <user>
  password: <pw>
  vin: <vin>
  capacity: 72.6
  language: de
  name: ev4
  targetSoC: 70

loadpoints:
- title: Garage
  charger: wallbox5
  mode: now
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3
```

### Log details

```text
no log's available
in my case evcc is an addon to HomeAssistant
```

### What type of operating system are you running?

Docker container

### Version

v0.113.2

andig, 2023-03-01:
Leider nicht nachvollziehbar

closed 2023-03-01 by andig

premultiply, 2023-03-01:
Sichtung der Ladedaten von `evcc charger` kann ggf. validieren ob die Daten von der Box richtig ausgewertet werden.