evcc-io/evcc #6553: Priorität von loadpoint wird nicht beachtet

nicx, 2023-03-01:
### Describe the bug

Ich habe 2 Loadpoints mit unterschiedlicher Priorität. Jeweils angeschlossen ein Auto unterhalb des targetSOC. Starte ich nun EVCC mit default Mode "PV", dann springen beide Loadpoints (Überschuss vorausgesetzt) auf "starte Ladevorgang in 3m". Je nachdem, welcher Loadpoint zuerst ein Fahrzeug erkannt hat, gewinnt dieser erst einmal. Der Überschuss reicht aktuell aber nur für 1 Loadpoint... damit wird der priorisierte Loadpoint nicht geladen und dessen Startvorgang bricht wieder ab.

Ich hätte erwartet das wenn der Überschuss nur für 1 Loadpoint ausreicht, dass dann auch nur der Loadpoint gestartet wird, der die höhere Priorität hat.

Das Spielchen geht dann recht häufig hin und her... da immer wieder die Timer beider Loadpoints loslaufen fürs starten/stoppen, da auch immer wieder der "freie" Überschuss ein anderer ist, je nachdem ob der nicht-priorisierte Loadpoint gerade gewonnen hat, oder auch nicht.

Dazu kommt noch meine Hausbatterie, die auch immer wieder in diesem Spiel mit einbezogen wird... selbst wenn sie ihre 100% prioritySOC konfiguriert und erreicht hat, wird sie immer wieder um wenige Prozent entladen und wieder beladen (trotz bufferSOC von ebenfalls 100%).

Mein Verständnis bzw. Wunsch wäre bei vorhandenem PV Überschuss und Modus "PV":
- Zuerst wird die Batterie auf 100% geladen. Sobald dies erledigt ist, ist die Batterie komplett aus dem Spiel.
- Danach wir der Loadpoint mit höherer Priorität aktiviert und bis zum targetSOC das Fahrzeugs geladen, danach ist dieser ebenfalls komplett aus dem Spiel
- Als letztes wird der Loadpoint mit niedrigerer Priorität beladen, ebenfalls bis zum targetSOC des Fahrzeugs.
- Sollte kein PV Überschuss vorhanden sein, wird weder Batterie, noch einer der Loadpoints beladen. Sollte Überschuss wieder da sein gehts einfach in der obigen Reihenfolge wieder los.

Ist das so simpel nicht abbildbar? Oder habe ich etwas falsch konfiguriert? Es kann natürlich auch sein, das ich das eine oder andere Feature nicht wirklich verstanden habe... dann freue ich mich sehr über Aufklärung :)

### Steps to reproduce

-

### Configuration details

```yaml
vehicles:
  - name: Opel Mokka
    type: template
    template: opel
    title: Opel Mokka
    user: *
    password: *
    capacity: 50
    mode: pv
    minSoc: 20
    targetSoc: 90
  - name: Tesla Model Y
    type: template
    template: tesla
    title: Tesla Model Y
    accessToken: *
    refreshToken: *
    capacity: 77
    mode: pv
    minSoc: 20
    targetSoc: 80

site:
  title: Zuhause
  meters:
    grid: Huawei DTSU666
    pv: Huawei SUN2000
    battery: Huawei LUNA2000
  residualPower: 100
  prioritySoC: 100 # give home battery priority up to this soc (empty to disable)
  bufferSoC: 100 # ignore home battery discharge above soc (empty to disable)

loadpoints:
  - title: Stellplatz # display name for UI
    charger: Easee Home Stellplatz # charger
    mode: pv
    vehicle: Tesla Model Y # set default vehicle (disables vehicle detection)
    resetOnDisconnect: true # set defaults when vehicle disconnects
    phases: 0  # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    minCurrent: 6 # minimum charge current (default 6A)
    maxCurrent: 16 # maximum charge current (default 16A)
    soc:
      poll:
        mode: charging
        interval: 60m
      estimate: true
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 5m # switch charger contactor not more often than this (default 5m)
    priority: 0
  - title: Garage # display name for UI
    charger: Easee Home Garage # charger
    mode: pv
    vehicle: Opel Mokka # set default vehicle (disables vehicle detection)
    resetOnDisconnect: true # set defaults when vehicle disconnects
    phases: 0  # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    minCurrent: 6 # minimum charge current (default 6A)
    maxCurrent: 16 # maximum charge current (default 16A)
    soc:
      poll:
        mode: charging
        interval: 60m
      estimate: true
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 5m # switch charger contactor not more often than this (default 5m)
    priority: 0
```

### Log details

```text
-
```

### What type of operating system are you running?

Docker container

VolkerK62, 2023-03-01:
In der Konfig oben, haben beide Loadpoints `priority: 0`.

Unabhängig davon:
Ich weiß nicht, wie `priority` umgesetzt ist, aber aus der Doku heraus würde ich vermuten, dass es nur greift, wenn geladen wird.
Du könntest den Start der priorisierten Loadpoints aber über `enable.delay` steuern, dass der sich als erstes einschaltet.

nicx, 2023-03-01:
@VolkerK62 stimmt, die konfig stammt aus einem meiner vielen versuche dazu... da ist aktuell natürlich bei einem loader die priority=1.

im grunde würde ich sowieso am liebsten ein fahrzeug bevorzugen, und nicht einen loadpoint. das feature hierzu ist ja ganz frisch vor ein paar tagen umgesetzt worden, ich teste das verhalten hier mal sobald es released wurde.

dein workaround mit dem delay könnte auch passen, das probier ich auch mal aus. danke für die idee :) ich bin gespannt wie es sich dann mit dem batterie-laden dabei verhält und ob ich so meine wunschreihenfolge "hinkonfiguriert" bekomme.

VolkerK62, 2023-03-01:
>sobald es released wurde.

ganz frisch, Version 0.114.0 ;-)

>ich bin gespannt wie es sich dann mit dem batterie-laden dabei verhält

Du hast beide Soc auf 100 stehen, da hat die Batterie immer Vorrang.

>Sollte kein PV Überschuss vorhanden sein, wird weder Batterie, noch einer der Loadpoints beladen.

Das Laden/Entladen der Hausbatterie regelt deren BMS. Das wird nicht direkt von evcc gesteuert.

nicx, 2023-03-01:
> > sobald es released wurde.
> 
> ganz frisch, Version 0.114.0 ;-)
> 

ok, aber leider wohl verbugged... sobald ich die priority auf vehicle ebene einfüge bekomme ich
vehicle soc: vehicle not available: cannot create vehicle 'template': yaml: line 12: did not find expected key

> > ich bin gespannt wie es sich dann mit dem batterie-laden dabei verhält
> 
> Du hast beide Soc auf 100 stehen, da hat die Batterie immer Vorrang.
> 

yup, desegen habe ich das,so konfiguriert, wenn auch in der doku steht das pufferSoc immer kleiner sein muss als batterySoc. ich hoffe mal dass das trotzdem funktioniert wie gewünscht.

> > Sollte kein PV Überschuss vorhanden sein, wird weder Batterie, noch einer der Loadpoints beladen.
> 
> Das Laden/Entladen der Hausbatterie regelt deren BMS. Das wird nicht direkt von evcc gesteuert.

klar, aber das starten des ladevorgangs regelt ja evcc. und nach einigen weiteren tests finde ich es ernüchtern dass egal welche delays ich einstelle beim neustart von evcc grundsätzlich sofort auch bei modus "pv" das laden beider loadpoints gestartet wird!auch wenn kein pberschuss vorhanden. jetzt ist es zb komplett nacht, und wenn ich mit meiner konfig evcc starte werden beide autos sofort geladen, bis dannnach dem disable delay timer beides gestoppt wird. gibts irgendwo ne option dass das laden grundsätzlich erst einmal nicht startet, sondern auch bei reboot von evcc der enable timer greift?

nicx, 2023-03-01:
ergänzung: ich revidiere meine erkenntniss, dass das laden losgeht nach reboot. die ui ist hier recht verwirrend. es scheint wohl nur der timer loszulaufen fürs umschalten von 3phasig auf 1phasig, mit der passenden medlung und dem countdown in der ui. wenn ich währenddessen auf die wallboxen schaue wird aber nicht geladen. das passt also, nur die ui meldung ist wie gesagt verwirrend (und eventuell ein bug?!).

andig, 2023-03-01:
Es ist unklar, worum es hier geht- verschoben nach Discussions.