evcc-io/evcc #6571: Dein Sponsortoken läuft vor 106.752 Tagen ab.

abeggled, 2023-03-01:
### Describe the bug

Seit der Version 0.114.0 erscheint folgende Meldung in den Einstellungen:

"Dein Sponsortoken läuft vor 106.752 Tagen ab. Hol dir ein Neues und aktualisiere deine Konfigurationsdatei.
Hinweis: Wir werden das zukünfig automatisieren."

Sponsorticket ist aktuell, extra neu generiert. Ist eine Lifetime $100 Variante

### Steps to reproduce

1. Upgrade auf 0.114.0
2. Gehe auf Einstellungen

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.a38.ch # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error
  wb01: trace

# unique installation id
plant: dc20acc4199bc233a617636f5c9ed11069b2ee03b9834a2d1827b8fed9915782

interval: 10s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3NzIzMTMwNTIsImlhdCI6MTY3NzcwNTA1MiwiaXNzIjoiZXZjYy5pbyIsInN1YiI6ImFiZWdnbGVkIn0.saAYPltKqybo-D-X7t1MpQpBozwoCfZhtV0LG_op-KY

meters:
- type: template
  template: kostal-plenticore
  id: 71
  host: wr.a38.ch
  port: 1502
  usage: pv
  modbus: tcpip
  name: pv1
- type: template
  template: kostal-plenticore
  id: 71
  host: wr.a38.ch
  port: 1502
  usage: battery
  modbus: tcpip
  name: battery2
- type: template
  template: kostal-ksem-inverter
  id: 71
  host: wr.a38.ch
  port: 1502
  usage: grid
  modbus: tcpip
  name: grid3

chargers:
- type: template
  template: webasto-next
  host: wb01.a38.ch
  port: 502
  name: wb01

vehicles:
- name: mx30
  type: custom
  title: Mazda MX-30 # display name for UI
  capacity: 32 # byttery capacity (kWh)
  phases: 1
  soc: # battery soc (%)
    source: mqtt
    topic: mazda/mx30/soc
  range: # optional electric range (km)
    source: mqtt
    topic: mazda/mx30/range
  status: # FahrzeugStatus (A...F)
    source: mqtt
    topic: mazda/mx30/state
  Odometer: # Kilometerstand
    source: mqtt
    topic: mazda/mx30/odo
  onIdentify: # set defaults when vehicle is identified
    mode: pv
    minSoC: 20
    targetSoC: 90

loadpoints:
- title: Garage links
  charger: wb01
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 23
  resetOnDisconnect: true
  vehicle: mx30
  soc:
    poll:
      mode: charging
      interval: 1h
    estimate: true

site:
  title: Heimetli
  meters:
    grid: grid3
    pvs:
    - pv1
    batteries:
    - battery2
  residualPower: 100

tariffs:
  currency: CHF # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.2381 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.2 # [currency]/kWh

mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable publishing

influx:
  url: http://influxdb.iqgroup.ch:8086
  database: db_evcc
  user: evcc
  password: LR1FMRyr37CwvMhrjKcr

messaging:
# Message config using evcc go-text-template rendering, evcc variables and sprig-functions
  events:
    start: # charge start event
      title: Ladung von {{.vehicleTitle}} gestartet
      msg: |
        Wallbox {{.title}} Ladung von {{.vehicleTitle}} im {{ toString .mode | upper }} Modus.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        #{{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    stop: # charge stop event
      title: Ladung von {{.vehicleTitle}} beendet
      msg: |
        Wallbox {{.title}} hat die Ladung von {{.vehicleTitle}} beendet
        with {{round (divf .chargedEnergy 1000) 2 }} kWh in {{.chargeDuration}}.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        #{{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    connect: # vehicle connect event
      title: "{{.vehicleTitle}} verbunden mit Wallbox {{.title}}"
      msg: |
        {{.vehicleTitle}} verbunden mit Wallbox {{.title}} mit {{round (divf .pvPower 1000) 2 }} kW PV.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        #{{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    disconnect: # vehicle connected event
      title: "{{.vehicleTitle}} Verbindung mit Wallbox {{.title}} getrennt"
      msg: |
        {{.vehicleTitle}} Verbindung miz Wallbox {{.title}} nacht {{.connectedDuration}} getrennt.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        #{{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}

  services:
    - type: pushover
      app: am8ehf548iwn97zdepo26133dqgite
      recipients:
        - gxgu789nqqa7g3zqctfgx16n9wxns4
```

### Log details

```text
keine informationen im debug log
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.114.0

nelli81, 2023-03-01:
Kleiner Tipp:
Nimm vielleicht das Token und die PW‘s aus dem Code…

naltatis, 2023-03-02:
@nelli81 Du kannst die Meldung erstmal ignorieren. Das ist eine reine Information die in deinem Fall falsch ausgespielt wird.

@andig die UI scheint hier einen negativen Ablaufzeitraum zu bekommen (`läuft vor(!) X Tagen ab`). Vermutlich ist das Szenario mit den Einmaltoken nicht richtig abgedeckt. Ich hatte erst gedacht, dass 106.752 eine Dezimalzahl ist, das sind aber scheinbar wirklich hunterttausend Tage. Also vermutlich ne Rechnung gegen ein nicht gesetztes Datum.

Bluppy, 2023-03-02:
Habe das gleiche Problem. Nach update auf 114 startet evcc nicht mehr.

telemetry need Sponsoring.

`# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
influx:  
  url: http://192.168.149.4:8086  
  database: evcc  
  user: *****  
  password: *****  
  
log: error  
levels:  
  cache: error  
  
# unique installation id  
plant: *****  
  
interval: 10s # control cycle interval  
  
tariffs:  
  currency: EUR # (default EUR)  
  grid:  
    # static grid price  
    type: fixed  
    price: 0.336 # [currency]/kWh  
  
  feedin:  
    # rate for feeding excess (pv) energy to the grid  
    type: fixed  
    price: 0.09 # [currency]/kWh  
  
sponsortoken: *****  
  
# sponsors can set telemetry: true to enable anonymous data aggregation  
# see https://github.com/evcc-io/evcc/discussions/4554  
telemetry: true  
  
meters:  
- type: template  
  template: kostal-plenticore   
  id: 71    
  host: 192.168.149.150    
  port: 1502    
  usage: pv    
  modbus: tcpip    
  name: pv1  
- type: template  
  template: kostal-plenticore   
  id: 71    
  host: 192.168.149.150    
  port: 1502    
  usage: battery    
  modbus: tcpip    
  name: battery2  
- type: template  
  template: kostal-ksem-inverter   
  id: 71    
  host: 192.168.149.150    
  port: 1502    
  usage: grid    
  modbus: tcpip    
  name: grid3  
  
chargers:  
- type: template  
  template: amtron   
  id: 255    
  host: 192.168.149.21    
  port: 502    
  modbus: tcpip    
  name: wallbox6  
  
vehicles:  
- type: template  
  template: kia   
  title: Ceed Auto    
  user: *****  
  password: *****  
  vin: *****  
  capacity: 8    
  phases: 1    
#  cache: 30m    
  language: en    
  mode: minpv    
  targetSoC: 100    
  minCurrent: 6    
  maxCurrent: 16    
  name: ev4  
- type: template  
  template: kia   
  title: Niro Auto    
  user: *****  
  password: *****  
  vin: *****  
  capacity: 64    
  phases: 3    
#  cache: 15m    
  language: en    
  mode: pv    
  targetSoC: 80    
  minCurrent: 6    
  maxCurrent: 16    
  name: ev5  
  
loadpoints:  
- title: Wallbox  
  charger: wallbox6  
  mode: off  
  phases: 3  
  mincurrent: 6  
  maxcurrent: 16  
  resetOnDisconnect: true  
  
site:  
  title: Mein Zuhause  
  bufferSoc: 80 # Hausbatterie wird oberhalb Soc 80% als Puffer genutzt  
  prioritySoc: 50 # Hausbatterie bekommt bis zum Soc 50% Priorität beim lade  
  residualPower: 300  
  meters:  
    grid: grid3  
    pvs:  
    - pv1  
    batteries:  
    - battery2  
  `

Bluppy, 2023-03-02:
habe das gleiche problem. evcc kommt nicht mehr hoch.

kann auch kein neuen sponsorkey erzeugen obwohl ich sponsor bin

Bluppy, 2023-03-02:
ich musste den token neu ausstellen und feststellen das paypal bei github rausgeflogen ist.

kad49007, 2023-03-02:
Hab die gleiche Meldung, nur mit monatlichen Sponsoring. Wenn ich einen neuen Token erstellen lasse, bekomme ich den bisherigen wieder. Scheint aber zum Glück keine funktionalen Auswirkungen zu haben.
[image]

andig, 2023-03-05:
Könntet ihr mir mal bitte eins der "kaputten" Token an info@evcc.io schicken?

andig, 2023-03-05:
Fixed in https://github.com/evcc-io/evcc/commit/e11c0d8d61eb0e05554fe7e43ea990f10a5b7df1, hoffentlich...

closed 2023-03-05 by andig