evcc-io/evcc #6581: RCT Inverter: Error messages

Jambola77, 2023-03-02:
### Describe the bug

When connecting an RCT 8.0 inverter there are some error messages displayed:

```
grid energy: invalid response: Cmd Response (05) Id Battery SoC target (8B9FF008) Data [63 0 0 0]
vor 2 Sekunden
battery 1 power: invalid response: Cmd Response (05) Id Inverter AC power [W] (DB2D69AE) Data [0 0 0 0]
vor 12 Sekunden
battery 1 soc: invalid response: Cmd Response (05) Id Total grid power [W] (91617C58) Data [69 147 108 51]
vor 52 Sekunden
grid energy: invalid response: Cmd Response (05) Id #INVALID (5B10CE81) Data [1]
vor 52 Sekunden
battery 1 soc: invalid response: Cmd Response (05) Id #INVALID (3623D82A) Data [0 2]
vor 52 Sekunden
battery 1 power: invalid response: Cmd Response (05) Id #INVALID (70A2AF4F) Data [0 0 0 0]
vor 1 Minute
grid energy: invalid response: Cmd Response (05) Id #INVALID (5B10CE81) Data [1]
vor 1 Minute
battery 1 soc: invalid response: Cmd Response (05) Id #INVALID (E96F1844) Data [0 0 0 0]
vor 1 Minute
pv 1 power: invalid response: Cmd Response (05) Id Battery power [W] (400F015B) Data [65 225 161 129]
vor 1 Minute
battery 1 power: invalid response: Cmd Response (05) Id Total grid power [W] (91617C58) Data [69 144 138 199]
vor 1 Minute
pv 1 power: invalid response: Cmd Response (05) Id #INVALID (0CB5D21B) Data [0 0 0 4]
vor 1 Minute
battery 1 power: invalid response: Cmd Response (05) Id #INVALID (97E203F9) Data [0]
vor 2 Minuten
battery 1 power: invalid response: Cmd Response (05) Id #INVALID (1AC87AA0) Data [69 144 193 251]
vor 2 Minuten
battery 1 power: invalid response: Cmd Response (05) Id #INVALID (663F1452) Data [2]
vor 2 Minuten
grid energy: invalid response: Cmd Response (05) Id Total grid power [W] (91617C58) Data [70 42 68 156]
vor 3 Minuten
pv 1 power: invalid response: Cmd Response (05) Id #INVALID (0CB5D21B) Data [0 0 0 4]
vor 3 Minuten
grid energy: invalid response: Cmd Response (05) Id Inverter state (5F33284E) Data [6]
vor 3 Minuten
pv 1 power: invalid response: Cmd Response (05) Id Battery power [W] (400F015B) Data [65 213 142 191]
vor 3 Minuten
grid meter: invalid response: Cmd Response (05) Id #INVALID (4E04DD55) Data [59 3 18 111]
vor 3 Minuten
battery 1 soc: invalid response: Cmd Response (05) Id Total grid power [W] (91617C58) Data [68 11 55 224]
vor 3 Minuten
pv 1 power: invalid response: Cmd Response (05) Id Battery power [W] (400F015B) Data [65 211 98 171]
vor 3 Minuten
grid energy: invalid response: Cmd Response (05) Id Battery SoC target (8B9FF008) Data [63 0 0 0]
vor 3 Minuten
battery 1 soc: invalid response: Cmd Response (05) Id Inverter state (5F33284E) Data [6]
vor 3 Minuten
grid energy: invalid response: Cmd Response (05) Id Total grid power [W] (91617C58) Data [68 10 217 87]
vor 3 Minuten
battery 1 soc: invalid response: Cmd Response (05) Id #INVALID (3623D82A) Data [0 2]
vor 3 Minuten
pv 1 power: invalid response: Cmd Response (05) Id Battery power [W] (400F015B) Data [65 204 20 104]
vor 3 Minuten
pv 1 power: invalid response: Cmd Response (05) Id #INVALID (3B7FCD47) Data [0 0 0 0]
vor 4 Minuten
battery 1 power: invalid response: Cmd Response (05) Id #INVALID (97E203F9) Data [0]
vor 4 Minuten
pv 1 power: invalid response: Cmd Response (05) Id #INVALID (3623D82A) Data [0 2]
vor 4 Minuten
battery 1 soc: invalid response: Cmd Response (05) Id #INVALID (1AC87AA0) Data [69 144 107 111]
vor 4 Minuten
battery 1 power: invalid response: Cmd Response (05) Id #INVALID (E96F1844) Data [0 0 0 0]
vor 4 Minuten
pv 1 power: invalid response: Cmd Response (05) Id #INVALID (4E04DD55) Data [59 3 18 111]
vor 4 Minuten
grid energy: invalid response: Cmd Response (05) Id #INVALID (1AC87AA0) Data [69 144 86 88]
vor 4 Minuten
battery 1 soc: invalid response: Cmd Response (05) Id #INVALID (E96F1844) Data [0 0 0 0]
```

evcc seems to work normally, but I am not sure if there is a bug in the API.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
As this is not my installation, I have to postpone the config details.

The RCT config is the same as in the documentation with grid, pv and battery.
```

### Log details

```text
see above
```

### What type of operating system are you running?

Linux

### Version

0.112.5 (update was not available for raspberry yesterday)

andig, 2023-03-02:
Vielleicht kann @mlnoga helfen?

mlnoga, 2023-03-03:
Ist die RCT App parallel irgendwo geöffnet? Die Firmware des Inverters kann nicht vernünftig mit Anfragen von mehreren Apps umgehen.

andig, 2023-03-05:
Seems resolved. Thank you @mlnoga

closed 2023-03-05 by andig

mergeli, 2023-03-17:
Hi. Ich habe bei mir das gleiche Problem und die gleichen Fehlermeldungen. Die RCT App ist nicht geöffnet. Woran könnte es noch liegen?

andig, 2023-03-17:
Du hast noch andere Dinge die zugreifen, muss ja nicht nur die App sein...