evcc-io/evcc #6595: schaltbare Steckdose als integrated Device - keine Funktion

StevieC121176, 2023-03-03:
Ich habe meine Homematic Steckdose über `EVCC configure` als` integrateddevice` konfiguriert und in meine yaml übertragen (dem Loadpint "Fahrrad" dann den Charger "wallbox10" zugeordnet)

evcc version 0.113.2 ```(5a964c75)```

```
 Choose a wallbox: Homematic / Homematic IP [Switchable sockets]

Please provide the following settings:

 IP address or hostname *****
 XML-RPC server port number (BidCos-Wired=2000, BidCos-RF=2001, HmIP=2010) 2010
 Device address/Serial number *****
 Standby power in W 15
 Username *****
 Password [? for help] *****
 Meter channel number (HMIP-PSM=6, HMIP-FSM+HMIP-FSM16=5, HM=2) 6
 Switch/Actor channel number (HMIP-PSM=3, HMIP-FSM+HMIP-FSM16=2, HM=1) 3
 Integrated device (no vehicle connection) Yes
 Icon Chose: bike

Testing the configuration of Homematic / Homematic IP [Switchable sockets] ...

Homematic / Homematic IP [Switchable sockets] Homematic / Homematic IP [Switchable sockets] was successfully added.

The configuration:

  type: template
  template: homematic 
  host: *****
  port: 2010  
  device: *****  
  standbypower: 15  
  user: *****
  password: *****  
  meterchannel: 6  
  switchchannel: 3  
  integrateddevice: true  
  icon: bike  
  name: wallbox10
```

Mein Loadpoint Fahrrad zeigt mir aber immer noch die Fahrzeugauswahl an. Habe ich etwas übersehen?

[image]
[image]

_Originally posted by @StevieC121176 in https://github.com/evcc-io/evcc/discussions/6483_

thierolm, 2023-03-03:
Schick mal bitte einen dump deines Config Files (`evcc dump --cfg`) ...

StevieC121176, 2023-03-03:
```
pi@raspberrypi:~ $ evcc dump --cfg
[main  ] INFO 2023/03/03 09:23:25 evcc 0.114.0 (d69077b4)
[main  ] INFO 2023/03/03 09:23:25 using config file: /etc/evcc.yaml
[db    ] INFO 2023/03/03 09:23:25 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2023/03/03 09:23:25 connecting evcc-933997641 at tcp://192.168.178.77:1883

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.117  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.118  
  port: 1502  
  usage: pv 
  modbus: tcpip 
  name: pv2
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.118  
  port: 1502  
  usage: battery  
  modbus: tcpip 
  capacity: 12.8 
  name: battery2
- type: template
  template: kostal-ksem 
  id: 71  
  host: 192.168.178.116  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid3

chargers:
- type: template
  template: daheimladen-mb 
  host: 192.168.178.74 
  port: 502
  name: wallbox9
- type: template
  template: homematic 
  host: 192.168.178.68  
  port: 2010  
  device: '**********'
  standbypower: 15  
  user: *****
  password: *****
  meterchannel: 6  
  switchchannel: 3 
  integrateddevice: true  
  icon: bike  
  name: wallbox10

vehicles:
- type: template
  template: enyaq
  title: Skoda Enyaq 80iV
  user: *****
  password: *****
  vin: *****
  capacity: 82
  phases: 3
  cache: 15
  mode: off
  minSoC: 30
  targetSoC: 99
  minCurrent: 6
  maxCurrent: 16
  identifiers:
  - *****
  name: ev6
  icon: car
- type: template
  template: skoda 
  title: Skoda Citigo iV  
  user: *****
  password: *****
  vin: *****
  capacity: 38  
  phases: 2  
  cache: 15  
  mode: off  
  minSoC: 30
  targetSoC: 99  
  minCurrent: 6  
  maxCurrent: 16  
  identifiers:
  - ***** 
  name: ev7
  icon: car
- type: template
  template: offline 
  title: E-Bike
  capacity: 0.75
  phases: 1  
  mode: off  
  minCurrent: 0.2
  maxCurrent: 0.2
  name: ev8
  icon: bike
- type: template
  template: id 
  title: Volkswagen ID5  
  user: *****
  password: *****
  vin: *****
  capacity: 82  
  phases: 3  
  icon: car  
  cache: 15m  
  mode: Off  
  minSoc: 30  
  targetSoc: 99  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev9

loadpoints:
- title: Garage
  charger: wallbox9
  mode: off
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
  Priority: 1
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true
- title: Fahrrad
  charger: wallbox10
#  vehicle: ev8
  mode: off
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
  Priority: 2

site:
  title: Mein Zuhause
  meters:
    grid: grid3
    pvs:
    - pv1
    - pv2
    batteries:
    - battery2
  residualPower: 100
  bufferSoC: 100
  prioritySoC: 90

influx:
  url: http://192.168.178.80:8086
  database: evcc
  user: *****
  password: *****

modbusproxy:
  - port: 5200
    uri: 192.168.178.74:502 # IP und Port des Gerätes, das abgefragt werden soll

mqtt:
  broker: 192.168.178.77:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: *****
  password: *****

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.234 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # [currency]/kWh

  planner:
    type: grünstromindex
    zip: "*****"

Version: `0.114.0 (d69077b4)`

pi@raspberrypi:~ $ 
```

thierolm, 2023-03-03:
Der `vehicle`Parameter am Fahrrad Loadpoint ist auskommentiert ...
```yaml
- title: Fahrrad
  charger: wallbox10
#  vehicle: ev8
  mode: off
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
  Priority: 2
```

StevieC121176, 2023-03-03:
Ja , war nur zu Testzwecken. Nun wird das Gastfahrzeug am Loadpoint angezeigt. Aber der Parameter `integrated Vehicle` soll ja die komplette Vehicle Auswahl ausblenden.

thierolm, 2023-03-03:
OK, dann versuch mal Folgendes und ziehe dir die neuste Version 0.114:
```yaml
- title: Fahrrad
  charger: wallbox10
  integrateddevice: true
  mode: off
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
  Priority: 2
```

Hofyyy, 2023-03-03:
Hab den gleichen Bug

erikgieseler, 2023-03-04:
Bei der Tasmota Implementierung besteht scheinbar das gleiche Problem:

```
- name: waermepumpe
  type: template
  template: tasmota
  host: 192.168.180.12
  user: admin
  password: 123
  channel: 2
  standbypower: -1
  integrateddevice: true

- name: heizstab
  type: template
  template: tasmota
  host: 192.168.180.13
  user: admin
  password: 123
  channel: 1
  standbypower: -1
  integrateddevice: true
```

VolkerK62, 2023-03-04:
MQTT-Explorer meldet `false`

[image]

StevieC121176, 2023-03-04:
hmn, bei mir ist es anders. Ich habe 2 Ladepunkte. Bei dem Ladepunkt als` integrateddevice` wird `chargerFeatureIntegratedDevice = NaN` angezeigt. Bei dem anderen `chargerFeatureIntegratedDevice = false`

thierolm, 2023-03-04:
Moin Freund*innen des integratedDevice Features.
Könnt ihr bitte die den Issue umbenennen.
Das Thema hat NICHTS mit einer der Schaltbaren Steckdosen zu tun!
Vielmehr ist es ein allgemeines charger/loadpoint Problem.

closed 2023-03-05 by andig

andig, 2023-03-06:
> Das Thema hat NICHTS mit einer der Schaltbaren Steckdosen zu tun!

Also irgendwie wars schon eins- ich hatte die Templates der Steckdosen nicht vollständig angehübscht :)

aendes, 2023-03-12:
Hallo,

wäre es auch sinnvoll, das custom template um die Funktion integrateddevice zu erweitern?

Mein Szenario ist, dass ich ein custom device per Script steuere.

andig, 2023-03-12:
> wäre es auch sinnvoll, das custom template um die Funktion integrateddevice zu erweitern?

Was meinst Du damit? Bitte neues Thema- hier sind wir fertig :)