evcc-io/evcc #6599: Warnung wenn Wallbox (DaheimLaden) mit nur 2 Phasen verbunden ist

Philipp23, 2023-03-03:
### Describe the bug

Habe meine DaheimLaden Wallbox (DaheimLader Smart) mit 2 Phasen (L1 und L2) ans Netz angeschlossen, um eine bessere Ausnutzung beim PV-Überschussladen zu erzielen.

Die Regelung des Stroms funktioniert perfekt, allerdings bekomme ich im Log ständig Warnung wegen der Phasenkonfiguration.
Siehe hier:
Mar 03 11:10:24 raspberrypi evcc[20982]: [lp-1  ] WARN 2023/03/03 11:10:24 invalid phase wiring between charge meter and charger

Die Warnung wird alle 10 Sekunden ausgegeben.

Könnte man die Warnung irgendwie unterdrücken oder eine entsprechende Änderung im Code vornehmen, dass eine solche Situation nicht als Warnung identifiziert wird?

### Steps to reproduce

In der evcc.yaml beim Loadpoint "phases: 2" einstellen.

### Configuration details

```yaml
chargers:
- type: template
  template: daheimladen-mb
  host: 192.168.2.1
  port: 502
  name: wallbox4
loadpoints:
- title: Carport
  charger: wallbox4
  mode: Off
  phases: 2
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
vehicles:
- type: template
  template: tesla 
  title: MY  
  accessToken: ***
  refreshToken: ***
  vin: ***
  capacity: 77.2  
  phases: 3  
  icon: car  
  mode: Off  
  minSoC: 0  
  targetSoC: 90  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev3
```

### Log details

```text
Mar 03 11:17:34 raspberrypi evcc[20982]: [lp-1  ] WARN 2023/03/03 11:17:34 invalid phase wiring between charge meter and charger
Mar 03 11:17:44 raspberrypi evcc[20982]: [lp-1  ] WARN 2023/03/03 11:17:44 invalid phase wiring between charge meter and charger
Mar 03 11:17:54 raspberrypi evcc[20982]: [lp-1  ] WARN 2023/03/03 11:17:54 invalid phase wiring between charge meter and charger
Mar 03 11:18:04 raspberrypi evcc[20982]: [lp-1  ] WARN 2023/03/03 11:18:04 invalid phase wiring between charge meter and charger
Mar 03 11:18:14 raspberrypi evcc[20982]: [lp-1  ] WARN 2023/03/03 11:18:14 invalid phase wiring between charge meter and charger
Mar 03 11:18:24 raspberrypi evcc[20982]: [lp-1  ] WARN 2023/03/03 11:18:24 invalid phase wiring between charge meter and charger
Mar 03 11:18:34 raspberrypi evcc[20982]: [lp-1  ] WARN 2023/03/03 11:18:34 invalid phase wiring between charge meter and charger
Mar 03 11:18:44 raspberrypi evcc[20982]: [lp-1  ] WARN 2023/03/03 11:18:44 invalid phase wiring between charge meter and charger
Mar 03 11:18:54 raspberrypi evcc[20982]: [lp-1  ] WARN 2023/03/03 11:18:54 invalid phase wiring between charge meter and charger
Mar 03 11:19:04 raspberrypi evcc[20982]: [lp-1  ] WARN 2023/03/03 11:19:04 invalid phase wiring between charge meter and charger
Mar 03 11:19:14 raspberrypi evcc[20982]: [lp-1  ] WARN 2023/03/03 11:19:14 invalid phase wiring between charge meter and charger
Mar 03 11:19:24 raspberrypi evcc[20982]: [lp-1  ] WARN 2023/03/03 11:19:24 invalid phase wiring between charge meter and charger
```

### What type of operating system are you running?

Linux

### Version

0.114.0

VolkerK62, 2023-03-03:
Bäh, das ist doof.  
Die Spannungserkennung ist extra reingekommen, damit der `phases` Wert automatisch anhand der anliegenden Spannungen eingestellt wird. Dabei ist aber nur 1p oder 3p "zulässig".

premultiply, 2023-03-03:
Es sind nur 1p (L1) oder 3p (L1+L2+L3) zulässig.
Andere Kombinationen sind im Standard nicht vorgesehen und werden auch von evcc nicht unterstützt.

closed 2023-03-03 by premultiply

Philipp23, 2023-03-05:
> Bäh, das ist doof. Die Spannungserkennung ist extra reingekommen, damit der `phases` Wert automatisch anhand der anliegenden Spannungen eingestellt wird. Dabei ist aber nur 1p oder 3p "zulässig".

Implementiert ist es jedenfalls so, dass die Anzahl der Phasen über die an den Phasen anliegende Spannung ermittelt wird.
Es folgt dann aber eine Unterscheidung rein zwischen 1p und 3p. Wobei in meinen Fall (2 spannungsversorgte Phasen) programmintern mit 3 Phasen weitergearbeitet wird.

Ich hab mir den Code jetzt ein wenig umgeschrieben, damit die Warnung nicht mehr ausgegeben wird und der Wert phases = 2 entsprechend in die anderen Methoden rein übergeben wird. Ob es was ändert, weiß ich nicht. Ich teste es einfach mal.

Dem Auto scheints egal zu sein. Zeigt zwar auch eine Warnung an, dass eine Phase fehlt und dadurch die Ladeleistung reduziert ist, aber es lädt. Und auch EVCC regelt im PV-Modus völlig korrekt die Leistung.