evcc-io/evcc #6615: Automatic Phase switching with Tinkerforge WARP Energy Manager

deadrabbit87, 2023-03-04:
**Is your feature request related to a problem? Please describe.**
Tinkerforge has released in 02.2023 the [WARP Energy Manager.](https://www.warp-charger.com/energy-manager.html#firmware) 

In combination with the WARP 2 Charger, automatic phase switching (1ph 3 ph) is know supported. 

**Describe the solution you'd like**
Integration of the WARP Energy Manager in evcc. Evcc should be able to control the used phases. 

The API of the energy manager: 
https://www.warp-charger.com/api.html?v=2#reference-energy_manager

andig, 2023-03-05:
Does the WEM only switch phases or does it also control the charger? There should only be a single controlling instance.

deadrabbit87, 2023-03-05:
I'm not sure, but i think it's depending on the configuration. I've installed my energy manager yesterday and for the moment it is not configured with a smart meter. Therefore it is only possible to switch between "schnell" and "off": 
[image]
According to the [api](https://www.warp-charger.com/api.html?v=2#reference-energy_manager), it should be possbile to switch the phases externally. See `phase_switching_mode.`

andig, 2023-03-05:
Certainly. The question is under which conditions we should do that. Just always???

deadrabbit87, 2023-03-05:
I don't know, what makes sense. Maybe when "PV-Überschussladen" is switched of?
[image]

TMA84, 2023-03-05:
I think you would have to set the charging mode in the Energy Manager via API and start and stop charging in the Warp Charger via API.

If the "PV Überschussladen" is disabled, there will probably be no automatic phase switching.

TMA84, 2023-03-05:
> Does the WEM only switch phases or does it also control the charger? There should only be a single controlling instance. I think we need some automatism to tell evcc that it should take control of an WEM if it is present. How would be determine that that's the case?

I think this is the best solution.

deadrabbit87, 2023-03-06:
Hi,

maybe helpful:

https://www.tinkerunity.org/topic/11399-warp-energy-manager/?do=findComment&comment=50770

TMA84, 2023-03-06:
@andig 
Unfortunately I can't program, but I could help with documentation and testing. But the integration would be very helpful.

deadrabbit87, 2023-03-06:
> @andig Unfortunately I can't program, but I could help with documentation and testing. But the integration would be very helpful.

Same as me. If there is anything to do, i would try to help.

andig, 2023-03-06:
Wir können das ja mal probehalber einbauen, aber das Grundproblem bleibt: es braucht irgendeinen Automatismus, der evcc mitteilt, ob wir die Phasenumschaltung übernehmen dürfen und sollen.

TMA84, 2023-03-07:
Ich hätte da gegebenenfalls einen anderen Lösungsansatz.
Könntet ihr nicht anstatt direkt die Wallbox zu steuern den Energy Manager den ausgewählten Ladestatus übergeben und dieser übernimmt dann die Steuerung?

Das könnte man per HTTP API oder auch per MQTT machen.
https://www.warp-charger.com/api.html?v=2#energy_manager_charge_mode

Zusätzlich könnte man noch den Wert des Zählers an den Energy Manager übergeben.
https://www.warp-charger.com/api.html?v=2#meter_values_update
Hier muss nur der Wert Power den aktuellen Gesamtleistung enthalten.
Die beiden anderen Werte können mit 0 übergeben werden.

deadrabbit87, 2023-03-07:
Was auf jeden Fall mal schon funktioniert: Ich habe min+PV im WARP Energy Manager eingestellt und PV in EVCC. 

Die Wallbox lädt dann erst mal nicht, weil kein Strom von EVCC an WB freigegeben wird. Erst wenn das geschieht, beginnt die Ladung. 

Phasenumschaltung funktioniert dann aber leider nicht automatisch. Setzt nan aber im WEM die Ladung auf Schnell, geht auch das. 

Was ich damit sagen will: Der Lösungsansatz von @TMA84 könnte so funktionieren.

deadrabbit87, 2023-03-16:
Gibt es diesbezüglich schon Neuigkeiten? 

Hab seit längerem beides parallel im Einsatz: EVCC steuert die Wallbox und im Energy Manager schalte ich nur von MIN+PV (1 Phase) und SCHNELL (3 Phasen) um. Funktioniert auch soweit parallel super, nur was halt fehlt, ist dass EVCC auch die Phasen umschaltet. 

Daher wäre die Lösung von @TMA84 schon super in EVCC.

rtrbt, 2023-03-22:
Kurzes Update: Wir haben uns mit @andig auf eine API-Erweiterung geeinigt, damit EVCC die Phasenumschaltung kontrollieren kann. Implementieren wir für die nächste Firmware-Version.

TMA84, 2023-03-22:
Das ist doch mal eine gute Nachricht.

deadrabbit87, 2023-03-22:
Eine sehr gute Nachricht!

andig, 2023-03-22:
PERFEKT! Echt toll, dass ihr Euch soviel Mühe macht, ein Fremdsystem einzubinden…

TMA84, 2023-04-17:
Hallo @rtrbt @andig,
gibt es da schon weiteres? Können wir vielleicht beim testen unterstützen?

rtrbt, 2023-04-19:
Leider noch nicht. Eventuell kommen wir diese Woche noch dazu.

rtrbt, 2023-05-02:
Oh sorry, @MattiasTF  hatte die Beta am Freitag nur @andig direkt geschickt. Ich zitiere mal aus der Mail:

> In den Einstellungen vom WEM kannst du jetzt den Phasenumschaltungsmodus auf „Externe Steuerung“ stellen. Dazu gibt es zwei neue API-Endpunkte:
> 
> /energy_manager/state -> external_control
> Das sagt dir, ob du eine Phasenumschaltung auslösen kannst.
> 0: Phasenumschaltung verfügbar
> 1: Phasenumschaltung nicht verfügbar, externe Steuerung deaktiviert
> 2: Phasenumschaltung nicht verfügbar, Laufzeitbedingungen nicht erfüllt
> 3: Phasenumschaltung nicht verfügbar, Phasen werden gerade umgeschaltet
> 
> Unter 2 fällt sowas wie Wallboxen antworten nicht, Schützfehler, über externen Eingang blockiert, etc.
> 
> /energy_manager/external_control(_update) -> phases_wanted
> Über external_control_update kannst du eine Phasenumschaltung auslösen.
> "phases_wanted":0 - Strom aus, einphasig
> "phases_wanted":1 - Strom an, einphasig
> "phases_wanted":3 - Strom an, dreiphasig
> 
> Über external_control kannst du kontrollieren, ob dein Kommando übernommen wurde oder nicht. In den Zuständen 1-3 (siehe oben) werden Kommandos ignoriert.
> 
> Bitte beachte, dass entgegen der vorherigen Ankündigung aktuell doch keine Totzeit zwischen zwei Phasenumschaltungen erzwungen wird. Eine Umschaltung dauert ca. 15 Sekunden und neue Kommandos werden in der Zeit ignoriert, aber der WEM hält dich nicht davon ab, vier Umschaltungen pro Minute durchzuführen, was dir dein Auto sicherlich übelnehmen wird. Die auf der Debug-Seite angezeigten Wartezeiten werden im extern gesteuerten Modus ignoriert. Die Totzeit liegt somit auch in der Verantwortung von EVCC.
> 
> Bitte beachte (die 2.): Die API ist noch nicht final und der Code hat Beta-Qualität. 😉

[energy_manager_firmware_1_0_1+EVCC_beta.bin.zip](https://github.com/evcc-io/evcc/files/11370618/energy_manager_firmware_1_0_1%2BEVCC_beta.bin.zip)

deadrabbit87, 2023-05-02:
Sehr cool, danke! Ist schon geflasht 😉. Dann hoffe ich, das @andig und @premultiply  so flott sind wie sonst auch 😛😉

deadrabbit87, 2023-05-08:
@andig Ich weiß, du hattest mal gesagt, ihr gebt keine Zeitangaben bis wann ihr das implementiert, daher die Frage: Wo in deiner Prioritätenliste von 1-100 steht der feature request?

Gruß

andig, 2023-05-12:
Kommt auf jeden Fall!

andig, 2023-05-14:
Wie sieht das bei Euch im mqtt aus- hat der EM ein anderes Topic als die Warp? Gibts dafür einen Standard (so wie `warp` für die Wallbox?

deadrabbit87, 2023-05-14:
Sieht bei mir so aus `wem/26ud/energy_manager/...:`
[image]

andig, 2023-05-14:
Und was hast du als warp topic konfiguriert?

deadrabbit87, 2023-05-14:
`warp2/22qS/...` Die Standardnamen müsste ich mal ändern...
[image]

andig, 2023-05-18:
Ihr könnt es gerne nochmal probieren, erfordert the Beta-FW. Wenn das auch nciht klappt wäre es Klasse, wenn mir jemand seine Warp mal über eine Public Broker zur Verfügung stellen könnte (info@evcc.io)

closed 2023-05-19 by andig