evcc-io/evcc #6633: REST API: add Timestamp in JSON result of /api/state

Diavolezza, 2023-03-05:
/api/state returns all actual data of an evcc-installation
It would be helpful for e.g. visualisation if there is a timestamp included in the JSON-result that contains date and time of the creation of all that data.

andig, 2023-03-05:
Is there a specific use case you have in mind? While certainly possible it seems unfortunate to double the amount of data by time-stamping every entry?

Diavolezza, 2023-03-05:
Not a timestamp for every entry but one timestamp in the whole JSON-result of /api/state. E.g. as "result.timestamp". That seems very little additional data.
And yes, I am switching to mqtt at the moment for displaying time-series. I thought I can omit opening the mqtt port using the REST API.

andig, 2023-03-05:
> Not a timestamp for every entry but one timestamp in the whole JSON-result of /api/state. E.g. as "result.timestamp". That seems very little additional data.

Yes, but what's that worth? The only info you'd really get is if the service is still alive and that is already available at `/api/health`.

Diavolezza, 2023-03-06:
If I have multiple responses from api/state (multiple JSON results), I know when they were created (on the host-machine) and I can do evaluations on that. I see that as a real benefit.

andig, 2023-03-13:
That does unfortunately not work as you'd expect it. The `/state` is a collection of values with individual production dates, not a single timestamp. Adding timestamp for the query time would be a task of the calling application. Hence wontfix.

closed 2023-03-13 by andig