evcc-io/evcc #6634: Installation EVCC auf Raspberry 4B klappt nicht

offermannb, 2023-03-05:
### Describe the bug

Hallo zusammen,
möchte EVCC auf einem Raspberry 4B installieren und kriege es einfach nicht hin.
Ich muss zugeben dummer Windows User zu sein und tue mich mit Konsolen schwer.

Installationsversuch über PC mit Ubuntu per ssh auf Raspberry 4b
Nach der Anleitung Erstinstallation geht es schon im ersten Schritt los

**pi@raspberrypi-iobroker:~ $ sudo apt install -y debian-keyring debian-archive-keyring apt-transport-https
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Paket debian-keyring ist nicht verfügbar, wird aber von einem anderen Paket
referenziert. Das kann heißen, dass das Paket fehlt, dass es abgelöst
wurde oder nur aus einer anderen Quelle verfügbar ist.

E: Für Paket »debian-keyring« existiert kein Installationskandidat.**

...und weiter geht es dann so

**pi@raspberrypi-iobroker:~ $ curl -1sLf
'https://dl.cloudsmith.io/public/evcc/stable/setup.deb.sh'
| sudo -E bash
Executing the setup script for the 'evcc/stable' repository ...

OK: Checking for required executable 'curl' ...
OK: Checking for required executable 'apt-get' ...
OK: Detecting your OS distribution and release using system methods ...
^^^^: ... Detected/provided for your OS/distribution, version and architecture:

:
: ... distro=debian version=11 codename=bullseye arch=aarch64
:
OK: Checking for apt dependency 'apt-transport-https' ...
OK: Checking for apt dependency 'ca-certificates' ...
OK: Checking for apt dependency 'gnupg' ...
OK: Checking for apt signed-by key support ...
OK: Importing 'evcc/stable' repository GPG key ...
OK: Checking if upstream install config is OK ...
OK: Installing 'evcc/stable' repository via apt ...
NOPE: Updating apt repository metadata cache ...
Failed to update via apt-get update
Fehl:1 http://your.repo.domain/repository bullseye InRelease
»your.repo.domain« konnte nicht aufgelöst werden.
OK:2 http://archive.raspberrypi.org/debian bullseye InRelease
Holen:3 http://de.archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
OK:4 https://deb.nodesource.com/node_16.x bullseye InRelease
OK:5 https://packages.grafana.com/enterprise/deb stable InRelease
Holen:6 http://de.archive.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Holen:7 http://de.archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]
Holen:8 http://de.archive.ubuntu.com/ubuntu focal InRelease [265 kB]
Holen:9 https://dl.cloudsmith.io/public/evcc/stable/deb/debian bullseye InRelease [5.118 B]
Holen:10 http://archive.canonical.com/ubuntu focal InRelease [12,1 kB]
Holen:11 https://dl.cloudsmith.io/public/evcc/unstable/deb/debian bullseye InRelease [5.105 B]
Fehl:3 http://de.archive.ubuntu.com/ubuntu focal-updates InRelease
Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY 3B4FE6ACC0B21F32 NO_PUBKEY 871920D1991BC93C
Fehl:6 http://de.archive.ubuntu.com/ubuntu focal-security InRelease
Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY 3B4FE6ACC0B21F32 NO_PUBKEY 871920D1991BC93C
Fehl:7 http://de.archive.ubuntu.com/ubuntu focal-backports InRelease
Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY 3B4FE6ACC0B21F32 NO_PUBKEY 871920D1991BC93C
Fehl:8 http://de.archive.ubuntu.com/ubuntu focal InRelease
Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY 3B4FE6ACC0B21F32 NO_PUBKEY 871920D1991BC93C
Fehl:10 http://archive.canonical.com/ubuntu focal InRelease
Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY 3B4FE6ACC0B21F32 NO_PUBKEY 871920D1991BC93C
Paketlisten werden gelesen…
W: GPG-Fehler: http://de.archive.ubuntu.com/ubuntu focal-updates InRelease: Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY 3B4FE6ACC0B21F32 NO_PUBKEY 871920D1991BC93C
E: Das Depot »http://de.archive.ubuntu.com/ubuntu focal-updates InRelease« ist nicht signiert.
W: GPG-Fehler: http://de.archive.ubuntu.com/ubuntu focal-security InRelease: Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY 3B4FE6ACC0B21F32 NO_PUBKEY 871920D1991BC93C
E: Das Depot »http://de.archive.ubuntu.com/ubuntu focal-security InRelease« ist nicht signiert.
W: GPG-Fehler: http://de.archive.ubuntu.com/ubuntu focal-backports InRelease: Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY 3B4FE6ACC0B21F32 NO_PUBKEY 871920D1991BC93C
E: Das Depot »http://de.archive.ubuntu.com/ubuntu focal-backports InRelease« ist nicht signiert.
W: GPG-Fehler: http://de.archive.ubuntu.com/ubuntu focal InRelease: Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY 3B4FE6ACC0B21F32 NO_PUBKEY 871920D1991BC93C
E: Das Depot »http://de.archive.ubuntu.com/ubuntu focal InRelease« ist nicht signiert.
W: GPG-Fehler: http://archive.canonical.com/ubuntu focal InRelease: Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY 3B4FE6ACC0B21F32 NO_PUBKEY 871920D1991BC93C
E: Das Depot »http://archive.canonical.com/ubuntu focal InRelease« ist nicht signiert.
^^^^: ... Failed to update via apt-get update - Context above.
!!!!: Oh no, your setup failed! :-( ... But we might be able to help. :-)
!!!!:**
!!!!: You can contact evcc for further assistance.
!!!!:
!!!!: URL: https://github.com/evcc-io/evcc
!!!!:

Kann mir hier jemand weiter helfen?

### Steps to reproduce

1.Versuch nach Installationsanleitung https://docs.evcc.io/docs/installation/linux
2.
3.
...

### Configuration details

```yaml
So weit komme ich erst gar nicht
```

### Log details

```text
Siehe oben
```

### What type of operating system are you running?

Linux

Sillium007, 2023-03-05:
`your.repo.domain` wie kommt das denn da hin?

Mal versucht vorher ein `sudo apt update` zu machen und dann erst den keyring installieren?

Hofyyy, 2023-03-05:
Also du scheiterst schon am ersten Schritt. Den Rest brauchst du garnicht probieren.

offermannb, 2023-03-05:
> `your.repo.domain` wie kommt das denn da hin?
> 
> Mal versucht vorher ein `sudo apt update` zu machen und dann erst den keyring installieren?

Habe ich bereits probiert, bringt leider nichts.

offermannb, 2023-03-05:
> Also du scheiterst schon am ersten Schritt. Den Rest brauchst du garnicht probieren.

Keine Ahnung warum das so ist, hat das irgendwer hier auf dem Raspi 4 hinbekommen? Oder muss ich den manuellen weg gehen?