evcc-io/evcc #6644: Support for server based databases like MySQL, Postgres or InfluxDB

janknieling, 2023-03-05:
**Is your feature request related to a problem? Please describe.**
At the moment it isn’t possible to use a central DBMS like Postgres (and TimescaleDB), MySQL or InfluxDB.

**Describe the solution you'd like**
I think it will be very useful to have the possibility to replace the local SQLite database against a central database like MySQL or Postgres.
Storing the data in a central DB instance is much better for the performance and it gives you the opportunity to operate EVCC on a smaller system like a Pi zero or something else.
I think Postgres is the better solution because then you can connect it to TimescaleDB too (https://www.timescale.com/ TimescaleDB is a time series DB on top of Postgres that can store efficently timeseries and relation data).

andig, 2023-03-05:
Already supported, just change the database type. However, not all SQL supports that, e.g. session filtering by time. Would need a PR to improve that.

janknieling, 2023-03-05:
Is there a (developer) documentation available where I can see which database types are supported at this time?
At the docu at docs.evcc.io I cant find any Typ of developer documentation or a hint for SQL server support. I only find there something about SQLite

janknieling, 2023-03-05:
Maybe I will find the time to improve the DB support that TimeseriesDB can used without any limitations. I think its a very useful db for something like EVCC where timeseries data’s for metrics will be produced and relational data for permissions or something else

StefanSchoof, 2023-03-05:
Evcc uses gorm https://github.com/evcc-io/evcc/blob/4963842e77f2ced5295abb253d92e5a45b168720/server/db/db.go#L12

> GORM provides official support for sqlite, mysql, postgres, sqlserver.
https://gorm.io/docs/write_driver.html#:~:text=GORM%20provides%20official%20support%20for,to%20implement%20the%20dialect%20interface.

andig, 2023-03-05:
> Maybe I will find the time to improve the DB support that TimeseriesDB can used without any limitations. 

Happy to take PRs to make Mysql, Postgres etc first class citizens. As @StefanSchoof wrote, Gorm ist the basis but we'll need to find a mechanism for adpting custom sql.

TimeseriesDB is not on the roadmap. If you want to fork take a look at the influx implementation.

closed 2023-03-05 by andig

janknieling, 2023-03-05:
Influx isn't an option. It doesn't makes sense to use it if you want to support a web based configuration where the co fig will be stored in the database backend. You can do this with influx too but it's not the best way to do it.
It think for this purpose TimescaleDB is the best option because it can store relational data like the configs and timeseries data

janknieling, 2023-03-05:
You should take a look on it: https://www.timescale.com
Timescale is based on Postgres (the sql queries are the same 
)
Or how are you planning to implement the web based configuration where the config will be stored in the db instead of the yaml files?

panzerdev, 2023-03-24:
To revive this conversation.

I am using Postgres at work and at home for everything and it'S great. Relational data as well as time based Data or even unstructured JSONB. If you do the indexes right the default out of the box postgres is totally fine as a timeseries DB. There is no real need for Timescale DB IMO.

I see two usecases here

1. Inserting Data about the Charging sessions like is done right now with sqllite (which is the less interesting usecase IMO)

2. Insert data about all meters and loadpoints(when the loadpoint is acutally used) All this in a configurable interval.

I have 2 Tables right now for evcc data:
1. is for the powers and is not yet full fleged because I don't have a battery
```
CREATE TABLE public.pv_data (
	created_at timestamp NOT NULL DEFAULT now(),
	pv_power numeric NOT NULL,
	home_power numeric NOT NULL,
	grid_power numeric NOT NULL,
	id serial4 NOT NULL,
	CONSTRAINT pv_data_pkey PRIMARY KEY (id)
);
```
2. table for loadpoints
```
CREATE TABLE public.loadpoint (
	created_at timestamp NOT NULL DEFAULT now(),
	id numeric NOT NULL,
	title text NOT NULL,
	vehicle text NOT NULL,
	charge_mode text NOT NULL,
	phase_action text NOT NULL,
	charge_power numeric NOT NULL,
	active_phases numeric NOT NULL,
	phases numeric NOT NULL,
	idx serial4 NOT NULL,
	CONSTRAINT loadpoint_pkey PRIMARY KEY (idx)
);
```

So evcc you just need to do an `insert` at a fixed interval into a configured db and this would most likely be even fine for all SQL based databases. The specifics of the databases are more visible in getting the data out then putting it in.

Does this help to bring the conversation further along?

maf-soft, 2023-03-24:
The sql command should be configurable in the evcc config with variables, to support any tables and columns or even call a stored procedure instead of insert.

maf-soft, 2023-03-24:
What do you mean with "do it yourself"? Create a fork just for myself and change the command in the code?

panzerdev, 2023-03-24:
You can do it as I do right now which is using the API and postprocessing into the format you need it want. You could also use Mqtt for that.

But bringing this level of configuration into the core seems too much for me tbh.

maf-soft, 2023-03-24:
At least an optional config parameter is very low effort and allows the 10% to do what they want without the need for additional middleware between evcc and the db. Internally the sql query would become a constant with variables anyway, wouldn't it?
But I have already learned that my "maximum flexibility" approach is somehow not popular here. Just wanted to help brainstorm how it would be most helpful.

andig, 2023-03-24:
> which is the less interesting usecase 

Even so it would be the minimum to stay consistent with what we have sofar.

panzerdev, 2023-03-24:
I'm not using the influx integration so the question is:

Does the data that gets put into SQLite also go to influx?

[image]
The data here could/should also be the result is the query against the loadpoint table.

The current implementation is so bare bones that I wouldn't even start talking about consistency here TBH

jinnatar, 2024-12-09:
Could this be reconsidered as a very bare bones "just let me give you a postgres DSN, no fancy time stuff" issue?

SQLite is great in many environments, it just has one major limitation in that it's absolutely allergic to network storage and always eventually fails on it, even though it might take a few months, it will corrupt the database eventually. This means evcc is forever bound to be run on a single static server where any data migrations are completely manual since all modern storage solutions are network based and thus incompatible with SQLite.

andig, 2024-12-09:
Happy to look at a PR that covers all types of queries. DSN alone won't work. That's the SQL hell for you...

jinnatar, 2024-12-09:
Looking at the history, a call was made to move away from pure gorm and focus solely on SQLite, is that something that is still realistically reversible, or rather, I assume there was a good reason to drop support?

andig, 2024-12-09:
Lack of features like date formatting etc...

jigfox, 2025-03-10:
Hi, I am interested in using evcc with Postgres or mysql, and I would be happy to implement it and create a PR for this. But before I do this, I'd like to know if such a PR would have a chance to get merged.

@andig could you point to queries that could be problematic in other databases than sqlite?

andig, 2025-03-10:
See https://github.com/evcc-io/evcc/pull/19651#issuecomment-2711342457