evcc-io/evcc #6654: Tariff Planner: Request failed with status code 400

Torsten81, 2023-03-06:
### Describe the bug

Hallo, hab gerade auf das neueste Nightly Build geupdatet und bekomme nun folgende Fehlermeldung.
Request failed with status code 400: API request failed http://192.168.178.190:7070/api/tariff/planner

Nun ist leider keine Ladeplanung mit Tibber mehr möglich, den Fehler scheint es ja schon einmal gegeben zu haben, habe ihn in den Issues gefunden. 

### Steps to reproduce

1. Fehler tritt nach dem Update auf die neueste Nightly Build Version auf
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
failed configuring grid tariff: cannot create tariff 'tibber': Post "https://api.tibber.com/v1-beta/gql": dial tcp: lookup api.tibber.com: i/o timeout
```

### What type of operating system are you running?

Linux

### Version

0.114.0 (75a488a7)

steve0564, 2023-03-06:
Me too!
#6655

andig, 2023-03-06:
Kein Wunder wenn Tibber nicht erreichbar ist.

closed 2023-03-06 by andig