evcc-io/evcc #6677: TWC3 showing approx. 200 W consumption

ugoetzen, 2023-03-07:
### Describe the bug

since the upgrade to version evcc 0.114.1  evcc shows approx. 200 W charging - even if no car is connected. evcc 0.114.0 was handling this correctly.
Tesla API hasn't changed recently.
[image]

### Steps to reproduce

TWC3 API call: (http://192.168.10.246/api/1/vitals)
{"contactor_closed":false,"vehicle_connected":false,"session_s":0,"grid_v":225.2,"grid_hz":50.047,"vehicle_current_a":0.8,"currentA_a":0.3,"currentB_a":0.4,"currentC_a":0.2,"currentN_a":0.5,"voltageA_v":5.1,"voltageB_v":0.0,"voltageC_v":0.0,"relay_coil_v":11.7,"pcba_temp_c":8.0,"handle_temp_c":4.6,"mcu_temp_c":7.3,"uptime_s":1339338,"input_thermopile_uv":-168,"prox_v":0.0,"pilot_high_v":11.6,"pilot_low_v":11.6,"session_energy_wh":41931.000,"config_status":5,"evse_state":1,"current_alerts":[]}

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.altenburg.home # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: xxx

interval: 10s # control cycle interval

sponsortoken: xxx

meters:
- type: template
  template: sonnenbatterie
  usage: grid
  host: sb-49335.altenburg.home
  port: 8080
  name: grid1
- type: template
  template: sonnenbatterie
  usage: pv
  host: sb-49335.altenburg.home
  port: 8080
  name: pv2
- type: template
  template: sonnenbatterie
  usage: battery
  host: sb-49335.altenburg.home
  port: 8080
  capacity: 12
  name: battery1

chargers:
- type: template
  template: twc3
  host: 192.168.10.246
  name: wallbox1

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.297 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.1227 # [currency]/kWh

vehicles:
- type: template
  template: tesla
  title: Tessarossa
  accessToken: xxx
  refreshToken: xxx
  vin: 5YJ3E7EB0MF96xxxx
  capacity: 84
  phases: 3
  icon: car
  mode: PV
  minSoc: 20
  targetSoc: 80
  minCurrent: 3
  maxCurrent: 20
  name: ev6
- type: template
  template: tesla
  title: TessY
  accessToken: xxx
  refreshToken: xxx
  vin: XP7YGCEL7NB00xxxx
  capacity: 79
  phases: 3
  icon: car
  mode: PV
  minSoc: 20
  targetSoc: 80
  minCurrent: 3
  maxCurrent: 20
  name: ev7

loadpoints:
- title: Carport
  charger: wallbox1
  mode: pv
  phases: 3
  mincurrent: 3
  maxcurrent: 16
  resetOnDisconnect: false
# Message config using evcc go-text-template rendering, evcc variables and sprig-functions
messaging:
  events:
    start: # charge start event
      title: Charge of {{.vehicleTitle}} started
      msg: |
        Wallbox {{.title}} started charging {{.vehicleTitle}} in {{ toString .mode | upper }} mode.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    stop: # charge stop event
      title: Charge of {{.vehicleTitle}} finished
      msg: |
        Wallbox {{.title}} finished charging {{.vehicleTitle}} 
        with {{round (divf .chargedEnergy 1000) 2 }} kWh in {{.chargeDuration}}.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    connect: # vehicle connect event
      title: "{{.vehicleTitle}} connected on wallbox {{.title}}"
      msg: |
        {{.vehicleTitle}} connected on wallbox {{.title}} at {{round (divf .pvPower 1000) 2 }} kW PV.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    disconnect: # vehicle connected event
      title: "{{.vehicleTitle}} disconnected of wallbox {{.title}}"
      msg: |
        {{.vehicleTitle}} disconnected of wallbox {{.title}} after {{.connectedDuration}}.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
  services:
  - type: pushover
    app: xxx # API Token/Key der in Pushover angelegten Aplication
    recipients: xxx

site:
  title: Altenburg
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery1
  prioritySoc: 60
```

### Log details

```text
[main  ] INFO 2023/03/07 09:18:00 evcc 0.114.1
[main  ] INFO 2023/03/07 09:18:00 using config file: /etc/evcc.yaml
[main  ] INFO 2023/03/07 09:18:00 starting ui and api at :7070
[db    ] INFO 2023/03/07 09:18:00 using sqlite database: /root/.evcc/evcc.db
[site  ] INFO 2023/03/07 09:18:01 site config:
[site  ] INFO 2023/03/07 09:18:01   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/03/07 09:18:01     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/03/07 09:18:01     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/03/07 09:18:01     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2023/03/07 09:18:01   vehicles:
[site  ] INFO 2023/03/07 09:18:01     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[site  ] INFO 2023/03/07 09:18:01     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/03/07 09:18:01 loadpoint 1:
[lp-1  ] INFO 2023/03/07 09:18:01   mode:        pv
[lp-1  ] INFO 2023/03/07 09:18:01   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/03/07 09:18:01   meters:      charge ✓
[lp-1  ] INFO 2023/03/07 09:18:01     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2023/03/07 09:18:01 phase timer inactive
[lp-1  ] DEBUG 2023/03/07 09:18:01 pv timer inactive
[site  ] DEBUG 2023/03/07 09:18:01 ----
[lp-1  ] DEBUG 2023/03/07 09:18:01 charge power: 202W
[site  ] DEBUG 2023/03/07 09:18:01 pv power: 2313W
[site  ] DEBUG 2023/03/07 09:18:01 battery soc: 20%
[site  ] DEBUG 2023/03/07 09:18:01 battery power: -1881W
[site  ] DEBUG 2023/03/07 09:18:01 grid power: -15W
[site  ] DEBUG 2023/03/07 09:18:01 giving priority to battery charging at soc: 20%
[site  ] DEBUG 2023/03/07 09:18:01 site power: -15W
[lp-1  ] DEBUG 2023/03/07 09:18:01 charge voltages: [5.1 0 0]V
[lp-1  ] DEBUG 2023/03/07 09:18:01 charge currents: [0.3 0.4 0.2]A
[lp-1  ] DEBUG 2023/03/07 09:18:01 charger status: A
[lp-1  ] INFO 2023/03/07 09:18:01 car disconnected
[site  ] DEBUG 2023/03/07 09:18:01 telemetry: charge: Δ0/0Wh @ 202W
[site  ] DEBUG 2023/03/07 09:18:11 ----
[lp-1  ] DEBUG 2023/03/07 09:18:11 charge power: 201W
[site  ] DEBUG 2023/03/07 09:18:11 pv power: 2271W
[site  ] DEBUG 2023/03/07 09:18:11 battery soc: 20%
[site  ] DEBUG 2023/03/07 09:18:11 battery power: -1845W
[site  ] DEBUG 2023/03/07 09:18:11 grid power: 2W
[site  ] DEBUG 2023/03/07 09:18:11 giving priority to battery charging at soc: 20%
[site  ] DEBUG 2023/03/07 09:18:11 site power: 2W
[lp-1  ] DEBUG 2023/03/07 09:18:11 charge voltages: [5.1 0 0]V
[lp-1  ] DEBUG 2023/03/07 09:18:11 charge currents: [0.3 0.4 0.2]A
[lp-1  ] DEBUG 2023/03/07 09:18:11 charger status: A
[site  ] DEBUG 2023/03/07 09:18:21 ----
[lp-1  ] DEBUG 2023/03/07 09:18:21 charge power: 201W
[site  ] DEBUG 2023/03/07 09:18:21 pv power: 2252W
[site  ] DEBUG 2023/03/07 09:18:21 battery soc: 20%
[site  ] DEBUG 2023/03/07 09:18:21 battery power: -1813W
[site  ] DEBUG 2023/03/07 09:18:21 grid power: -19W
[site  ] DEBUG 2023/03/07 09:18:21 giving priority to battery charging at soc: 20%
[site  ] DEBUG 2023/03/07 09:18:21 site power: -19W
[lp-1  ] DEBUG 2023/03/07 09:18:21 charge voltages: [5.1 0 0]V
[lp-1  ] DEBUG 2023/03/07 09:18:21 charge currents: [0.3 0.4 0.2]A
[lp-1  ] DEBUG 2023/03/07 09:18:21 charger status: A
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.114.1

ugoetzen, 2023-03-07:
Das kann ich mir nicht vorstellen. Ich habe zwar keinen direkten Zähler vor der WB aber der Hausverbrauch sinkt nachts auf ca. 250W mit TWC3 in Standby. Ich denke das ist die schlechte API der TWC3. Hier mal eine Ausweitung der 
TWB_currentA_a
TWB_currentB_a 
TWB_currentC_a mit Ladeaktivität
[image]

Hier wird das Thema auch diskutiert: 
https://tff-forum.de/t/tesla-wallbox-im-ruhezustand/

fynnsh, 2023-03-07:
Sieht wirklich danach aus als wäre hier die API von dem Wallconnector schuld. Sehe da als einzige Möglichkeit die Leistung zu entfernen wenn kein Auto angeschlossen ist. (`vehicle_connected`)

fynnsh, 2023-03-07:
Evtl geht das auch mit contactor_closed, ich gucke nachher mal bei mir was da die richtigen Werte liefert

premultiply, 2023-03-07:
Ist halt grob geschätzte SCHEINleistung...

andig, 2023-03-08:
Wie lösen wir das jetzt: Scheinleistung nur anzeigen bei Status C? Oder wieder ausbauen?

premultiply, 2023-03-08:
> Scheinleistung nur anzeigen bei Status C?

Finde ich hier noch den besten Kompromiss.

ugoetzen, 2023-03-08:
Fände ich suboptimal, das das Fahrzeug ja die meiste Zeit am Kabel hängt und nicht läd. Da würden dann nach wie vor falsche Verbräuche angezeigt werden, wie gerade jetzt:
Mit der [image]
So wie es in der 0.114.0 war, fand ich es optimal.

premultiply, 2023-03-08:
Da lädt es nicht.

fynnsh, 2023-03-08:
Ja, mit Status C sollte es eigentlich gelöst sein...

fynnsh, 2023-03-08:
Es beeinflusst ja nicht die geladenen Werte oder verfälscht sonstige Statistiken dann

PR #6712 (2023-03-09): TWC3: Hide apparent power on standby
Fixes #6677
commit 36bd8cc1 (2023-03-09): TWC3: return VA power only if status C (#6712)

Fixes #6677
closed 2023-03-09 by andig